ZoneMinder/zoneminder #650: Use of undefined constants ZM_SKIN_DEFAULT and ZM_CSS_DEFAULT in index.php

troywill, 2015-01-03:
I'm attempting to build ZoneMinder from the Git repository and am getting a fatal PHP error due to ZM_SKIN_DEFAULT not being defined on line 63 of index.php and ZM_CSS_DEFAULT not being defined on line 72 of index.html.

If I comment out the four lines of code testing for ZM_SKIN_DEFAULT and ZM_CSS_DEFAULT Zoneminder runs. I'm wondering if I borked my build somewhere or if these constants on line 63 and line 72 are buggy.

SteveGilvarry, 2015-01-03:
Are they in your config table? The Pull linked below has them, but I know in the past I had issues with new config fields on old db version. There is some command to refresh options, but escapes me right now.
https://github.com/ZoneMinder/ZoneMinder/pull/645/files

troywill, 2015-01-03:
I wiped my previous zoneminder database and loaded the schema from the current Git db/zm_create.sql.

I see that Options/System has a `SKIN_DEFAULT` field with value set to `classic` and `CSS_DEFAULT` set to `classic.`

SteveGilvarry, 2015-01-03:
Those are the correct values, will need @connortechnology to check it. I am wondering where they get values loaded as index.php is first page and can't see config values being loaded at this stage.

connortechnology, 2015-01-03:
Yeah this is the problem with the way we do the config table.  a zmupdate.pl -f fixes it.

troywill, 2015-01-04:
I got the time to run `zmupdate.pl -f.` A fatal error still occurs due to ZM_SKIN_DEFAULT not being defined.

It appears to me that zmupdate.pl -f simply dumps the Config table into a perl structure, deletes the Config table, and then writes the same Config table back to the database. I don't see how that would have any bearing on the ZM_SKIN_DEFAULT variable not being defined in index.php.

troywill, 2015-01-04:
I believe the fix is to move the `require_once( 'includes/config.php' );` statement prior to the `   elseif ( ZM_SKIN_DEFAULT )` statement.

SteveGilvarry, 2015-01-04:
That was my thought but haven't had time to check that it wasn't dependant on skin being already set when calling config.php. Did you try it?

troywill, 2015-01-04:
Yes.
```php
require_once( 'includes/config.php' ); // Pull in Config values from /etc/zm.conf and Config table in database
print( ZM_SKIN_DEFAULT ); // prints `classic`
```

SteveGilvarry, 2015-01-04:
Good work.

connortechnology, 2015-01-04:
holy crap... must be a merge screw up, mine has the require for config moved up.

closed 2015-01-04 by troywill