ZoneMinder/zoneminder #655: update feature should be removed or easily disabled

wwat, 2015-01-03:
...  via some documented interface (*not* manually via the configuration interface)

`Error check failed: '501 Protocol scheme 'http' is not supported'` results from a check for an update, which is an undesired feature in itself. A program should just run, and another program should worry about updates being available and as such it is a misfeature. I also question the error message itself. It should say *what* doesn't support http. Is it zoneminder? Some server? If so, what server? Communicate clearly what is wrong, thank you very much. 

AFAIK, there is no way to disable the feature completely (for example at compile time).

knnniggett, 2015-01-03:
On a running system, turn off CHECK_FOR_UPDATES under options.
To set it before you build zoneminder, use zmeditconfigdata.sh like so:
https://github.com/ZoneMinder/ZoneMinder/blob/master/distros/fedora/zoneminder.cmake.f20.spec#L73

Run this *before* you run make/cmake!

wwat, 2015-01-03:
That's not part of the 1.28 release.

knnniggett, 2015-01-03:
It's a standalone script.  

You can either:
a) Download the file into your working folder and run it.
b) wait until the next release, which will be "soon"

wwat, 2015-01-03:
I welcome the release in that case, and for now I did the run-time option. You can close this issue when the script is released.

closed 2015-01-04 by knnniggett