ZoneMinder/zoneminder #656: socket_sendto( /var/run/zm/zms-256303s.sock ) failed: No such file or directory

wwat, 2015-01-03:
I get the following message in my log:

```
socket_sendto( /var/run/zm/zms-256303s.sock ) failed: No such file or directory
```

I can only classify this as a bug in the application (version 1.28), because it is displaying an internal error and not an end user understandable message (if the application is misconfigured, then it should report that itself).

connortechnology, 2015-01-03:
This is caused when zms hasn't run or has crashed. Typically this means that cgi isn't turned on or appropriately configured in apache.  Or that zms is crashing.  A quick google would have told you that.

Look in your zms logs. Find out of apache is configured to do cgi.

knnniggett, 2015-01-03:
You can't cherry pick error messages like that w/o giving us context, and then miraculously expect us to produce an answer for you.  We are not magicians.  

Any followup responses need to include a full log, what distro you are running on, and how you installed zoneminder.  It matters.

wwat, 2015-01-03:
I tried to export a log and then I see in my log (the useless message):
Unable to open log export file zm-log.txt	

A correct error message would have stated the complete path to zm-log.txt in the error message. I am trying to get it to run on NixOS (it is currently not packaged on NixOS). Since you probably aren't a NixOS expert, it's probably useless to explain in detail what I did. 

It's running as a systemd service and it calls this service (and in particular, no other one):
/var/setuid-wrappers/sudo -u wwwrun ${unstable.zoneminder}/bin/zmpkg.pl start

Via <ip>/zm I can access the zm web interface and mysql integration is already working.

wwat, 2015-01-03:
This seems relevant (and the reason the process crashes):  
zmc -m 1 returns: 
[0x6f5e528] pulse audio output error: PulseAudio server connection failure: Connection refused
[0x6f993e8] main input error: open of `rtsp://root:@192.168.1.152/h264' failed
[0x6f993e8] main input error: Your input can't be opened
[0x6f993e8] main input error: VLC is unable to open the MRL 'rtsp://root:@192.168.1.152/h264'. Check the log for details.

knnniggett, 2015-01-03:
Your camera url is incorrect. Set a password in the camera and update the url.
If that doesn't fix it then we need to see your monitor config, and the make/model of camera.

connortechnology, 2015-01-03:
If there isn't supposed to be a pasword, then you need to remove the : after root

wwat, 2015-01-03:
If I use /usr/bin/vlc and rtsp://root:@192.168.1.152:554/h264 as the URL, it all works. What do I need to tell Zoneminder in that case? There is also a dropdown with various options (and I have no idea which one to choose, I used the default one). I find it weird that I need to configure something which vlc also doesn't ask. 

The camera is a FCS-5051.

wwat, 2015-01-03:
This URL also doesn't work:
```
[0x7ab0528] pulse audio output error: PulseAudio server connection failure: Connection refused
[0x7ae6428] main input error: open of `rtsp://192.168.1.152:554/h264' failed
[0x7ae6428] main input error: Your input can't be opened
[0x7ae6428] main input error: VLC is unable to open the MRL 'rtsp://192.168.1.152:554/h264'. Check the log for details.
^C^C
```

It also annoys me that I cannot do Control-C, Control-C to exit the program again (that is, there is no signal handler).

wwat, 2015-01-03:
What is the recommended user to run under anyway (and should this be different for the various services)? I am currently running as user wwwrun, but perhaps that's not a good idea. 

```
nixos zmc[5198]: Home directory not accessible: Permission denied
```
"Home directory", please.... How about you just spell out home directory of user myuser located at local filesystem in directory /home/bar currently has permissions <current permissions>, but <permissions> are required, please execute chmod 0755 on this directory as the owner (wwwrun) of this directory.

connortechnology, 2015-01-03:
debian based distros run as www-data.  It's entirely up to the distribution.  However it typically has to be whatever user apache runs as.

closed 2015-01-06 by knnniggett

rlrosa, 2015-03-20:
not sure if this is the place to post.
i had a similar problem on ubuntu 14.04 and it was due to a problem with cgi. everything reported to be correctly cfg and enabled but still got error and no video stream.

fixed it by installing: libapache2-mod-perl2
as mentioned here: http://askubuntu.com/a/461797/83198

wwat, 2015-03-21:
@rlrosa Thanks for your comment; the flaw lies *entirely* with ZoneMinder. A proper program would only do work when its environment is completely correct or it would provide one such example environment, i.e. it would have some concept of a "reset". Obviously, @knnniggett  and @connortechnology  do not agree on this, but then again, they would not recognize a good program in the first place.