ZoneMinder/zoneminder #658: Horrible error message when pressing probe

wwat, 2015-01-04:
When I press probe, I get the following message:
```
Unable determine arp path, status is '1'
```
1) it's not correct English
2) arp is spelt ARP (if that's what it's about)
3) 'status' (???) 
4) '1'(???) (does this mean on or off)

tl;dr WTF?!

kylejohnson, 2015-01-04:
Please feel welcome to submit a PR - seems like an easy fix.  I am assuming 1 is the exit code when attempting to run the arp command.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

SteveGilvarry, 2015-01-04:
Means it couldn't find the file path for the arp command, as opposed to the protocol ARP. Try running 'type -p arp' on your system and you will probably get a 1 returning.

SteveGilvarry, 2015-01-05:
PR done and merged, hopefully my new message is better.

closed 2015-01-05 by knnniggett