ZoneMinder/zoneminder #659: Fake camera support

wwat, 2015-01-04:
It would be useful to have fake cameras for each possible supported source, such that the software can be tested without having to have a real camera around.

SteveGilvarry, 2015-01-04:
I assume you mean camera emulators, there has been talk of these and Axis has some in SDK, but nothing in the public domain that we can use. I would love an RTSP camera emulator to test with.

wwat, 2015-01-04:
Yes, that's what I meant. If I would be building something like Zoneminder from scratch, it would have been one of the first things, I would build.

SteveGilvarry, 2015-01-04:
Cameras were analog devices accessed through capture devices back then, so emulating was probably harder.
In case you are not aware the active development team is pretty small, so if you have the skills to build features like this please consider contributing.

SteveGilvarry, 2015-01-05:
What about having a document on how to setup at least streams for testing? I am looking at live555 for doing Rtsp streams and looks pretty simple.
Makes more sense than developing our own. Testing Pan would need something like axis sdk.

knnniggett, 2015-01-05:
What's wrong with using a camera manufacturer’s demo cameras?
I've used Foscam's demo cameras, streamed from China to my zoneminder development vm here in the US. Works fine.

Axis makes streams from all their cameras available here:
http://www.axis.com/camerademo/
I've not tried these, but I don't see why they wouldn't work.

wwat, 2015-01-05:
Not much, but then they should be integrated into the web-interface as an easy to select option.

knnniggett, 2015-01-05:
I don't agree with that.  
Those cameras are limited in the number of simultaneous users.  Making it a simple click for anyone to view them from zoneminder would be shooting ourselves in the foot.  We would soon find ourselves not able to access those cameras.

wwat, 2015-01-05:
I think those companies should just upgrade their infrastructure then.

knnniggett, 2015-01-05:
It's got nothing to do with their infrastructure.  A single camera will only support a limited number of users (streams).  Increasing the number of streams, makes the camera more expensive.  Zoneminder is targeted towards the home user with budget cameras.

wwat, 2015-01-05:
I don't think you quite get what I mean. You are talking about a URL. There is nothing requiring them to connect the users directly to the camera.

knnniggett, 2015-01-05:
Inserting a proxy between the recorder and zonemidner defeats the purpose of the test.  If there is a problem, we won't be able to show the problem isn't within the proxy itself.

knnniggett, 2015-01-05:
In any case, we did talk about adding exactly this kind of check into our Travis tests.  That is the best place to perform this kind of thing, but I don't know what became of it.

kylejohnson, 2015-01-05:
As I see it, this issue relates to testing automated builds of ZoneMinder.  We've already got support for that: take a look at our [.travis.yml](https://github.com/ZoneMinder/ZoneMinder/blob/master/.travis.yml#L50) file.

If you'd like to expand support, please submit a PR.  Or if you'd like to make camera publicly available for our automated testing purposes, please let us know.

What we shouldn't do is use the Axis provided demo cameras as part of our testing framework - that is not what they're there for, and as @knnniggett said, we'll probably get on our bad side if we do that.

closed 2015-01-05 by kylejohnson