ZoneMinder/zoneminder #661: zmaudit Error : Can't execute: Out of range value for column 'TotScore' at row 1

yarlicurly, 2015-01-06:
In mysql tables of zoneminder TotScore(int unsigned) is parameter of Events table, which is sum of Score(smallint) from Frames table. 
 I am currently working on 6 zoneminders with 4 monitors in each and exact same settings. But the first monitor(which have exact same settings as others) of 2 zoneminders have undesired Events, regular case is to create a Event for around 3000 frames and duration around 300, but for these monitors zm creates 3 Events( one is what expected and other two with frame count aound 0, 50, 700 and soon ) and I am assuming that because of this Totscore value is less than zero.

Being new to zoneminder I am not sure on how to debug the root cause of the problem. I would really appreciate if you can share your thoughts/knowledge on the same. 

Thanks in advance.

connortechnology, 2015-01-08:
I'm not sure I understand how you are getting to a negative totscore.  

The only way to get to a negative when adding unsigned integers is to wrap around... but int unsigned has a max value of 4294967295.  Is that what is happening?  Is your totscore > 4294967295?

yarlicurly, 2015-01-08:
@connortechnology : Thanks for your message. In my tables TotScore range is -30 to 4800 aprox. 
And strangely the negative totscore is because of selected monitors in Linked Monitors which I overlooked earlier. 
After unselecting the linked monitors and handling the erroneous opened events, monitors are working fine.

connortechnology, 2015-01-09:
But totscore is supposed to be unsigned... it shouldn't have any negative values... 

can you verify that totscore is an unsigned int(10) ?

yarlicurly, 2015-01-09:
Yes it is originally unsigned int(10), I have changed it to int(11) to check the values which are causing the error and then I have changed it back to unsiged int.

knnniggett, 2015-08-15:
This might be related to #319 which was recently fixed. Please refer to that for the fix.

If your system was writing "-1" values to the Score column in the Frames table, then that could explain how totscore could go negative as well.  Note that, if you have a table full of -1's, this fix will not correct that.  You will have delete those events out.

If the related fix does not solve this issue then please reopen it and we will investigate further.

closed 2015-08-15 by knnniggett