ZoneMinder/zoneminder #667: Error decoding frame (cam made public)

speeskeek, 2015-01-08:
Hi there,

I got zoneminder running for a few weeks now with 1 camera (cheap chinese thin). It's only @ 1,7fps - becouse I couldn't get the real strem to work. Now I am using the 'snapshot' way. But, it works good, detects things, not to much load and I'm happy.

Now I received a second cam (yet a few bucks cheaper :) ), and I can get it good to work with VLC using the stream : rstp:IP:554/user=admin_password=tlJwpbo6_channel=1_stream=1.sdp

When adding to ZoneMinder, It gives me a stream, good quality, and looks ok (no visible foults in the image, steady framerate, ..). The log however complains:
2015-01-08 08:45:35.674682	zmc_m6	23653	ERR	8: 00 00 01 06 e5 01 b9 80	zm_remote_camera_rtsp.cpp	303
2015-01-08 08:45:35.611255	zmc_m6	23653	ERR	Error while decoding frame 800	zm_remote_camera_rtsp.cpp	302
2015-01-08 08:45:34.357514	zmc_m6	23653	ERR	8: 00 00 01 06 e5 01 b7 80	zm_remote_camera_rtsp.cpp	303
2015-01-08 08:45:33.611396	zmc_m6	23653	ERR	Error while decoding frame 750	zm_remote_camera_rtsp.cpp	302
2015-01-08 08:45:31.655856	zmc_m6	23653	ERR	8: 00 00 01 06 e5 01 b5 80	zm_remote_camera_rtsp.cpp	303
2015-01-08 08:45:31.610902	zmc_m6	23653	ERR	Error while decoding frame 700	zm_remote_camera_rtsp.cpp	302
2015-01-08 08:45:29.688637	zmc_m6	23653	ERR	8: 00 00 01 06 e5 01 b3 80	zm_remote_camera_rtsp.cpp	303
2015-01-08 08:45:29.631817	zmc_m6	23653	ERR	Error while decoding frame 650

Anyone can test how to add this camera correctly?
Here is my config:
OpenMediavault OS (so debian)
[image]
[image]
[image]
[image]

speeskeek, 2015-01-09:
Some more info on the codec
When executing (I'm trying to make stills with ffmpeg, and read those stills):
ffmpeg -ss 2 -i rtsp://192.168.1.122:554/user=admin_password=tlJwpbo6_channel=1_stream=1.sdp -y -f image2 -sameq -t 5 /var/www/foo.jpeg

I get following:

ffmpeg version 0.8.16-6:0.8.16-1, Copyright (c) 2000-2014 the Libav developers
  built on Sep 16 2014 23:10:48 with gcc 4.7.2
The ffmpeg program is only provided for script compatibility and will be removed
in a future release. It has been deprecated in the Libav project to allow for
incompatible command line syntax improvements in its replacement called avconv
(see Changelog for details). Please use avconv instead.
[rtsp @ 0x11efb20] Estimating duration from bitrate, this may be inaccurate
Input #0, rtsp, from 'rtsp://192.168.1.122:554/user=admin_password=tlJwpbo6_channel=1_stream=1.sdp':
  Metadata:
    title           : RTSP Session
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0.0: Video: h264 (Main), yuv420p, 352x288, 25 fps, 25 tbr, 90k tbn, 50 tbc
Incompatible pixel format 'yuv420p' for codec 'mjpeg', auto-selecting format 'yuvj420p'
[buffer @ 0x11f17a0] w:352 h:288 pixfmt:yuv420p
[avsink @ 0x11f6920] auto-inserting filter 'auto-inserted scaler 0' between the filter 'src' and the filter 'out'
[scale @ 0x11f3480] w:352 h:288 fmt:yuv420p -> w:352 h:288 fmt:yuvj420p flags:0x4
Output #0, image2, to '/var/www/foo.jpeg':
  Metadata:
    title           : RTSP Session
    encoder         : Lavf53.21.1
    Stream #0.0: Video: mjpeg, yuvj420p, 352x288, q=2-31, 200 kb/s, 90k tbn, 25 tbc
Stream mapping:
  Stream #0.0 -> #0.0
Press ctrl-c to stop encoding
[image2 @ 0x11f23e0] Could not get frame filename number 2 from pattern '/var/www/foo.jpeg'
av_interleaved_write_frame(): Input/output error

connortechnology, 2015-01-09:
Um, I use RTSP with h264. 

RTSP actually uses the ffmpeg libraries, so whatever they support, RTSP supports (I think).

speeskeek, 2015-01-09:
yes, according to me I is a simple config or dependency that is wrong _somewhere_.
Because like I said, it looks nice in the 'stream-view window' of zoneminder.

SteveGilvarry, 2015-01-09:
My mistake I thought it was just MPEG, will have to have a look at it now. I was sure I have seen errors about h264 with RTSP.

knnniggett, 2015-08-10:
Closing due to inactivity.  If this is camera is still causing problems, the op should try again with a newer version of ffmpeg, 0.8 is rather old.

closed 2015-08-10 by knnniggett

shubhampratapsingh1, 2018-02-13:
use following command :

$ rosrun usb_cam usb_cam_node _video_device:=/dev/video0 _image_height:=480 _image_width:=640  _pixel_format:=yuyv _framerate:=10