ZoneMinder/zoneminder #669: zmc won't start automatically + Can't open memory map file

jonnsl, 2015-01-10:
If I start zmc manually (e.g. ```# zmc -m 6```) the error goes from:
```
Memory map file '/dev/shm/zm.mmap.6' should have been 896 but was instead 0 zmwatch.pl
```
to:
```
Can't open memory map file '/dev/shm/zm.mmap.6': Permission denied	zmwatch.pl
```
and eventually:
```
Can't open memory map file /dev/shm/zm.mmap.6, probably not enough space free: Permission denied
```

```
# cat /proc/sys/kernel/shmmax
18446744073692774399
```
I installed zoneminder from this package: https://aur.archlinux.org/packages/zoneminder-xinetd (I removed the dependency cambozola, and changed the dependency from perl-x10 to perl-x10-zoneminder)
zoneminder version: 1.27.0
perl version: 5.20.1-1
php version: 5.6.4-1
nginx version: 1.6.2-2
ffmpeg:
```
ffmpeg version 2.5.2 Copyright (c) 2000-2014 the FFmpeg developers
  built on Dec 26 2014 11:50:46 with gcc 4.9.2 (GCC)
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-avisynth --enable-avresample --enable-fontconfig --enable-gnutls --enable-gpl --enable-libass --enable-libbluray --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libspeex --enable-libtheora --enable-libv4l2 --enable-libvorbis --enable-libvpx --enable-libx264 --enable-libx265 --enable-libxvid --enable-runtime-cpudetect --enable-shared --enable-swresample --enable-vdpau --enable-version3 --enable-x11grab
  libavutil      54. 15.100 / 54. 15.100
  libavcodec     56. 13.100 / 56. 13.100
  libavformat    56. 15.102 / 56. 15.102
  libavdevice    56.  3.100 / 56.  3.100
  libavfilter     5.  2.103 /  5.  2.103
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  1.100 /  1.  1.100
  libpostproc    53.  3.100 / 53.  3.100
```

zoneminder log:
```
2015-01-10 01:45:44.890660	zmdc	31057	INF	Server shutdown at 15/01/10 01:45:44	zmdc.pl	
2015-01-10 01:45:34.808760	zmdc	31057	INF	'zmwatch.pl ' exited, signal 14	zmdc.pl	
2015-01-10 01:45:34.775050	zmdc	31057	INF	'zmwatch.pl ' stopping at 15/01/10 01:45:34	zmdc.pl	
2015-01-10 01:45:34.750930	zmdc	31057	INF	'zmupdate.pl -c' exited, signal 14	zmdc.pl	
2015-01-10 01:45:34.650410	zmdc	31057	INF	'zmupdate.pl -c' stopping at 15/01/10 01:45:34	zmdc.pl	
2015-01-10 01:45:34.619320	zmdc	31057	INF	'zmfilter.pl ' exited, signal 14	zmdc.pl	
2015-01-10 01:45:34.592070	zmdc	31057	INF	'zmfilter.pl ' stopping at 15/01/10 01:45:34	zmdc.pl	
2015-01-10 01:45:34.567750	zmdc	31057	INF	'zmaudit.pl -c' exited, signal 14	zmdc.pl	
2015-01-10 01:45:34.538750	zmdc	31057	INF	'zmaudit.pl -c' stopping at 15/01/10 01:45:34	zmdc.pl	
2015-01-10 01:45:34.309410	zmpkg	31188	INF	Command: stop	zmpkg.pl	
2015-01-10 01:45:14.692440	zmwatch	31103	ERR	Can't open memory map file '/dev/shm/zm.mmap.6': Permission denied	zmwatch.pl	
2015-01-10 01:45:12.756652	zmc_m6	31150	INF	Got signal 2 (Interrupt), exiting	zm_signal.cpp	40
2015-01-10 01:45:04.664710	zmwatch	31103	ERR	Can't open memory map file '/dev/shm/zm.mmap.6': Permission denied	zmwatch.pl	
2015-01-10 01:44:56.142759	zmu	31171	FAT	Can't open memory map file /dev/shm/zm.mmap.6, probably not enough space free: Permission denied	zm_monitor.cpp	392
2015-01-10 01:44:54.610890	zmwatch	31103	ERR	Can't open memory map file '/dev/shm/zm.mmap.6': Permission denied	zmwatch.pl	
2015-01-10 01:44:44.654085	zmc_m6	31150	INF	Monitor-6: 2000 - Capturing at 30.30 fps	zm_monitor.cpp	2831
2015-01-10 01:44:44.583890	zmwatch	31103	ERR	Can't open memory map file '/dev/shm/zm.mmap.6': Permission denied	zmwatch.pl	
2015-01-10 01:44:34.556940	zmwatch	31103	ERR	Can't open memory map file '/dev/shm/zm.mmap.6': Permission denied	zmwatch.pl	
2015-01-10 01:44:24.529950	zmwatch	31103	ERR	Can't open memory map file '/dev/shm/zm.mmap.6': Permission denied	zmwatch.pl	
2015-01-10 01:44:14.501970	zmwatch	31103	ERR	Can't open memory map file '/dev/shm/zm.mmap.6': Permission denied	zmwatch.pl	
2015-01-10 01:44:11.197176	zmc_m6	31150	INF	Monitor-6: 1000 - Capturing at 29.41 fps	zm_monitor.cpp	2831
2015-01-10 01:44:04.474950	zmwatch	31103	ERR	Can't open memory map file '/dev/shm/zm.mmap.6': Permission denied	zmwatch.pl	
2015-01-10 01:43:54.439510	zmwatch	31103	ERR	Can't open memory map file '/dev/shm/zm.mmap.6': Permission denied	zmwatch.pl	
2015-01-10 01:43:44.411660	zmwatch	31103	ERR	Can't open memory map file '/dev/shm/zm.mmap.6': Permission denied	zmwatch.pl	
2015-01-10 01:43:37.326082	zmc_m6	31150	INF	Priming capture from rtsp://888888:987654@192.168.1.151:554/cam/realmonitor?channel=1&subtype=0	zm_ffmpeg_camera.cpp	112
2015-01-10 01:43:37.305720	zmc_m6	31150	INF	Starting Capture	zmc.cpp	195
2015-01-10 01:43:34.384770	zmwatch	31103	ERR	Memory map file '/dev/shm/zm.mmap.6' should have been 896 but was instead 0	zmwatch.pl	
2015-01-10 01:43:24.363990	zmwatch	31103	ERR	Memory map file '/dev/shm/zm.mmap.6' should have been 896 but was instead 0	zmwatch.pl	
2015-01-10 01:43:14.341660	zmwatch	31103	ERR	Memory map file '/dev/shm/zm.mmap.6' should have been 896 but was instead 0	zmwatch.pl	
2015-01-10 01:41:34.277730	zmwatch	31103	INF	Watchdog pausing for 30 seconds	zmwatch.pl	
[... 101 lines cut ...]
[h264 @ 0x6c5e900] bytestream overread -3
[h264 @ 0x6c5e900] bytestream overread -9
[h264 @ 0x6c5e900] error while decoding MB 43 19, bytestream -9
[h264 @ 0x6c5e900] Cannot use next picture in error concealment
[h264 @ 0x6c5e900] concealing 490 DC, 490 AC, 490 MV errors in P frame
[h264 @ 0x6be6000] RTP: missed 3 packets
[h264 @ 0x6c5e900] Frame num gap 3 1
[h264 @ 0x6eb5440] left block unavailable for requested intra mode at 0 18
[h264 @ 0x6eb5440] error while decoding MB 0 18, bytestream 3885
[h264 @ 0x6eb5440] Cannot use next picture in error concealment
[h264 @ 0x6eb5440] concealing 577 DC, 577 AC, 577 MV errors in P frame
[h264 @ 0x6be6000] RTP: missed 2 packets
[h264 @ 0x6be8b40] left block unavailable for requested intra4x4 mode -1 at 0 24
[h264 @ 0x6be8b40] error while decoding MB 0 24, bytestream 10294
[h264 @ 0x6be8b40] concealing 313 DC, 313 AC, 313 MV errors in I frame
[h264 @ 0x6be6000] RTP: missed 3 packets
[h264 @ 0x6be6000] RTP: missed 4 packets
[h264 @ 0x6be8fa0] Frame num gap 3 1
[h264 @ 0x6be6000] RTP: missed 1 packets
[h264 @ 0x6c5e900] left block unavailable for requested intra mode at 0 12
[h264 @ 0x6c5e900] error while decoding MB 0 12, bytestream 33387
[h264 @ 0x6c5e900] concealing 841 DC, 841 AC, 841 MV errors in I frame
```

Hope this is enough info.

connortechnology, 2015-01-10:
if you run zmc as root, the memory map file will be owned by root and nothing else will be able to mmap it because generally run as www-data.  run zmc as www-data

jonnsl, 2015-01-10:
you are correct. if I run ```runuser -l http -c '/usr/bin/zmc -m 8'``` there are no more errors about the memory map file.
The name of the monitor isn't a link, as if there was an error, but if I access '?view=watch&mid=8' I can see the image from the camera just fine. 
Shouldn't zmc be called automatically by /usr/bin/zmpkg.pl?

closed 2015-01-11 by jonnsl