ZoneMinder/zoneminder #670: Nonfunctioning event replay controls (pause,play,and others); disappearing socket

troywill, 2015-01-11:
I can record and play back events fine. However, none of the 9 event replay controls (Prev,Rewind,Step Back,Pause,Play,Step Forward,Fast Forward,Zoom Out,Next) do anything.

I believe a socket is initially getting created in zoneminder/ajax/stream.php

    if ( !@socket_sendto( $socket, $msg, strlen($msg), 0, $remSockFile ) )
    {
    ajaxError( "socket_sendto( $remSockFile ) failed: ".socket_strerror(socket_last_error()) );
    }

For example, on my last view of an event `/srv/zoneminder/socks/zms-528567s.sock` was created but it quickly disappeared. 

After the event stops playing I end up with errors in my apache log that the socket couldn't be found:

    ERR [socket_sendto( /srv/zoneminder/socks/zms-912267s.sock ) failed: No such file or directory], referer: http://192.168.1.14/zm/index.php?view=event&eid=176&filter[terms][0]    [attr]=MonitorId&filter[terms][0][op]=%3D&filter[terms][0][val]=1&sort_field=StartTime&sort_asc=1&page=1

    ERR [getCmdResponse stream error: socket_sendto( /srv/zoneminder/socks/zms-912267s.sock ) failed: No such file or directory - checkStreamForErrors()], referer: http://192.168.1.14/zm/index.php?view=event&eid=176&filter[terms][0][attr]=MonitorId&filter[terms][0][op]=%3D&filter[terms][0]     [val]=1&sort_field=StartTime&sort_asc=1&page=1

I'm conjecturing that none of the controls work because they depend on a socket that is getting deleted. If anyone has a clue how I could troubleshoot the socket set up in zoneminder/ajax/stream.php such that my controls are working I'd be very grateful.

knnniggett, 2015-01-11:
I can't remember the last time a "socket_sendto" error was the result of anything but a configuration error.

The first place to start is your apache config.  Make sure you ScriptAlias is working and does not overlap.  Other places to look are to verify your PATH_ZMS is correct.  Make sure your socketdir exists and is writable by the web account.

If the issue is just with a single camera, then your monitor config is suspect.

Zoneminder creates two sockets.  I forget which is which.

troywill, 2015-01-11:
Thanks for the suggestions. Since Zoneminder is recording and playing back events I think my ScriptAlias and PATH_ZMS are alright. I checked to see if PATH_ZMS was alright by hiding my zms and nph-zms binaries, and after I did that I'm unable to play back events. I put zms and nph-zms back and am able to play back events again.

I'll try to grep through the code again now and see what is function is removing the sockets.

knnniggett, 2015-01-12:
How much memory does your system have?

troywill, 2015-01-12:
```
              total        used        free      shared  buff/cache   available
Mem:           997M        189M        633M        9.3M        174M        671M
Swap:          127M          0B        127M
```
(It's a an ASUS netbook)

I've just compiled from source on my laptop now targeting the Apache DocumentRoot and setting a /zoneminder directory for ZM_SOCKDIR,ZM_CONTENTDIR/ZM_TMPDIR to try to isolate where the problem is arising. Will find out soon if there are errors.

knnniggett, 2015-01-12:
How many cameras are you trying to use?
With that little ram, your machine's ramdisk (a.k.a. tmpfs) will be limited to just 512MB (default max is 50% of your physical ram).
You can easily hit that limit with only a handful of high resolution cameras, or several low res cameras.

For example, I've got 8 VGA'ish cameras on my system and my ramdisk is over 600mb.
If you are hitting that limit, that might explain why the socket appears to go away.

troywill, 2015-01-12:
I'm getting the same type of monitor errors after building from source on my laptop that has quite a bit of memory. For example, I get this whenever pressing a control button (e.g. Pause or Play):
```
web_php[12189]: ERR [socket_sendto( /zoneminder/socks/zms-109495s.sock ) failed: Connection refused]
web_php[12189]: ERR [getStreamCmdResponse stream error: socket_sendto( /zoneminder/socks/zms-109495s.sock ) failed: Connection refused - checkStreamForErrors()
```
`/zoneminder/socks/zms-109495s.sock` does exist:
```
ls -l /zoneminder/socks/zms-109495s.sock
srwxr-xr-x 1 http http 0 Jan 11 23:35 /zoneminder/socks/zms-109495s.sock

troywill, 2015-01-17:
Is possible the following warning has an effect on the socket error I'm (and at least one other Zoneminder on Arch Linux user am having?:
```
2015-01-16 15:33:16.959869	zms	1054	WAR	Shared data not initialised by capture daemon, some query functions may not be available or produce invalid results
/tmp/zoneminder/src/ZoneMinder-1.28.0/src/zm_monitor.cpp	424
2015-01-16 15:33:25.875775	zms	994	INF	Got signal 15 (Terminated), exiting	/tmp/zoneminder/src/ZoneMinder-1.28.0/src/zm_signal.cpp	40
```
Then shortly after socket errors appear in the log:
```
2015-01-16 15:33:27.536751	web_php	946	ERR	socket_bind( /srv/zoneminder/socks/zms-547319w.sock ) failed: Address already in use	zoneminder/includes/functions.php	2330
2015-01-16 15:33:27.617882	web_js	439	ERR	getCmdResponse stream error: socket_bind( /srv/zoneminder/socks/zms-547319w.sock ) failed: Address already in use - checkStreamForErrors()	?view=event	
2015-01-16 15:33:33.484932	web_php	946	ERR	socket_bind( /srv/zoneminder/socks/zms-547319w.sock ) failed: Address already in use	zoneminder/includes/functions.php	2330
2015-01-16 15:33:33.543698	web_js	946	ERR	getCmdResponse stream error: socket_bind( /srv/zoneminder/socks/zms-547319w.sock ) failed: Address already in use - checkStreamForErrors()	?view=event	
2015-01-16 15:33:35.224071	web_php	946	ERR	socket_bind( /srv/zoneminder/socks/zms-547319w.sock ) failed: Address already in use	zoneminder/includes/functions.php	2330
2015-01-16 15:33:35.283494	web_js	946	ERR	getCmdResponse stream error: socket_bind( /srv/zoneminder/socks/zms-547319w.sock ) failed: Address already in use - checkStreamForErrors()	?view=event	
2015-01-16 15:33:37.445702	web_php	946	ERR	socket_bind( /srv/zoneminder/socks/zms-547319w.sock ) failed: Address already in use	zoneminder/includes/functions.php	2330
2015-01-16 15:33:37.492482	web_js	946	ERR	getCmdResponse stream error: socket_bind( /srv/zoneminder/socks/zms-547319w.sock ) failed: Address already in use - checkStreamForErrors()	?view=event	
```
Just a wild guess, but could the Zoneminder tmp dir on Arch Linux not being on a tmpfs filesystem be a problem? Here's how zoneminder is currently built on Arch Linux:
```
cmake -DCMAKE_INSTALL_PREFIX=/usr \
      -DZM_PERL_SUBPREFIX=/lib/perl5 \
      -DZM_WEBDIR=/srv/http/zoneminder \
      -DZM_CGIDIR=/srv/http/cgi-bin \
      -DZM_WEB_USER=http \
      -DZM_CONTENTDIR=/var/cache/zoneminder \
      -DZM_LOGDIR=/var/log/zoneminder \
      -DZM_RUNDIR=/srv/zoneminder \
      -DZM_TMPDIR=/srv/zoneminder/tmp \
      -DZM_SOCKDIR=/srv/zoneminder/socks .
```
ZM_RUNDIR,ZM_TMPDIR, and ZM_SOCKDIR are all on a regular filesystem.

Thanks.

knnniggett, 2015-01-17:
No, you absolutely do not want zoneminder to use a tmpfs tmpdir.  That is known to cause other problems and is why the default has been changed to /var/tmp.  

What are the file permissions on the folders in question?  Can the web user account write to them?

troywill, 2015-01-17:
Yes, I remember the problem with the tmpdir, caused me to lose more of my hair.

The web user account on Arch Linux is http and it owns the directories
```
[troy@kangaroo zoneminder]$ ls -ld /srv/zoneminder/{tmp,socks} /var/cache/zoneminder/* /srv/http/{cgi-bin,zoneminder}
drwxr-xr-x  2 http http 4096 Jan 16 18:32 /srv/http/cgi-bin
drwxr-xr-x 12 http http 4096 Jan 16 18:32 /srv/http/zoneminder
drwxr-xr-x  2 http http 4096 Jan 16 18:32 /srv/zoneminder/socks
drwxr-xr-x  2 http http 4096 Jan 16 18:32 /srv/zoneminder/tmp
drwxr-xr-x  2 http http 4096 Jan 16 18:32 /var/cache/zoneminder/events
drwxr-xr-x  2 http http 4096 Jan 16 18:32 /var/cache/zoneminder/images
drwxr-xr-x  2 http http 4096 Jan 16 18:32 /var/cache/zoneminder/temp
```

troywill, 2015-01-18:
I made the interesting discovery ( well interesting to me anyway ) that I'm _not_ getting socket errors when use Chrome on my Android phone. I _am_ getting socket errors when using Chromium on i686 Arch Linux.

cspenceiv, 2015-01-22:
I was unable to replicate the same issues using Chromium or Firefox on i686 Arch Linux. It is also behaving well with Windows 7 x64 using Chrome and Firefox.

knnniggett, 2015-08-13:
I'm closing this out because the information provided does not describe a bug, but appears to be a configuration issue.  If you are still experiencing this issue, feel free to contact us in the ZoneMinder IRC channel or post your question in the user forum.

closed 2015-08-13 by knnniggett

Jbonnett, 2016-04-25:
bug or not, lots of people have this issue. Looking for information on it just leads to a closed loop of tickets closed as dupe or as "not a bug." Clearly something about this system needs to be made easier to configure if all these people have the same problem.

Jbonnett, 2016-04-25:
i actually have a new version of this issue, which is a actual bug: on mysql 5.7, with the default configuration you will see 
`FAT [SQL-ERR 'SQLSTATE[42000]: Syntax error or access violation: 1055 Expression #2 of SELECT list is not in GROUP BY clause and contains nonaggregated column 'zm.Events.MonitorId' which is not functionally dependent on columns in GROUP BY clause; this is incompatible with sql_mode=only_full_group_by', statement was 'select Events.Id as Id, MonitorId, Name, Cause, StartTime, date_format( StartTime, '%y/%m/%d %H:%i:%S' ) as StartTimeShort, EndTime, Width, Height, Length, Frames, AlarmFrames, TotScore, AvgScore, MaxScore, Archived, Videoed, Uploaded, Emailed, Messaged, Executed, Notes, min(Frames.FrameId) as MinFrameId, max(Frames.FrameId) as MaxFrameId, min(Frames.Delta) as MinFrameDelta, max(Frames.Delta) as MaxFrameDelta from Events left join Frames on Events.Id = Frames.EventId where Events.Id = ? group by Frames.EventId limit 1']`

in the apache log. This is caused by only_full_group_by mode, which is the new default in 5.7.  Assuming that zoneminder is the only thing using mysql on this system, you can fix this by running `SET global sql_mode = 'STRICT_TRANS_TABLES,NO_ZERO_IN_DATE,NO_ZERO_DATE,ERROR_FOR_DIVISION_BY_ZERO,NO_AUTO_CREATE_USER,NO_ENGINE_SUBSTITUTION';`

SteveGilvarry, 2016-04-25:
@Jbonnett second post not actually related at all to socket issues, but correct it is a bug, along with php db updates pushing '' instead of NULL.
Details on the workaround sql_mode already in the guide for Ubuntu 16.04.
https://wiki.zoneminder.com/Ubuntu_Server_16.04_64-bit_with_Zoneminder_1.29.0_the_easy_way.

Problem with zms_path and sockets is ongoing and I agree something needs to be done to help people work it out. We really have not seen any actual bugs demonstrated here, they tend to be some sort of configuration that has been missed or changed from default. But maybe we need to find a way to detect mis-configuration and issue warnings.

Jbonnett, 2016-04-25:
Fixing the SQL mode fixed the socket issue for me. Zms path was right. I
suspect that the socket creator was hitting the same bug, but I don't have
logs.