ZoneMinder/zoneminder #672: The README.opensuse file is misleading regarding mysql configuration file.

jornmoe, 2015-01-11:
After having issues with creating and saving monitors and alarm zones I finaly found out what was wrong (see the MySQL 5.6 compatibility issues #649) . Basically the instructions in README.opensuse mislead me to belive I was running mariadb, and that the my.cnf file was located in /etc/. But I am running mysql-community-server and its' configuration file is in /usr/!

I have reread the README.opensuse in 1.28, and it clearly states that it is in /etc/my.cnf you need to comment out "sql_mode=NO_ENGINE_SUBSTITUTION,STRICT_TRANS_TABLES". 
That may be correct if you use mariadb implementation of mysql. However, if you use the mysql-community-server from the opensuse repo for 13.2, the effective settings are in /usr/my.cnf

I guess that if you during installation of opensuse selects you want to install a LAMP server, you get mariadb(?), but I preffer to install OS with as little software as possible and add software as needed, and then tends to select software I have used before (hence mysql-community-server). And honestly, I didn't know of mariadb until i started reading the above mentioned README.OpenSuse file. The wording in that file led me to belive that I actually was running mariadb, even if I did not.

It would be nice if the /distros/opensuse/README.OpenSuse reflected the difference in where the my.cnf file is for the two differnt versions of mysql you get with opensuse. How do I go forward to propose changes to this file?

newburns, 2015-01-11:
You can fork the project.
Make the change, then push a "pull request"
Make sure you have good comments on your commit

knnniggett, 2015-01-11:
The distros/opensus folder needs some love.  We need someone to maintain this folder over time.  While I have never used Opensuse, I am familiar with redhat rpm packaging so I can help anyone willing to keep it up to date.

Note that the opensuse rpms found on the ZoneMinder WiKi are done by a third party and won't necessarily reflect what you see on github.  <- I'd like to change this someday.

I've been in contact with the author, but so far I haven't been able to get him to make the step towards integrating his work into ours.  

But to answer your question, you can fork the repo like @newburns suggested.  Since you intend to edit just a single file, you can do it all from your web browser and won't have to git-clone the repo to your local hard drive.

knnniggett, 2015-08-09:
Closing due to inactivity.  The OpenSuse distro folder is currently without an active maintainer.  Feel free to submit a pr addressing this issue.

closed 2015-08-09 by knnniggett