ZoneMinder/zoneminder #674: Make Monitor Probe useful

connortechnology, 2015-01-13:
For a local camera, it should scan /dev/video*.  If there is already a monitor setup for /dev/video0 but not for /dev/video1 then it should setup /dev/video1.  It should detect optimal palette, colorspace, maximum size, etc.

For remote cameras I guess we just need to add more camera support.

knnniggett, 2015-01-13:
Note that the ONVIF feature will supersede the current probe method for ip cameras.

connortechnology, 2015-01-13:
Note that ONVIF isn't supported on older cameras...

knnniggett, 2015-01-14:
Yes, of course. I was merely pointing out that we don't need to add support for ONVIF cameras using this method.  I could probably have said that more clearly.

connortechnology, 2015-01-14:
but maybe this should run the onvif code...

closed 2017-03-18 by SteveGilvarry