ZoneMinder/zoneminder #677: CMAKE ZM_PERL_SUBPREFIX Slash Issue

cspenceiv, 2015-01-14:
Applicable version: 1.28.0
Build environment: Arch Linux x86_64 ABS

The current INSTALL file states:
<i>"ZM_PERL_SUBPREFIX Use a different directory for the zm perl modules. NOTE: This is a subprefix, e.g. lib will be turned into <prefix>/lib, default: <libarch>/perl5"</i>

This explanation is either ambiguous/wrong or the scripts are broken behind the scenes. Consider the following examples generated on an Arch ABS x86_64 setup. Keep in mind that pkg/zoneminder is essentially a build directory.

<b>ZM_PERL_SUBPREFIX=lib/perl5</b>
Places the files that should be at pkg/zoneminder/usr/lib/perl5/ into pkg/zoneminder/usr/tmp/zoneminder/src/ZoneMinder-1.28.0/lib/perl5/. This is way off.

<b>ZM_PERL_SUBPREFIX=/usr/lib/perl5/</b>
Places the files at pkg/zoneminder/usr/usr/lib/perl5/. 

<b>ZM_PERL_SUBPREFIX=/lib/perl5</b>
Places the files at pkg/zoneminder/usr/lib/perl5/ as originally desired.

Without code changes, you would want to amend the INSTALL file to say:
<i>"ZM_PERL_SUBPREFIX Use a different directory for the zm perl modules. NOTE: This is a subprefix, e.g. /lib will be turned into <prefix>/lib, default: <prefix>/lib/perl5"</i>

Otherwise, it appears that there is an interesting directory slash bug somewhere.

knnniggett, 2015-01-15:
Zoneminder will install the perl modules into the following folder:
${DESTDIR}${PREFIX}${PERL_SUBPREFIX}

That is why your third attempt works and the others don't.
I've never tried using a relative folder, and I don't see how that could ever work.  I'll update the documentation.

knnniggett, 2015-01-15:
Fixed in commit d8df44fefe4ea0266e270fd5de0d2e149cda1980

Note that your original quote isn't quite right.  I think github may have stripped out some of it when you pasted it. 
`ZM_PERL_SUBPREFIX Use a different directory for the zm perl modules. NOTE: This is a subprefix, e.g. lib will be turned into <prefix>/lib, default: <libarch>/perl5`

This is essentially correct.  I added a "/" for clarification:
`ZM_PERL_SUBPREFIX Use a different directory for the zm perl modules. NOTE: This is a subprefix, e.g. /lib will be turned into <prefix>/lib, default: <libarch>/perl5`

cspenceiv, 2015-01-15:
I just fixed my original post. You were right. I forgot that GitHub recognizes the &lt; and &gt; symbols as HTML.

That fix should make it a little less confusing and save people like me a compile or two.

In respect to using a relative folder, it depends if the prefix was using a trailing slash. But most projects assume that the user does not include a trailing slash so any additional variable content is added to the end of the prefix with a leading slash. Since double forward slashes are acceptable, it always works to assume you need a leading slash.

One solution to absolutely make sure it works: ${DESTDIR}${PREFIX}/${PERL_SUBPREFIX}

But if the documentation is clear, it shouldn't be needed and this is such a minor thing anyway.

knnniggett, 2015-06-17:
Closing this one out as commit 0cf82a0b71cd2ada18b2e388cc3ee62fc0beacf9 rewrites the code in question.

closed 2015-06-17 by knnniggett