ZoneMinder/zoneminder #678: Can't create swap file/directory

speeddragon, 2015-01-16:
Error log is full of 
"Can't open /var/tmp/zm/zmswap-m1/zmswap-q088358/zmswap-i00923.jpg: No such file or directory"

ZM have permission and the directory is 777.
It creates the zmswap-m1 but not the subfolder.

troywill, 2015-01-17:
Are you running Arch Linux? If not, what are you running? (I've been encountering this on Arch Linux.)

speeddragon, 2015-01-17:
Running Debian 7.5

troywill, 2015-01-17:
OK, thanks. I'm investigating the problem. I think I looked at the code and logs and found that zoneminder creates the subfolder (/zmswap-q822199 in your log) and then deletes the directory it had just created. I'll try to get the problem analyzed today time permitting.

knnniggett, 2015-08-09:
Closing due to inactivity.  Zoneminder packages are now available through official Debian sources.  Recommend you install zoneminder via that method.

closed 2015-08-09 by knnniggett