ZoneMinder/zoneminder #681: Gentoo Systemd ZM V1.28.0- New old bug with PATH_SOCK and PATH_SWAP

christophey2k, 2015-01-19:
Hello all the team, 
Have upgraded one of my server with Gentoo Linux, Systemd  and ZoneMinder V1.28.00
and have the big problem with .sock files .....
(have another srv with older zm1.27.99 with identical setup that run fine)

Error:
web_php	1838	ERR	socket_sendto( /var/run/zm/zms-139675s.sock ) failed: Aucun fichier ou dossier de ce type	includes/functions.php 2325

Aucun fichier ou dossier = no file or folder
And have no camera image ...

My setup:
# cmake -DZM_DB_HOST="localhost" -DZM_DB_NAME="zm" -DZM_DB_USER="zmuser" -DZM_DB_PASS="zmpass" -DZM_WEBDIR="/var/www/zoneminder.mydomain.fr/htdocs" -DZM_CGIDIR="/var/www/zoneminder.mydomain.fr/cgi-bin" -DZM_CONTENTDIR="/ZM4To/zoneminder_data" -DZM_EXTRA_LIBS="ssl" -DCMAKE_VERBOSE_MAKEFILE="ON" -DCMAKE_INSTALL_PREFIX="/usr" -DCMAKE_CXX_FLAGS_RELEASE="-D__STDC_CONSTANT_MACROS" -DZM_RUNDIR="/run/zm" -DZM_LOGDIR="/var/log/zm" -DZM_TMPDIR="/ZM4To/swap_zm" -DZM_WEB_USER="apache" -DZM_WEB_GROUP="apache" .

# mkdir /ZM4To/swap_zm
# chown -R apache:apache /ZM4To/swap_zm
# chmod 0755 /ZM4To/swap_zm

# mkdir /run/zm
# chown -R apache:apache /run/zm
# chmod 0755 /run/zm

# nano -w /etc/tmpfiles.d/zoneminder.conf
<--->
d  /run/zm	 0755	 apache apache
<--->
# chmod 755 /etc/tmpfiles.d/zoneminder.conf

# nano -w /etc/systemd/system/zoneminder.service
<--->
# ZoneMinder systemd unit file
# This file is intended to work with Gentoo Linux distributions

[Unit]
Description=ZoneMinder CCTV recording and security system
After=network.target mysqld.service apache2.service
Requires=mysqld.service apache2.service

[Service]
User=apache
Type=forking
PIDFile=/run/zm/zm.pid
ExecStart=/usr/bin/zmpkg.pl start
ExecReload=/usr/bin/zmpkg.pl restart
ExecStop=/bin/bash -c '[[ "$(/usr/bin/pgrep zmdc.pl)" > 0 ]] && /usr/bin/zmpkg.$

[Install]
WantedBy=multi-user.target
<--->
Under ZM web interface have this settings:

SETUP IN "Options" -> "Path"	:
-----------------------------------------
	-> PATH_ZMS		=	/cgi-bin/nph-zms
	-> PATH_MAP		=	/dev/shm
	-> PATH_SOCKS		=	/run/zm
	-> PATH_LOGS		=	/var/log/zm
	-> PATH_SWAP		=	/ZM4To/swap_zm

Thank for watching

knnniggett, 2015-01-22:
See how you've got -DZM_RUNDIR="/run/zm" but the sock file is being created under /var/run/zm?
Yes, /var/run is probably a symlink back to /run, but this discrepancy tells me that you should verify the path values under Options match the values you built zoneminder with.  

These can get out of sync if you start with a database from a previous build rather than create a new database.

christophey2k, 2015-01-22:
Hello, thank for your reply, i start with full new empty db, and i change path with /run/zm in cmake and zm with same error
----
web_php	30815	ERR	socket_sendto( /run/zm/zms-765587s.sock ) failed: Aucun fichier ou dossier de ce type	includes/functions.php	2325
2015-01-22 21:17:26.132701	web_php	30814	ERR	socket_sendto( /run/zm/zms-953437s.sock ) failed: Aucun fichier ou dossier de ce type	includes/functions.php	2325
---
in /run/zm
have 2 files  : 
drwxr-xr-x  2 apache apache  80 22 janv. 21:17 .
drwxr-xr-x 22 root   root   540 19 janv. 11:12 ..
srwxr-xr-x  1 apache apache   0 22 janv. 21:09 zmdc.sock
-rw-r--r--  1 apache apache   5 22 janv. 21:09 zm.pid

in /ZM4To/swap_zm directory have nothing ...

christophey2k, 2015-01-22:
another,  i try to recompile with /var/run/zm and 
the log stay blocked with error in socket_sendto( /run/zm/zms-937320s.sock ) failed
it seem i can't change the value of -DZM_RUNDIR to /var/run/zm

christophey2k, 2015-01-24:
i try many configuration with fresh full new (compil & DB) install playing with directory variables with no success

for example
i found "zmdc.sock and zms-424616w.sock" file in sock directory (designed by PATH_SOCKS = ZM_SOCKDIR) 

and have an error who say system can't found
web_php 30815 ERR socket_sendto( /run/zmzms-424616s.sock.sock

in:  https://github.com/ZoneMinder/ZoneMinder/releases
you explain :
ZoneMinder now requires the polkit package and the URI::Encode perl module.
ZM_TMPDIR now defaults to /var/tmp
and
ZM_SOCKDIR now defaults to /var/run
but in the DOCUMENTATION file in source code for cmake
Possible configuration options:
    ZM_RUNDIR        Location of transient process files, default: /var/run/zm
    ZM_TMPDIR        Location of temporary files, default: /tmp/zm
 
nothing about ZM_SOCKDIR

ok 
(cmake) ZM_SOCKDIR = (ZM Option Path) PATH_SOCKS
(cmake) ZM_TMPDIR = (ZM option path) PATH_SWAP

christophey2k, 2015-01-26:
Sorry !!! sorry, my prob come from invalid apache directive with an typo error (a double slash... )in
"ScriptAlias /cgi-bin/nph-zms      /var/www/zoneminder.fqdn.fr/cgi-bin/nph-zms"
That cause my error with
"web_php 30815 ERR socket_sendto( /run/zm/zms-765587s.sock ) failed"

And for me i need to setup in cmake ZM_TMPDIR outside of /tmp directory ... for me in /zm
no test for ZM_SOCKDIR all are in the same /zm directory

No have Zoneminder V1.28 that works under my gentoo systemd
libvlc not work and ffmpeg give me a blue image but i go to search for that...

closed 2015-01-26 by christophey2k