ZoneMinder/zoneminder #683: The monitorprobe.php code does not work in FreeBSD

cspenceiv, 2015-01-22:
While looking into some issues regarding a <a href="https://github.com/ZoneMinder/ZoneMinder/pull/682">pull request</a>, I loaded up FreeBSD 10.1 to test a couple things with monitorprobe.php.

First of all, monitorprobe.php as written will not work with FreeBSD since it has no concept of <code>type -p</code>. Running <code>type -p arp</code> will result in a return value of 127. This will trigger the if statement to fall into the Fatal trap on line 295.

Second, the $output will contain <code>arp is /usr/bin/arp</code> since it does not understand the -p argument. This will cause the next system command to fail since it will send <code>arp is /usr/bin/arp -a</code>.

knnniggett, 2015-01-29:
I plan to push a change to pr #682 which will make this feature work for any distro.  I just need to make time to do it.

knnniggett, 2015-02-21:
Fixed in #709

closed 2015-02-21 by knnniggett