ZoneMinder/zoneminder #684: short PHP open tags paragraph on zoneminder.com Troubleshooting page

troywill, 2015-01-24:
Could the paragraph on the short PHP open tags on the Troubleshooting page be removed? (http://www.zoneminder.com/wiki/index.php/Troubleshooting.) (Since zoneminder no longer uses short open tags.)

troywill, 2015-01-24:
Thanks for the update. I applied it.

knnniggett, 2015-01-24:
We have not yet released a version of zoneminder that no longer requires short tags.
Note that anyone can update the WiKi.  Typically other users in the forum have been managing it.

troywill, 2015-01-24:
Sorry. I read https://github.com/ZoneMinder/ZoneMinder/pull/620 and presumed (incorrectly) that applied to current version.

knnniggett, 2015-06-17:
Closing.  Feel free to reopen this if you notice any short php tags in the next release.

closed 2015-06-17 by knnniggett