ZoneMinder/zoneminder #685: RTSP fails when hostname is used for source

dgilmour77, 2015-01-28:
ZM console allows an RTSP source URL to be entered as a host name, and will send the host name in the URL when performing an RTSP DESCRIBE.  However, when the source provides a control url, it may do so with an IP address, not a hostname.  This causes the logic added in the following fix:

https://github.com/ZoneMinder/ZoneMinder/commit/e764938a0d652a0e24db0fa6dde416b583a281df

to fail, generating a 404 error in response to a SETUP.

This is shown in the following zmc log file based on 1.28+.  Close to the end of this log excerpt, it is clear that the URL passed along with the SETUP is malformed because the fix above did not anticipate a hostname/IP mismatch.
======
.
01/27/15 19:01:40.085098 zmc_m1[3722].DB2-/home/david/zoneminder/src/zm_rtsp.cpp/74 [Received RTSP response: RTSP/1.0 200 OK
Server: VLC/2.0.3
Content-Length: 0
Cseq: 1
Public: DESCRIBE,SETUP,TEARDOWN,PLAY,PAUSE,GET_PARAMETER

 (124 bytes)]
01/27/15 19:01:40.085215 zmc_m1[3722].DB2-/home/david/zoneminder/src/zm_rtsp.cpp/48 [Sending RTSP message: DESCRIBE rtsp://kam.gilmour.net:8554/cam.sdp RTSP/1.0
User-Agent: ZoneMinder/1.28.0
CSeq: 2

]
01/27/15 19:01:41.086964 zmc_m1[3722].DB2-/home/david/zoneminder/src/zm_rtsp.cpp/74 [Received RTSP response: RTSP/1.0 200 OK
Server: VLC/2.0.3
Date: Wed, 28 Jan 2015 03:01:45 GMT
Content-Type: application/sdp
Content-Base: rtsp://192.168.1.54:8554/cam.sdp
Content-Length: 355
Cache-Control: no-cache
Cseq: 2

v=0
o=- 15596756719339326206 15596756719339326206 IN IP4 kam
s=Unnamed
i=N/A
c=IN IP4 0.0.0.0
t=0 0
a=tool:vlc 2.0.3
a=recvonly
a=type:broadcast
a=charset:UTF-8
a=control:rtsp://192.168.1.54:8554/cam.sdp
m=video 0 RTP/AVP 96
b=RR:0
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1
a=control:rtsp://192.168.1.54:8554/cam.sdp/trackID=0
 (564 bytes)]
01/27/15 19:01:41.087106 zmc_m1[3722].DB1-/home/david/zoneminder/src/zm_rtsp.cpp/372 [Processing SDP 'v=0
o=- 15596756719339326206 15596756719339326206 IN IP4 kam
s=Unnamed
i=N/A
c=IN IP4 0.0.0.0
t=0 0
a=tool:vlc 2.0.3
a=recvonly
a=type:broadcast
a=charset:UTF-8
a=control:rtsp://192.168.1.54:8554/cam.sdp
m=video 0 RTP/AVP 96
b=RR:0
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1
a=control:rtsp://192.168.1.54:8554/cam.sdp/trackID=0
']
01/27/15 19:01:41.087146 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'v=0']
01/27/15 19:01:41.087164 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'o=- 15596756719339326206 15596756719339326206 IN IP4 kam']
01/27/15 19:01:41.087184 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 's=Unnamed']
01/27/15 19:01:41.087253 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'i=N/A']
01/27/15 19:01:41.087265 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'c=IN IP4 0.0.0.0']
01/27/15 19:01:41.087283 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 't=0 0']
01/27/15 19:01:41.087292 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'a=tool:vlc 2.0.3']
01/27/15 19:01:41.087305 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/321 [Ignoring general SDP attribute 'tool:vlc 2.0.3']
01/27/15 19:01:41.087314 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'a=recvonly']
01/27/15 19:01:41.087324 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/321 [Ignoring general SDP attribute 'recvonly']
01/27/15 19:01:41.087333 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'a=type:broadcast']
01/27/15 19:01:41.087343 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/321 [Ignoring general SDP attribute 'type:broadcast']
01/27/15 19:01:41.087351 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'a=charset:UTF-8']
01/27/15 19:01:41.087362 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/321 [Ignoring general SDP attribute 'charset:UTF-8']
01/27/15 19:01:41.087370 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'a=control:rtsp://192.168.1.54:8554/cam.sdp']
01/27/15 19:01:41.087381 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/321 [Ignoring general SDP attribute 'control:rtsp://192.168.1.54:8554/cam.sdp']
01/27/15 19:01:41.087390 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'm=video 0 RTP/AVP 96']
01/27/15 19:01:41.087408 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'b=RR:0']
01/27/15 19:01:41.087420 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'a=rtpmap:96 H264/90000']
01/27/15 19:01:41.087438 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'a=fmtp:96 packetization-mode=1']
01/27/15 19:01:41.087457 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/301 [Ignoring SDP fmtp attribute 'packetization-mode' for media 'video']
01/27/15 19:01:41.087468 zmc_m1[3722].DB3-/home/david/zoneminder/src/zm_sdp.cpp/168 [Processing SDP line 'a=control:rtsp://192.168.1.54:8554/cam.sdp/trackID=0']
01/27/15 19:01:41.088258 zmc_m1[3722].DB1-/home/david/zoneminder/src/zm_sdp.cpp/382 [Looking for codec for video payload type 96 / H264]
01/27/15 19:01:41.088274 zmc_m1[3722].DB1-/home/david/zoneminder/src/zm_sdp.cpp/422 [Got dynamic payload type 96, H264]
01/27/15 19:01:41.088310 zmc_m1[3722].DB2-/home/david/zoneminder/src/zm_rtsp.cpp/48 [Sending RTSP message: SETUP rtsp://kam.gilmour.net:8554/cam.sdp/rtsp://192.168.1.54:8554/cam.sdp/trackID=0 RTSP/1.0
Transport: RTP/AVP/TCP;unicast
User-Agent: ZoneMinder/1.28.0
CSeq: 3

]
01/27/15 19:01:41.123788 zmc_m1[3722].DB2-/home/david/zoneminder/src/zm_rtsp.cpp/74 [Received RTSP response: RTSP/1.0 404 Not found
Content-Length: 362

connortechnology, 2015-01-28:
Interesting... good work.  

A fix should be fairly easy. I'll look into it today.

connortechnology, 2015-01-28:
Can we not instead of comparing the two urls... simply look at the beginning for something like a protocol string?  Or parse for '://' or something like that?  opinions?

knnniggett, 2015-01-28:
If I am reading things right, mURL will contain just the hostname or ip.  It doesn't contain the protocol.

I was thinking we check if murl is a hostname.  Do this around line 402 where trackurl is being defined. If murl is not an ip,then do a dns lokoup and set trackurl to its corresponding ip.  If is is not, then we set trackurl = murl.  
https://github.com/ZoneMinder/ZoneMinder/blob/e764938a0d652a0e24db0fa6dde416b583a281df/src/zm_rtsp.cpp#L402

dgilmour77, 2015-01-29:
Thanks, sorry I wasn't able to make and test a fix myself

knnniggett, 2015-01-29:
Well, I don't have enough time either.  I'll get to it it when I can, but once I get back to working on zoneminder (soon) I have other priorities to work on first.

I assume you are just manually created a DNS A record to do your name resolution?  Check your camera to see if you tell it what its dns name should be.  It's a stretch, but maybe that might be enough to get the camera to respond to a DESCRIBE with the name of the device, rather than the ip.  This would just be a workaround until someone fixes the source code.

dgilmour77, 2015-01-29:
It's just a Raspberry Pi being used as a test camera, with v4l2 and vlc generating the stream, so fortunately I have full control over the way DESCRIBE works.  Still trying to get mjpeg streaming from it working using v4l2 drivers and vlc without much luck, but my "real" cameras will arrive soon.  I can always use IP addresses for all sources and give static IPs or DHCP reservations to the cameras if it comes to that.  Thanks for a great app.

dgilmour77, 2015-01-29:
Reopened.

knnniggett, 2015-01-29:
That close button sure is easy to hit... done the same thing myself.

knnniggett, 2015-01-30:
So I remember what I was thinking after I drive home for the weekend, these are my own scratch notes, which I plan to insert near  line 370.
```c++
    std::string Header = response.substr( 0,sdpStart );
    Debug( 1, "Processing DESCRIBE response header '%s'", Header.c_str() );

    lines = split( Header, "\r\n" );
    for ( size_t i = 0; i < lines.size(); i++ )
    {
        if ( ( lines[i].size() > 13 ) && ( lines[i].substr( 0, 13 ) == "Content-Base:" ) )
	{
		Debug( 1, "Recieved new Content-Base in DESCRIBE reponse header. Updating device Url to: '%s'", Header.c_str() );
		mUrl = trimSpaces( lines[i].substr( 13 ) );
	}
    }
```

knnniggett, 2015-01-31:
pr #687 fixes this.  Please test to make sure it works correctly. 
Set your debug to level 5, and post your log.  Please do this even if everything appears to work correctly.

dgilmour77, 2015-01-31:
Sorry, was unable to successfully test because of a problem in my environment - sent you mail with details.

knnniggett, 2015-02-21:
@dgilmour77 Did you say you were running Debian Jessie?
If we merged this to master, perhaps @connortechnology could build a deb for you to test with.

dgilmour77, 2015-02-23:
Andrew, yes, I finally got it running with Jessie and it works great, but I couldn’t get the .deb to work – ended up editing the config manually.  So if you merge to master I can just update the complete set of source files and test.

knnniggett, 2015-02-23:
Glad this fixed your camera. We went ahead and merged it over the weekend so it is now in master.

dgilmour77, 2015-02-23:
Thanks for the help.  It’s really a nice system, now that I have it working.

connortechnology, 2015-06-17:
I've re-opened this because our fix has caused other problems, and I think we need to at least discuss the remaining issues.  

The case that this breaks is when the camera is behind a nat firewall.  I was trying to help someone, and so they forwarded the relevant port and gave me their external ip.  Of course the camera had an internal ip 192.168.whatever, so in it's describe it used the internal ip instead of the external and so it didn't work.  Now.. udp wasn't going to work here anyways, and after switch to tcp it worked just fine. 

So... maybe this needs to be a low-level advanced setting... or maybe we just need to add some text saying if the camera is behind a firewall you are going to have to use the tcp version.

knnniggett, 2015-06-17:
Are you there is a firewall between the zoneminder server and camera or are you saying there is a firewall between the client browser and the camera?

connortechnology, 2015-06-17:
between zm and the camera.  In this case, the zm server is mine, because I am trying to diagnose someone's problem.  

That sounds like a crazy case, but people are trying to use zm to monitor cameras at various locations, accessing them through firewalls.

This fix breaks the firewall case.

knnniggett, 2017-05-18:
This was fixed in #1028

closed 2017-05-18 by knnniggett