ZoneMinder/zoneminder #686: Streaming an Event as MPEG fails.

kylejohnson, 2015-01-28:
In my example, I'm trying to open this URL: `http://zmdev01/cgi-bin/nph-zms?source=event&mode=mpeg&format=webm&event=219&frame=1`, and it produces these errors:
```
Jan 28 17:05:25 zmdev01 zms[26582]: ERR [Got signal 11 (Segmentation fault), crashing]
Jan 28 17:05:25 zmdev01 zms[26582]: ERR [Signal address is 0x8, from 0x7f0aae233a09]
Jan 28 17:05:25 zmdev01 zms[26582]: ERR [Backtrace 0: /usr/lib/cgi-bin/nph-zms(_Z14zm_die_handleriP9siginfo_tPv+0x273) [0x48d668]]
Jan 28 17:05:25 zmdev01 zms[26582]: ERR [Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0xfc90) [0x7f0ab0c4dc90]]
Jan 28 17:05:25 zmdev01 zms[26582]: ERR [Backtrace 2: /lib/x86_64-linux-gnu/libc.so.6(+0x98a09) [0x7f0aae233a09]]
Jan 28 17:05:25 zmdev01 zms[26582]: ERR [Backtrace 3: /usr/lib/cgi-bin/nph-zms(_ZN11VideoStream11EncodeFrameEPKhibj+0x180) [0x46f7c8]]
Jan 28 17:05:25 zmdev01 zms[26582]: ERR [Backtrace 4: /usr/lib/cgi-bin/nph-zms(_ZN11EventStream9sendFrameEi+0x28e) [0x442c04]]
Jan 28 17:05:25 zmdev01 zms[26582]: ERR [Backtrace 5: /usr/lib/cgi-bin/nph-zms(_ZN11EventStream9runStreamEv+0x52a) [0x443828]]
Jan 28 17:05:25 zmdev01 zms[26582]: ERR [Backtrace 6: /usr/lib/cgi-bin/nph-zms(main+0xf1b) [0x433f60]]
Jan 28 17:05:25 zmdev01 zms[26582]: ERR [Backtrace 7: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f0aae1bcec5]]
Jan 28 17:05:25 zmdev01 zms[26582]: ERR [Backtrace 8: /usr/lib/cgi-bin/nph-zms() [0x432ea9]]
Jan 28 17:05:25 zmdev01 zms[26582]: INF [Backtrace complete, please execute the following command for more information]
Jan 28 17:05:25 zmdev01 zms[26582]: INF [addr2line -e /usr/lib/cgi-bin/nph-zms 0x48d668 0x7f0ab0c4dc90 0x7f0aae233a09 0x46f7c8 0x442c04 0x443828 0x433f60 0x7f0aae1bcec5 0x432ea9]

```

Running `addr2line`, I get this:
```
^Ckjohnson@zmdev01:~/ZoneMinder/src$ addr2line -e /usr/lib/cgi-bin/nph-zms 0x48d668 0x7f0ab0c4dc90 0x7f0aae233a09 0x46f7c8 0x442c04 0x443828 0x433f60 0x7f0aae1bcec5 0x432ea9
/home/kjohnson/ZoneMinder/src/zm_signal.cpp:89
??:0
??:0
/home/kjohnson/ZoneMinder/src/zm_mpeg.cpp:569
/home/kjohnson/ZoneMinder/src/zm_event.cpp:1227 (discriminator 2)
/home/kjohnson/ZoneMinder/src/zm_event.cpp:1449
/home/kjohnson/ZoneMinder/src/zms.cpp:326 (discriminator 2)
??:0
??:?
``1

Results are the same regardless of which `format` I specify.

kylejohnson, 2015-01-28:
Of note is that I am able to stream a Monitor as MPEG, in both webm and mp4 formats.

kylejohnson, 2015-01-29:
The event successfully streams via MPEG and webm / mp4 if I enable debugging, level 6 (only level I tried), for components `_zms|_nph-zms`.

As soon as I turn off debugging, the stream fails immediately.

SteveGilvarry, 2015-01-29:
Can you run zms under gdb? I did that with zmc and zma at first but now I just debug in Netbeans, I will see if I can debug mine, but my Dev machine is currently setup for h264 development. I will see if I can still record jpeg, wrote it so I could...

kylejohnson, 2015-01-29:
@SteveGilvarry I'm not sure how to call zms under gdb, because zms is called via CGI.  So I'd need to figure out how to have CGI call programs through gdb.  I'll see what I can find.

SteveGilvarry, 2015-01-29:
@kylejohnson gdb ./zms from CGI directory. Then just need to emulate the query string options are command line args

connortechnology, 2015-01-29:
I believe I have this fixed.

kylejohnson, 2015-01-29:
Commit 9d636ad12829ffc86f6cf916ed21ea961b9c894a seems to have fixed this issue.  Thank you @connortechnology

closed 2015-01-29 by kylejohnson