ZoneMinder/zoneminder #688: Multiple issues..

TheHammer101, 2015-02-01:
Hi gents, first off since I am completely new to Git, and ZM. I gotta give you huge props for the amazing work you're doing!

That being said I would love it if someone could help me with a few issues as I am trying to set up a camera recording server for the company I work for with about 190 remote Panasonic IP cameras (mostly very low resolution, standard quality, .5 frame rate). I've got the cameras added and accessible with little to no problem.

I am running:
Ubuntu Server 14.04.01 with LAMP, OpenSSH, and shared memories to shmall = 1572864/shmmax = 6442450944
Xeon proc
12GB of RAM
Ubuntu sofware RAID0 boot disks (temporarily to get set up with before I move to RAID1)
a 3x1TB Seagate ZFS ZRAID1 array setup for the recordings mounted to /recording (will be expanding this size later)

I have everything set up and working on a small scale, but I have absolutely no idea how to fix the issues I'm seeing right now! If you would, read on, and let me know how I can help you help me. lol

Issue#1: When viewing recorded events of Panasonic BL-C???A cameras which are recording JPG streams of 320x240 with a frame rate of 0.5 frames per second the WebGUI playback controls are very sluggish to respond. Even with only 4 monitors added! No errors in the log file I can see.
---------------------------------------------------------------------------------------------

Issues#2: When viewing recorded events of Panasonic BL-C???A cameras which are recording JPG streams of 320x240 with a frame rate of 0.5 frames per second the fast forward button merely changes the number of playback speed, but does not increase the actual playback speed. I do not see any errors in the log for this.

---------------------------------------------------------------------------------------------

Issue#3: When viewing the "Montage" I can only see 6 cameras at one time (20 monitors are set to Monitor, Enabled), and the rest show white space similar to an image that's failed to load in a web page.
[image]
When clicking on the missing image it brings up the single camera live view, but the page doesn't load! Any other pages from ZM do not load as well! When I close the Montage page the single camera live view I brought up from the Montage page, or any other pages I've tried to open loads up with no problem. *shrugs* It throws the following errors:

2015-01-31 19:15:37.401302	zms	4120	INF	Got signal 15 (Terminated), exiting	zm_signal.cpp	40
2015-01-31 19:15:36.376705	zms	4121	INF	Got signal 15 (Terminated), exiting	zm_signal.cpp	40
2015-01-31 19:15:36.132085	zmc_m2	1869	WAR	Unable to capture image, retrying	zm_remote_camera_http.cpp	1105
2015-01-31 19:15:36.020956	zmc_m2	1869	ERR	Connection dropped by remote end	zm_remote_camera_http.cpp	614
2015-01-31 19:15:35.922301	zmc_m2	1869	WAR	Select timed out	zm_remote_camera_http.cpp	156
2015-01-31 19:15:35.614037	zms	4119	INF	Got signal 15 (Terminated), exiting	zm_signal.cpp	40
2015-01-31 19:15:35.518923	zms	4122	INF	Got signal 15 (Terminated), exiting	zm_signal.cpp	40
2015-01-31 19:15:35.436297	zms	4123	INF	Got signal 15 (Terminated), exiting	zm_signal.cpp	40
2015-01-31 19:15:35.422785	zms	4124	INF	Got signal 15 (Terminated), exiting	zm_signal.cpp	40
2015-01-31 19:15:33.932024	zmc_m52	1976	WAR	Unable to capture image, retrying	zm_remote_camera_http.cpp	1105
2015-01-31 19:15:33.847760	zmc_m52	1976	ERR	Connection dropped by remote end	zm_remote_camera_http.cpp	982
2015-01-31 19:15:33.379451	zmc_m52	1976	WAR	Select timed out	zm_remote_camera_http.cpp	156
2015-01-31 19:15:05.864305	zmc_m42	1933	WAR	Unable to capture image, retrying	zm_remote_camera_http.cpp	1105
2015-01-31 19:15:05.792190	zmc_m42	1933	ERR	Connection dropped by remote end	zm_remote_camera_http.cpp	982
2015-01-31 19:15:05.718104	zmc_m42	1933	WAR	Select timed out	zm_remote_camera_http.cpp	156
2015-01-31 19:15:01.929343	zms	4120	WAR	Shared data not initialised by capture daemon, some query functions may not be available or produce invalid results	zm_monitor.cpp	424
2015-01-31 19:15:01.929323	zms	4122	WAR	Shared data not initialised by capture daemon, some query functions may not be available or produce invalid results	zm_monitor.cpp	424
2015-01-31 19:15:01.929303	zms	4124	WAR	Shared data not initialised by capture daemon, some query functions may not be available or produce invalid results	zm_monitor.cpp	424
2015-01-31 19:15:01.929116	zms	4119	WAR	Shared data not initialised by capture daemon, some query functions may not be available or produce invalid results	zm_monitor.cpp	424
2015-01-31 19:15:01.928913	zms	4123	WAR	Shared data not initialised by capture daemon, some query functions may not be available or produce invalid results	zm_monitor.cpp	424
2015-01-31 19:15:01.856301	zms	4121	WAR	Shared data not initialised by capture daemon, some query functions may not be available or produce invalid results	zm_monitor.cpp	424
2015-01-31 19:15:01.786379	zms	4120	INF	Authenticated user 'Daniel'	zm_user.cpp	241
2015-01-31 19:15:01.785148	zms	4124	INF	Authenticated user 'Daniel'	zm_user.cpp	241
2015-01-31 19:15:01.784487	zms	4122	INF	Authenticated user 'Daniel'	zm_user.cpp	241
2015-01-31 19:15:01.783807	zms	4119	INF	Authenticated user 'Daniel'	zm_user.cpp	241
2015-01-31 19:15:01.783386	zms	4123	INF	Authenticated user 'Daniel'	zm_user.cpp	241
2015-01-31 19:15:01.783219	zms	4121	INF	Authenticated user 'Daniel'	zm_user.cpp	241

TheHammer101, 2015-02-01:
I suppose that I'm not sure on some of these issues where to look to fix them myself, and I am unsure of what to provide to help others help me.

newburns, 2015-02-01:
Answer to number 3

http://www.zoneminder.com/wiki/index.php/FAQ#I_have_several_monitors_configured_but_when_I_load_the_Montage_view_in_FireFox_why_can_I_only_see_two.3F_or.2C_Why_don.27t_all_my_cameras_display_when_I_use_the_Montage_view_in_FireFox.3F

TheHammer101, 2015-02-01:
I see, interesting. I am actually using WaterFox (slim difference, but one of those settings was missing) I get a never finishing loading page in Chrome, and IE won't even load the single monitor views.. I have the Cambozola installed without JRE though, so I'm sure that's suspect. Otherwise, "Issue 3" is resolved, thank you for your prompt help!

TheHammer101, 2015-02-02:
Maybe streaming an MPG would be better than viewing the JPG style.. but when I do, I get so many errors I don't know how to shake a stick at it.

TheHammer101, 2015-02-04:
When I try to use MPEG streaming I get the following errors:

2015-02-04 12:25:46.127409	web_js	30099	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused - checkStreamForErrors()	?view=watch	
2015-02-04 12:25:45.890630	web_php	30272	ERR	socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused	includes/functions.php	2330
2015-02-04 12:25:44.262808	web_js	30099	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused - checkStreamForErrors()	?view=watch	
2015-02-04 12:25:44.052625	web_php	30272	ERR	socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused	includes/functions.php	2330
2015-02-04 12:25:42.708787	web_js	30099	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused - checkStreamForErrors()	?view=watch	
2015-02-04 12:25:42.616857	web_php	30099	ERR	socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused	includes/functions.php	2330
2015-02-04 12:25:42.250030	web_js	30272	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused - checkStreamForErrors()	?view=watch	
2015-02-04 12:25:41.536528	web_php	30272	ERR	socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused	includes/functions.php	2330
2015-02-04 12:25:40.535310	web_js	30272	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused - checkStreamForErrors()	?view=watch	
2015-02-04 12:25:40.455619	web_php	30272	ERR	socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused	includes/functions.php	2330
2015-02-04 12:25:39.454705	web_js	30272	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused - checkStreamForErrors()	?view=watch	
2015-02-04 12:25:39.371994	web_php	30272	ERR	socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused	includes/functions.php	2330
2015-02-04 12:25:38.370299	web_js	30272	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused - checkStreamForErrors()	?view=watch	
2015-02-04 12:25:38.316044	web_php	30272	ERR	socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused	includes/functions.php	2330
2015-02-04 12:25:37.312817	web_js	30272	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused - checkStreamForErrors()	?view=watch	
2015-02-04 12:25:37.258771	web_php	30272	ERR	socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused	includes/functions.php	2330
2015-02-04 12:25:36.261223	web_js	30272	ERR	getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused - checkStreamForErrors()	?view=watch	
2015-02-04 12:25:36.149055	web_php	30272	ERR	socket_sendto( /var/run/zm/zms-270724s.sock ) failed: Connection refused	includes/functions.php	2330
2015-02-04 12:25:35.167801	zms	30274	FAT	Could not open codec. Error code -1 "Operation not permitted"	zm_mpeg.cpp	320

I would assume that this has something to do with the change to avcon from ffmpeg, but even after installing ffmpeg I have nearly identical errors.

SteveGilvarry, 2015-06-17:
@TheHammer101 I am going to close this issue out, if any of the issues remain and you believe they are bugs please raise them as individual issues. More chance someone will look if the title describes a single issue. Usage questions please use zoneminder.com forums

closed 2015-06-17 by SteveGilvarry