ZoneMinder/zoneminder #689: Image disappears (flickers) in monitor and event playback

hans109h, 2015-02-02:
I've been trying to debug an issue that has come up recently with little luck.

Preface:  I've been using zoneminder for a few years now with out this issue.  I build from the Master branch.  I'm using Ubuntu 14.10.

Issue: When either watching a monitor or playing back an event everything will be working fine and then all the sudden there is a random flicker.  The entire image goes blank (like when an image isn't found in a web browser but a placeholder remains), then the image comes back.

When I look in either the /tmp/zm directory or the events directory, all images are there.  I haven't been able to find any related errors related to this.

Any suggestions on where to continue looking for the problem would be very helpful.

Hans

hans109h, 2015-02-04:
I've isolated this issue to only occur with Firefox, and it doesn't occur with Cambozla.

knnniggett, 2015-02-09:
The only time I have experienced anything similar to what you have described is when on a Windows client, running Firefox and Avast antivirus.  Disabling Avast antivirus stops the flickering.

hans109h, 2015-02-09:
I've done a little more investigating, and it's definitely unique to Firefox, as it doesn't do it in Chromium.  Right now I have framerates for low, medium, and high bandwidths set to 5, 10, 15 respectively (and the other associated settings are the defaults as well such as bitrate), the flick does not occur at the low and medium setting, but once it goes about 10 frames per second it begins to flicker.   I will play with it more when I have more time to determine which setting is triggering the flicker.

I did find this which seems to be related:

https://bugzilla.mozilla.org/show_bug.cgi?id=625012

It says that it has been resolved, however the issue I'm experiencing seems the same, so I filed this:

https://bugs.launchpad.net/ubuntu/+source/firefox/+bug/1418179

SteveGilvarry, 2015-06-17:
Closing as appears to be Firefox specific issue.

closed 2015-06-17 by SteveGilvarry