ZoneMinder/zoneminder #690: Foscam FI8918W Control does not work with FI8918W.pm but works with FI8908W.pm

jplx, 2015-02-03:
I purchased a Foscam camera FI8918W last year with firmware 11.37.2.56.
I used it on Fedora 20 with zoneminder 1.27.0-1 (rpm downloaded). This year, I am running on Fedora 21 with zoneminder 1.28.0-3 (compiled from the srpm source).
In both cases, the camera control does not work with the file FI8918W.pm but it does work with FI8908W.pm
Has anybody reported this issue? Is there any update or needed patch for FI8918W.pm?

knnniggett, 2015-02-03:
As you can see below, there is no FI8918W.pm script that is packaged with ZoneMinder:
https://github.com/ZoneMinder/ZoneMinder/tree/master/scripts/ZoneMinder/lib/ZoneMinder/Control

That means the script you are trying to use was written by a third party, and who knows what is wrong with it.  

If you want to fix it, try and compare the two scripts and see what has changed.  The control scripts for Foscam are usually very similar, if not virtually identical, to one another.

connortechnology, 2015-02-03:
We have a PR for a better FI8918W script... perhaps we should help get it merged. Maybe I should get a FI8918W, or maybe find one online to test against.... we need to get these things into our code and get rid of the old script references... maybe we should do some releases of old releases that include these scripts.

The problem is likely the ZM_VERSION thing.

jplx, 2015-02-04:
Thw FI8918W.pm I used came from:
https://github.com/arunh/FoscamFI8918W/blob/master/FoscamFI8918W.pm
It does not look very similar to the FI8908W.pm but I am not a Perl programmer.
I will try to debug with wireshark to hopefully identify what is different in the control packets.
It will take a couple of days...

connortechnology, 2015-02-04:
I can't seem to find a publicly accessible FI8918W... so...

knnniggett, 2015-02-05:
@jplx 
Since the author has posted his script on github, try creating an issue (in his repo, not ours) that asks the author to either fix the script to work with newer versions of zoneminer.  You could also ask him to create a pull quest in the zoneminder repo to get his script added to the source tree.  That way, it won't ever break again if we make a design change to the way the scripts work.

jplx, 2015-02-05:
Thank you for the advice: I will work on it.
Now, here are some results from the packet analysis with wireshark.

With your FI8908W.pm file, I see a correct control packet:
244 GET /decoder_control.cgi?command=4&user=admin&pwd=   HTTP/1.1

With the FI8918W.pm file, wireshark does not see any control packet. A small window pops up in the feed window:
---
The page at localhost says:
Control response was status = undefined message = /usr/bin/zmcontrol.pl --autostop
--command=moveConRight --id=3=>
---
In the log window, the following errors are reported (in chronological order):
-----
FAT Can't connect: Connection refused      zmcontrol.pl   (no line number)
ERR /usr/bin/zmcontrol.pl --autostop --command=moveConRight --id=3=>     The file is:
     /usr/share/zoneminder/www/includes/functions.php        The line is 2330
----
The fatal error occurs first.

Is there a way to display the variables ($cmd) being worked on in the script FI8918W file? I tried to insert the following statement that I found in other modules:
Info($cmd);
but it did not do anything. Any hint? If I could follow what the script is doing, I should be able to identify the problem. A short description of what the script is doing would help too :-)

connortechnology, 2015-02-05:
Have you tried running it from the command line to see if there are any errors output?

jplx, 2015-02-06:
I launched zm with "zmpkg.pl" command line but did not get any message on the terminal.
I inserted some flags in the working FI8908.pm like: Error( "Flag1" );
When using camera control, i saw in the log that the code went through subroutines 'new' and 'open' once and then through 'printMsg' and 'sendCmd' for every command.
I inserted similar flags in the non-working FI8918W.pm file and found that the code did not go through any flag in that file. Then, in zmcontrol.pl, I increased the MAX_CONNECT_DELAY to 100 (as opposed to 10) but it did not make any difference.
The failure occurs at line 139 of zmcontrol.pl  It is: while (!connect( CLIENT, $saddr ))
For some reason the connection does not happen.
I will try to make Arun Horne aware as he seems to be the last person who modified the FI8918.pm file.

knnniggett, 2015-02-06:
No, that isn't what @connortechnology meant.

You need to run *zmcontrol.pl* from the command line using the appropriate parameters as shown in the log file you previously posted.

Also, to test your script for Perl errors do execute this:
`perl /path/to/your/script/FI8908W.pm`

If the code is correct, the script should almost immediately return with no output. If you get output then that means there is a problem with the scripts and the output will tell you where.

jplx, 2015-02-07:
The test for Perl errors came back empty: so, there is no syntax error. I ran it on the FoscamFI8918W.pm file, which is the one failing.
Following your suggestion, I ran the same exact control command from the log:
/usr/bin/zmcontrol.pl --autostop --command=moveConRight --id=3=>
It came back with an error:
-bash: syntax error near unexpected token `newline'
Then, I suspected the => at the end of the line and ran:
/usr/bin/zmcontrol.pl --autostop --command=moveConRight --id=3
This time, there was no error and the camera rotated.
To verify this further, I edited the FoscamFI8918W.pm file as follows:
In the sub sendCmd routine, I inserted a substitution statement to get rid of => in the $cmd variable:
my $cmd = shift;       # Current line 132
$cmd=~ s/=>//;         # Added to remove =>
my $result = undef;  # previous line 133, is now line 134
With this workaround, the control works and the log does not report any error.
However, I still do not know where the => is coming from. Is it a bug?
What do you suggest I do as next step? I tried to connect to Arun Horne on Linkedin but have not heard back yet.

garrettben, 2015-02-19:
I just checked one in that may work for you.

jplx, 2015-02-19:
Thank you for letting me know and I would be glad to test it. Can you let
me know how to get it?
What is the URL?

garrettben, 2015-02-19:
Just view the comment on github and click my username.

jplx, 2015-02-20:
No, it does not work

2015-02-19 18:46:41.110138web_php4399ERR/usr/bin/zmcontrol.pl
--command=moveStop --id=1=>/usr/share/zoneminder/www/includes/functions.php
23302015-02-19 18:46:41.074700zmcontrol10850FATCan't connect: Connection
refusedzmcontrol.pl2015-02-19 18:46:31.073670zmcontrol10853INFControl
server 1/FoscamFI8918W-BEN starting at 15/02/19 18:46:31zmcontrol.pl2015-02-19
18:46:31.040360zmcontrol10850INFStarting control server 1/FoscamFI8918W-BEN
zmcontrol.pl

Even with my workaround, located after line 113:
$cmd=~ s/=>//; # JP added
(To get rid of the => at the end of the line). Now, it does not appear to
have any effect...

The only file that works is the FI8908W.pm included in the zoneminder
package. After using it and going back to yours, it worked until I used the
preset home. It may be an initialization issue.

garrettben, 2015-02-20:
Sorry to hear that.  Good luck on your search.

arjunroychowdhury, 2015-03-23:
The 3rd party perl files have errors - I've fixed them to work for my foscams. I'll post the updates soon. Keep a watch at http://www.zoneminder.com/wiki/index.php/Various_Learnings_from_getting_Zomeminder_1.28.1_working_well_on_Ubuntu_Server_14.04 --> I'll update the PTZ part for both 9831 and 8918 - I have both

SteveGilvarry, 2015-06-17:
@pliablepixels did you end up with a PR to make Foscam 8918 work? If so let us know and we will close this out.

jplx, 2015-06-18:
No, I have not been able to test it because the camera has been moved to a
remote area that I have not visited for a while.

pliablepixels, 2015-06-19:
@SteveGilvarry just did a PR for both 9831 and 8918. Note that its very important for format ControlDevice in control
For HD cameras it is usr=xxx&pwd=yyyy (Not "user" but "usr")
For MJPEG it is user=xxxx&pwd=yyyy

knnniggett, 2015-08-15:
Fixed in #895 .
Closing.

closed 2015-08-15 by knnniggett