ZoneMinder/zoneminder #693: Release 1.28.1

knnniggett, 2015-02-05:
Creating an issue to track any bugs or issues or whatnot we run into. Lets make this a clean release!

### Documentation
- [ ] Wiki – Update Wiki documentation to reflect any changes to existing features
- [ ] Local – Update internal documentation to reflect any changes to existing features
- [ ] Wiki – Add new documentation to Wiki for any new features
- [ ] Local – Add new internal documentation for any new features

### Build
Tested with the following build method:
- [ ] AutoTools
- [ ] CMake

### Distro
These tests were performed on the following Linux distributions:
- [ ] Ubuntu LTS
- [ ] Ubuntu
- [ ] CentOS/Redhat
- [ ] Fedora
- [ ] Other

### Browser
These tests where performed using recent versions of the following browsers:
- [ ] FireFox
- [ ] Chromium
- [ ] Chrome
- [ ] Internet Explorer
- [ ] Safari
- [ ] Mobile (state browser)
- [ ] Other

### Startup
- [ ] Verify initscript can auto-start ZoneMinder during the boot process
- [ ] Verify initscript can manually stop, start, and restart ZoneMinder

### Filters
- [ ] Verify a filter can be manually executed
- [ ] Verify a background filter runs automatically

### Authentication
- [ ] None - Trivial
- [ ] Plain – Set authentication to plain and verify
- [ ] Hashed – Set authentication to hashed and verify
- [ ] Remote – Set authentication to remote and verify

### Streaming Methods
- [ ] Local – Verify successful stream from this camera type
- [ ] Http – Verify successful stream from this camera type
- [ ] Ffmpeg – Verify successful stream from this camera type
- [ ] Libcurl – Verify successful stream from this camera type
- [ ] Libvlc – Verify successful stream from this camera type

### Recording Functions
- [ ] Monitor - Verify this function works
- [ ] Modect - Verify this function works
- [ ] Record - Verify this function works
- [ ] Mocord - Verify this function works
- [ ] Nodect - Verify this function works

### PTZ Control
- [ ] Verify ptz control works

### Skins
- [ ] Classic – verify field alignments
- [ ] Flat – verify field alignments

### Log Files
- [ ] Inspect log files for anything unusual

### Upgrade
- [ ] zmupdate - Verify an upgrade for a recent minor release
- [ ] zmupdate – Verify an upgrade for a past major release

### Optional Features
- [ ] X10 - verify basic communication with an x10 device
- [ ] E-mail – verify events are sent through e-mail
- [ ] Upload – verify events are uploaded
- [ ] eyeZm – verify this works
- [ ] Cambozola – verify from Internet Explorer

### API
 - [ ] Verify that you can query monitors and events

knnniggett, 2015-02-05:
I have begun build & testing the next release.  
Making a special note to focus on testing native RTSP support.

knnniggett, 2015-02-07:
I have completed testing 1.28.1 and did not find any issues. As you can see from the list above, I did not test everything.  Most of the unchecked items, I am not able to test.

@connortechnology Have you built 1.28.1 or master recently with autotools?  I pretty much use cmake now for all my builds.  If you have built zoneminder recently with autotools and have found no issues then I will not test it on my end.

christophey2k, 2015-02-08:
Hello,
Under Gentoo (from fresh and empty dir and db) if not specified event directory with ZM_CONTENTDIR in cmake, there is not event directory (var/lib/zoneminder not exist) and impossible to add new monitor in zm - 
And when setting ZM_CONTENDIR the directory was not self create .
And when create manually directory and setting ZM_CONTENDIR that not work too
when i create a monitor, nothing appear in the list, the monitor was removed immediatly

my ZM_CONTENDIR is under another hard drive

After a restart the log say:

deleting	zmaudit.pl	
2015-02-08 03:54:15.315850	zmaudit	560	INF	Filesystem monitor link 'BEWAN-iCam200G-CUISINE' does not point to valid monitor directory	zmaudit.pl	
2015-02-08 03:54:15.314760	zmaudit	560	INF	deleting	zmaudit.pl	
2015-02-08 03:54:15.313910	zmaudit	560	INF	Filesystem monitor '0' does not exist in database	zmaudit.pl

christophey2k, 2015-02-09:
ZM create directory (verified) for the new monitor but not in db ?, it's SQL issue ?
MySQL V5.6.22

knnniggett, 2015-02-09:
@christophey2k Your posts are not relevant to the building and testing of zm 1.28.1.  Please do not hijack threads for your own purpose.

closed 2015-02-15 by knnniggett