ZoneMinder/zoneminder #694: zms is unreliable, and I don't want to support it.  Yay <video>?

kylejohnson, 2015-02-05:
# Overview
This issue is to lead the discussion on the pull request that I plan on following up with.  This PR will be part of the current `classic` skin, but will perhaps live in a separate branch for a while, until tested.  That branch will otherwise stay up-to-date with `master`, such that the only change is what I am proposing.

Please read, comment and argue, so that we can make the right decision.

What is driving this is:  With the new UI rewrite, I do not want to support the `zms` code that is involved in creating, playing back, pausing, stopping, fast forwarding, and rewinding.

As such, I want to use HTML5's `<video>` element for both live streams and existing events.

I was able to get `zms` to work well enough with both webm and h264 that I could use `zms` as the `src` attribute for the `<video>` tag.  Unfortunately, while `zms` would spit out an initial stream, it was terribly unreliable, and the stream looked terrible.  

What I propose is to cut `zms` out of the picture entirely.

# wat?
`ffserver`.  `ffserver` serves HTTP streams of various feeds.  That is, for a given feed (read: monitor), you can have multiple streams (read: webm, h264).  

Currently, I have two feeds - one for each of my cameras.
I then have a total of 4 streams, 2 for each camera - webm and h264.
Finally, I am using the `<video>` element and pointing its `src` at these streams:
```html
<video controls autoplay>
 <source type="video/webm" src="http://$myserver:8090/$monitorname.webm">
 <source type="video/h264" src="http://$myserver:8090/$monitorname.flv">
</video>
```

The ffserver.conf looks like this

```
HTTPPort 8090
MaxHTTPConnections 2000
MAXClients 100
MaxBandwidth 1000000
CustomLog /tmp/ffserver.log

<Feed front.ffm>
File /tmp/front.ffm
FileMaxSize 1G
Launch ffmpeg -r 10 -i 'http://kjohnson:sosafe@192.168.11.20/mjpg/video.mjpg?streamprofile=ZM' -codec copy -an
ACL ALLOW 127.0.0.1
</Feed>

<Feed back.ffm>
File /tmp/back.ffm
FileMaxSize 1G
Launch ffmpeg -r 10 -i 'rtsp://kjohnson:sosafe@192.168.11.21:554/axis-media/media.amp?videocodec=h264' -codec copy -an
ACL ALLOW 127.0.0.1
</Feed>

<Stream front.flv>
Feed front.ffm
Format flv
NoAudio

VideoCodec libx264
VideoSize 704x480
VideoFrameRate 10
VideoBitRate 256
VideoBufferSize 800
VideoGopSize 12

AVOptionVideo flags +global_header
PreRoll 0
StartSendOnKey
</Stream>

<Stream front.webm>
Feed front.ffm
Format webm
NoAudio

VideoCodec libvpx
VideoSize 704x480
VideoFrameRate 10
VideoBitRate 256
VideoBufferSize 800
VideoGopSize 1

AVOptionVideo flags +global_header
#AVOptionVideo cpu-used 0
AVOptionVideo qmin 1
AVOptionVideo qmax 31
#AVOptionVideo quality good
PreRoll 0
StartSendOnKey
</Stream>

<Stream back.flv>
Feed back.ffm
Format flv
NoAudio

VideoCodec libx264
VideoSize 1280x720
VideoFrameRate 10
VideoBitRate 256
VideoBufferSize 800
VideoGopSize 12

AVOptionVideo flags +global_header
PreRoll 0
StartSendOnKey
</Stream>

<Stream back.webm>
Feed back.ffm
Format webm
NoAudio

VideoCodec libvpx
VideoSize 1280x720
VideoFrameRate 10
VideoBitRate 256
VideoBufferSize 800
VideoGopSize 12

AVOptionVideo flags +global_header
#AVOptionVideo cpu-used 0
AVOptionVideo qmin 1
AVOptionVideo qmax 31
#AVOptionVideo quality good
PreRoll 0
StartSendOnKey
</Stream>

<Stream index.html>
Format status
</Stream>
```

# Architecture

I want to focus on live streaming initially.  What I propose for live streaming is like this:

When a monitor is saved, edited or deleted, a bunch of php happens:
```
kill ffserver

# Write the ffserver.conf file
for each $monitors as $monitor {
  # Write the <feed> section

  # Write the <stream webm> section
  # Write the <stream h264> section
}

ffserver &
```
Finally, modify the `watch.php` and `montage.php` files to use the `<video>` tag, along with the `streams` provided by `ffserver`

Juanicardinali, 2015-02-11:
Great idea ! .. If I want to help you, wich files i have to see to modify the code ? .. How Can I make me an idea of how the architecture of ZM works to understand it ?

knnniggett, 2015-08-10:
Is this superseded by what @SteveGilvarry is doing in the video branch?
@fixxxermet Can we close this out?

connortechnology, 2015-08-10:
No.  Nothing has been done to get rid of zms.

liucougar, 2015-11-19:
for h264 streaming cameras, the override_ffserver option of ffmpeg may be helpful:
http://superuser.com/questions/818736/server-video-stream-without-reencoding-on-linux

SteveGilvarry, 2017-03-18:
Known task to fix streaming

closed 2017-03-18 by SteveGilvarry