ZoneMinder/zoneminder #695: When editing a monitor, using RTSP, host cannot contain random punctuation

connortechnology, 2015-02-05:
However the password might have punctuation in it.

We need to be consistent on all the various methods, including the control tab.  

I think username/password should be separate fields, that way we can actually do useful input validation.

SteveGilvarry, 2017-03-18:
Use Username and Password fields already in the db and just do this.

connortechnology, 2018-09-30:
I hate you stale bot.

GlenCooper, 2019-11-04:
Just commenting because I've reached this point, but need sleep (sorry).  Hopefully this comment will keep the stale-bot at bay for a little longer.

parvit, 2022-07-17:
Hi all,
i'm working on this and i've just got the system working from source.
I'll let you know as i have a change ready.

parvit, 2022-07-22:
@connortechnology I've added the PR responding to your request.

parvit, 2022-07-28:
Hello, is there any feedback on this?

parvit, 2022-08-22:
@connortechnology 
Thanks for merging the PR, if we may close this ticket it would allow me to claim the bounty.

As a side note: I would like to work also on #355, but it's locked so i don't know if i can proceed, can we unlock that one? thanks.

closed 2022-08-22 by connortechnology

connortechnology, 2022-08-22:
I don't see anything locked about #355.  You might want to discuss your approach on it before digging in too far.  Would be great to have it done though.

I've modified your code a bit.  in php8 passing null to a lot of functions generates errors.  Also I'm not sure strlen is the most efficient way to determine if a string is empty or not.  I also took out the setting of auth_type in ffmpeg.  It shouldn't be necessary and is just generating errors in the logs.

parvit, 2022-08-22:
This is what i see from my side on #355
[image]

connortechnology, 2022-08-22:
Okay it is unlocked.

parvit, 2022-08-22:
It worked, thanks.