ZoneMinder/zoneminder #697: ZM's Develop actually

Juanicardinali, 2015-02-06:
Hi everyone ! I'm relatively new at ZoneMinder. Me and a partnet are trying to implement a video surveillance system for the University. Actually ZM is the only one who runs in Debian. We read a lot here, and in the wiki, and we learn that ZM not have the feature to stream in the same format that the camera send the video, instead of that, ZM do a transcoding to jpeg to found motion detection (please, if I'm wrong let mi know !) For our work, we need that ZM does de streaming with the same format that the camara send it. My question is, is this nearly possible ? If not, where can I found good documentation to understand the source code to try to do something ? Thanks for your time.

kylejohnson, 2015-02-11:
Hello @Juanicardinali.

You're right, ZoneMinder does not yet have that feature.  Work on that is being done in #452 and #39, among others.  In addition, I am working on implementing ffserver as a middle man to allow streaming of the native camera stream without transcoding it, see #694.

If you want to help, please help with the first two issue numbers above.

If you need something now, look into ffserver, however that doesn't currently work with ZoneMinder.

Juanicardinali, 2015-02-11:
Thanks forma answerd ! Today I will make me a time yo ser al that you say.
Maybe yo help un the issues i will need sole help to understend wich files
have to modify anda how the code work un the file. Sure later I will write
you again :) Have a nice day !

closed 2015-06-17 by SteveGilvarry