ZoneMinder/zoneminder #698: 2nd network port breaks 'Probe' in 1.28 on Ubuntu 14.04?

Southcross, 2015-02-07:
2015-02-07 09:45:16.699755	web_php	3681	FAT	Unable determine arp path, status is '127'	/usr/share/zoneminder/skins/classic/views/monitorprobe.php	295

I am still only in the setup/testing phase...
The only thing I did to the machine (that I am aware of to break it) is I set up a 2nd ethernet port (eth1) for a network dedicated to CCTV traffic.  I can Ping/send receive data/ARP -a through the 2nd network., the "probe" command just stopped working.  I was thinking I needed to add "-i eth1" to the .php script, but I see that the "%command -a" takes place after line 295 giving the error?

knnniggett, 2015-02-07:
Status 127 means that PHP can't find the arp tool on your machine.

Southcross, 2015-02-07:
I understand the origination of the error '127', prior to v1.28(?) arp needed to be called via its real path, this is (should) be fixed in 1.28 to use "type" to get the path., When I use the "type -p arp" command, as is used in the php, I get a valid response:

xxxx@lamp:~$ type -p arp
/usr/sbin/arp

knnniggett, 2015-02-07:
You got a valid response, when running under a normal user account.
You need to run that test as the web account user, and also verify the arp command can be executed by the web user account.

If you are handy with php, temporarily change the script so it sends the raw output to your browser.

I've got two nic's on my production server and the probe works just fine.

Southcross, 2015-02-08:
I was thinking the very same thing, that it may be permissions related, when I was doing the brakes on my car yesterday... but I do not have the experience with Linux and PHP to do so

Southcross, 2015-02-08:
I guess googling is my friend LOL:

ok you can "sudo -u www-data <command>" to run a command as a specific user, you just need your superuser sudo password

"sudo -u www-data arp -a"
ARP works just fine... 

"sudo -u www-data type -p arp"
sudo: type: command not found

-interesting side note, you can not "type -p" the type command?

connortechnology, 2015-02-08:
type is a bash bulitin, isn't it.  

Let me guess, www-data's shell is something like /bin/sh

which is maybe why we should use which instead of type.

knnniggett, 2015-02-08:
The "which" command had its own set of problems, which (pun not intended) caused us to switch to "type" several months ago.  

In any case, I am working with another user on pr #682 
Rather than try to autodetect the path, it creates an a new field under Options to let the user enter the full path to the arp tool of choice.

Naturally, it still won't work if PHP does not have permission to execute the command, but I've changed the error reporting to be a little better (hopefully).

Southcross, 2015-02-08:
thanks Connortechnology, the command "which" fixed my problem. 

If you find a solution to the "type" command not working, great, unless the "which" command breaks something else I can live with the band-aid.

closed 2015-02-08 by knnniggett

mxc, 2015-08-09:
In debian systems the www-data account's shell is /usr/sbin/nologin