ZoneMinder/zoneminder #712: Zone edit dots are gone

river100, 2015-02-19:
I did a build of vers 1.28.0 using the method on the main github page to build the deb file.
Ubuntu 12.04 64bit desktop. this was done on 02-18-2015.
 I did make one change to the source file "zm_event.cpp" before 
building to stop the fast event playback when using mpeg stream.
When trying to edit a zone the green field is there but the dots are gone. If you hold the mouse over a corner you can find where it should be because the point x /y lights up but still no dot.
I tried this with chrome, firefox, and internet explorer.

knnniggett, 2015-02-19:
@river100 Other than to check your log for errors, I unfortunately do not have a suggestion for you at the moment. However, I do want to say that I have read and continue to read your posts both here and in the forum.  We are not ignoring you.  However, I simply don't have a good answer to your posts at this time.  You are hitting on areas I personally am completely unfamiliar with.  I do re-read open issues, and what I found is that, sometimes much later, I'll think of something I hadn't thought of before.

Perhaps someone else can speak to this particular issue.

kylejohnson, 2015-02-19:
In Firefox, have a look at the Console tab of the Inspect Element window, and then refresh F5 the zone edit page.  Please let us know if there are any errors reported.

Perhaps now is a good time for me to submit a PR with my new Zone JavaScript code, using <canvas> tags and html5.

river100, 2015-02-19:
I get this 
GET http://192.168.1.27/zm/skins/classic//css/graphics/point-g.gif [HTTP/1.1 404 Not Found 2ms]

river100, 2015-02-19:
just looked and it's here,
/usr/share/zoneminder/skins/classic/graphics/point-g.gif

river100, 2015-02-19:
knnniggett  you read minds too ?

river100, 2015-02-19:
I was wondering what I must have done. no responses to my events play back post.
not sure if the fix, ( it wasn't my fix) will mess anything up but it did work.  problem is you have to build from source and edit the zm_event.cpp file before building.

connortechnology, 2015-02-19:
I know what the problem is.  I will have a fix shortly.

Please give more details on the event playback fix.

knnniggett, 2015-02-19:
He is referring to #676 
There is a simple patch that fixes it, but I don't know if it breaks anything else.

river100, 2015-02-19:
OK, I just did another build, this time on ubuntu server 14.04 LTS 32bit.
Not sure if the distro being different but the "dots" are there in zone edit page. 
BTW  why doesn't V4L1 show when using a capture card ?
didn't show on 12.04 LTS either, only V4L2

connortechnology, 2015-02-19:
If you did an update from master, it worked because I fixed the problem.

As to v4l1, v4l1 has been removed from linux for a long time now. You would have to build zm on a system old enough to still support it.

connortechnology, 2015-02-19:
wait a minute, I fixed this in master, not 1.28

Now I'm going to have to check 1.28.1.  Sigh.

connortechnology, 2015-02-19:
1.28.1 seems to work fine. I believe it was just master that was affected.

river100, 2015-02-19:
One was built on 12.04 LTS and I think I got 1.28.0 from it.
Today I used 14.04 LTS 32bit and the vers is 1.28.1.
I'll dbl check the vers when I get back.
It's working on 1.28.1

river100, 2015-02-20:
Ok, sorry took so long, very long day.
here are the versions built from source instructions on main github page

12.04.2 LTS 64bit desktop ZM index page shows 1.28.0, it had the zone point bug.
dpkg -s gives   1.28.0+1-trusty-SNAPSHOT2014121201

14.04.2 LTS server 32bit  ZM index page shows 1.28.1. it was done feb-19 around noon
and the zone points were there.
 dpkg -s gives  Version: 1.28.0+1-trusty-SNAPSHOT2015021201

river100, 2015-02-20:
just did another on the machine with 12.04.2 LTS 64bit and ZM index page is showing 1.28.1
the zone point bug is not there.
 I did have to edit the zm_event.cpp file to get the play back speed to be normal
using mpeg / asf streaming

river100, 2015-02-20:
Wanted to mention that I've been using a version built by a forum member 3 years ago.
ubuntu 11.10 64 with ZM 1.25 as I could never get the asf streaming to work on my own.
I used a clonzilla image and went from there. Tried once and a while but no cigar.
The work that y'all have done on Zoneminder is great. I can now get a totally up to date 
Zoneminder working in 20 minutes.  BIG improvement.
Now I can update my friends to the good stuff.

closed 2015-02-21 by river100