ZoneMinder/zoneminder #715: Flat skin stopped working

river100, 2015-02-20:
i have 2 versions, one Ubuntu 14.04.2 LTS 32bit W/ ZM VERS 1.28.1 done on 12/19.
The flat skin works on it.  the cookies show flat
the other I built today 02/20 Ubuntu 12.04.2 LTS W/ ZM VERS 1.28.1 built this morning.
flat skin doesn't work, cleared cookies, and the cookies still show classic instead of flat.

connortechnology, 2015-02-20:
Are you running our master branch, or the 1.28.1 release?

river100, 2015-02-20:
not sure, I used the instructions on the main zoneminder github page.

river100, 2015-02-20:
what would they give me ??

river100, 2015-02-20:
I just copied pasted from them

river100, 2015-02-20:
The instructions say they will give me an install from the master branch.

"A fresh build based on master branch running Ubuntu 1204 LTS. Will likely work for other versions as well."

however the vers reported on ZM index page says 1.28.1
hope that helps

river100, 2015-02-20:
I'll try one on 14.04.2 later. some of these I've done on bare metal servers, others in ESXi 5.0

knnniggett, 2015-02-20:
@river100 So you did a "git clone" per the instructions?
That means you have built from the master branch. You really are not running 1.28.1, but a beta that is newer than 1.28.1.  It matters because we are in the middle of trying to fix the issue you are experiencing.

connortechnology, 2015-02-20:
Go into options to the system tab, select classic for skin and flat for css.  Then go to the display tab and do the same.
flat is no longer actually a skin, but just a different set of CSS files.

river100, 2015-02-20:
yes i did do a git clone per instructions here they are as I did them

aptitude install -y apache2 mysql-server php5 php5-mysql build-essential libmysqlclient-dev libssl-dev libbz2-dev libpcre3-dev libdbi-perl libarchive-

zip-perl libdate-manip-perl libdevice-serialport-perl libmime-perl libpcre3 libwww-perl libdbd-mysql-perl libsys-mmap-perl yasm automake autoconf 

libjpeg8-dev libjpeg8 apache2-mpm-prefork libapache2-mod-php5 php5-cli libphp-serialization-perl libgnutls-dev libjpeg8-dev libavcodec-dev libavformat-dev 

libswscale-dev libavutil-dev libv4l-dev libtool ffmpeg libnetpbm10-dev libavdevice-dev libmime-lite-perl dh-autoreconf dpatch;

apt-get install -y git

git clone https://github.com/ZoneMinder/ZoneMinder.git zoneminder;

cd zoneminder;

ln -s distros/ubuntu1204 debian;

  **** NOTE:  the packages below were complained about when dpkg-checkbuilddeps was run *****

apt-get install -y quilt libvlccore-dev libvlc-dev libcurl4-gnutls-dev
apt-get install -y libcurl4-nss-dev
apt-get install -y libcurl4-openssl-dev libpolkit-gobject-1-dev
apt-get install -y libav-tools

dpkg-checkbuilddeps;
dpkg-buildpackage;
dpkg-reconfigure zoneminder;
Also had to link the apache config files

river100, 2015-02-20:
OK, I guess with all the vers I've tried I forgot that it has to be set in 2 spots.
Tried it and  Flat skin does work ! sorry about that.
Again VERY nice work guys. BTW left this line out of above
dpkg -i /home/user/zoneminder_1.28.0+1-trusty-SNAPSHOT2015021201_amd64.deb; apt-get install -f;

closed 2015-02-20 by river100

knnniggett, 2015-02-20:
@river100 This relates to a different issue I am working on, but I think you might be able to help.

Do the source files you are working with contain the (new) onvif subfolder?  If no, then you don't have to answer the following.

Did you at any time, run into an error that looked like this:
http://pastebin.com/pjveXkFh

Probably not, but I am looking to confirm that the latest master branch, which includes experimental onvif support, builds on Ubuntu, using the same build method you just previously described. 

We had a user report that it does not build on Debian 7 using the Debian version of the same method you followed.  Trying to isolate where the problem may be.

river100, 2015-02-20:
I'll check,  I have so many of them lol

river100, 2015-02-20:
I do have the folder and files n it. I did a find search of my term' logs and nothing with pastebin came up.

river100, 2015-02-20:
that was from the 32 bit 14.04 ubuntu

river100, 2015-02-20:
BTW I still get the arp errors on both onvif and monitor probe, but I fure y'all know about them.
none of it works on the ones I've done.

knnniggett, 2015-02-20:
@river100 onvif support is not compiled in by default so the onvif probe won't work.  I still need to add code to turn off the onvif link when it is not compiled in.  maybe this weekend.