ZoneMinder/zoneminder #724: Build with ZM_TARGET_DISTRO=el7 fails to start under systemd

clipo1979, 2015-02-22:
Fresh build from latest Git on CentOS 7, latest update.

cmake -DZM_TARGET_DISTRO=el7 .

all compiles and installs OK, update mysql (Mariadb) with the right info then upon starting systemd gives the following error:

Feb 22 15:11:18 localhost.localdomain systemd[1]: Starting ZoneMinder CCTV recording and security system...
Feb 22 15:11:18 localhost.localdomain zmpkg[2829]: INF [Command: start]
Feb 22 15:11:18 localhost.localdomain zmdc[2845]: INF [Server starting at 15/02/22 15:11:18]
Feb 22 15:11:20 localhost.localdomain zmdc[2845]: FAT [Can't bind: Address already in use]
Feb 22 15:11:28 localhost.localdomain zmdc[2841]: FAT [Can't connect: Permission denied]
Feb 22 15:11:28 localhost.localdomain zmpkg[2829]: ERR [Unable to run "/usr/local/bin/zmdc.pl startup", output is "Starting server"]
Feb 22 15:11:28 localhost.localdomain systemd[1]: zoneminder.service: control process exited, code=exited status=255
Feb 22 15:11:28 localhost.localdomain systemd[1]: Failed to start ZoneMinder CCTV recording and security system.
Feb 22 15:11:28 localhost.localdomain systemd[1]: Unit zoneminder.service entered failed state.

When building from the same source with no distro tag all starts ok and runs perfect.

Any ideas whats missing.

knnniggett, 2015-02-22:
This really is not a systemd problem.
Exactly how are you attempting to start zoneminder?  
Can't bind means it can't create a socket.
You need to verify sockdir exists, has the right permission, a previous instance isn't already use that (address already in use)....

clipo1979, 2015-02-23:
It's a fresh install on a clean system. I used the target distro option, on another clean install using no distro option but manually specifing options it works fine. 

I looked inside the CMakeCache and the distro defined options ie log dir have not been listed correctly but do seem to be used in the build. The file states the el7 options have been used.

Now you have suggested a possible area I will have a look and post back With more information.

When you run cmake with say target distro of elk dose all the defined paths get copied into the cache file.

clipo1979, 2015-02-23:
Forgot to say. I tried to start with the systemd script and also when that failed with zmpkg.

knnniggett, 2015-02-23:
Again, the problem is with your SOCKDIR.  Start looking there.

clipo1979, 2015-03-01:
knnniggett,

You were right the SOCK directory permissions and the file permissions were all set to root not apache,
I've also found out that the SWAP and TEMP directoy permissions were all wrong and set to root.

Changing the permissions to apache fixed the problem and ZM is working perfect.

I've looked in the cmake file to see if it was defined wrong but all looks ok to me, I'll try and do a little more digging but this is a little bit out of my depth.

#install(DIRECTORY run DESTINATION /var DIRECTORY_PERMISSIONS OWNER_WRITE OWNER_READ OWNER_EXECUTE GROUP_WRITE GROUP_READ GROUP_EXECUTE WORLD_WRITE WORLD_READ WORLD_EXECUTE)
install(DIRECTORY sock swap DESTINATION /var/lib/zoneminder DIRECTORY_PERMISSIONS OWNER_WRITE OWNER_READ OWNER_EXECUTE GROUP_READ GROUP_EXECUTE WORLD_READ WORLD_EXECUTE)
install(DIRECTORY zoneminder DESTINATION /var/log DIRECTORY_PERMISSIONS OWNER_WRITE OWNER_READ OWNER_EXECUTE GROUP_READ GROUP_EXECUTE WORLD_READ WORLD_EXECUTE)
install(DIRECTORY zoneminder DESTINATION /run DIRECTORY_PERMISSIONS OWNER_WRITE OWNER_READ OWNER_EXECUTE GROUP_READ GROUP_EXECUTE WORLD_READ WORLD_EXECUTE)
install(DIRECTORY zoneminder-upload DESTINATION /var/spool DIRECTORY_PERMISSIONS OWNER_WRITE OWNER_READ OWNER_EXECUTE GROUP_READ GROUP_EXECUTE WORLD_READ WORLD_EXECUTE)
install(DIRECTORY events images temp DESTINATION /var/lib/zoneminder DIRECTORY_PERMISSIONS OWNER_WRITE OWNER_READ OWNER_EXECUTE GROUP_READ GROUP_EXECUTE WORLD_READ WORLD_EXECUTE)

clipo1979, 2015-03-14:
knnniggett,

I could do with a few pointers, have spent the last few weeks reading up and playing with the code but am stuck with the directory owners when they are copied in the Redhat CMakeList.txt file.

When the
 
install(DIRECTORY sock swap DESTINATION /var/lib/zoneminder DIRECTORY_PERMISSIONS OWNER_WRITE OWNER_READ OWNER_EXECUTE GROUP_READ GROUP_EXECUTE WORLD_READ WORLD_EXECUTE)

command is called it moves the directories to the folder but leaves them with root owner not apache as it needs to be, any ideas where Im going wrong.

knnniggett, 2015-03-17:
Give me a little more time to set up a development environment for zoneminder on centos 7 at my dayjob.  Things are more difficult w/o support from the rpmfusion repo.

While you wait, I'm not sure if you know that it only takes a couple more steps to just roll your own rpm rather than build from source.  

perform the following under your home folder.  Do NOT do this as root!

- Set up your system for rpmbuild:
http://wiki.centos.org/HowTos/SetupRpmBuildEnvironment
- put the zoneminder source files under SOURCES
- put the zoneminder spec file into SPECS
- run "rpmbuild -ba zonemidner.el7.spec". 

You should get an rpm that will install the folders in question with the correct ownership.

knnniggett, 2015-03-18:
The short answer to your question about the permissions on the sock/swap/run folders is because it has to.  The config files under the distros subfolder are intended primarily for packaging zoneminder into rpm/deb, and because of the way the packaging scripts work, we have to keep the permissions to be that of the account performing the build.  The RPM spec file contains the information to modify the permissions of these folders during installation.

Obviously that does not help anyone who wants to build from source (I still encourage you to try and roll your own rpm.  Once you get the hang of it, you won't go back to building from source).  I need to work through a few things to determine a solution that doesn't break something else. stay tuned.

clipo1979, 2015-03-22:
knnniggett,
Thank you for you answers, I'll have a go at rolling my own RPM.

I was looking for a more automated way to install from source and the distro build script was brilliant with these few minor problems. I can see why it is that way now I have done a fair amount of reading up and your help.

Maybe we could put another script similar to the zmlinkcontent one that would correct the permissions if you wanted to install from source. I'll have a look into this when I get more free time. My day job is taking all my time at the moment so no time to learn and play.

I've used the zmrepo RPM to install on my production machine and all seem ok at the momnet.
The only issue I've come across with is that the default installer for Centos7 as with 6 makes the home directory the majority of the drive space and the install uses the system spaces that then fills up quick.

when I get around to it I'll reinstall and make the entire drive the root.

closed 2015-03-26 by clipo1979

knnniggett, 2015-03-26:
Note that I am still planning to make a change to address this and a related issue.

clipo1979, 2015-03-26:
The issue is now resolved.

The sock and other /var/lib directories had the wrong owners so zoneminder could not were to the folders.

Permissions should have been apache not root

The cause is because of the rpm build and could so sdolved with a script if needed.

clipo1979, 2015-03-26:
Knight,

Do you want this reopening?

What's is the other related problem and your possible permanent solution

knnniggett, 2015-04-01:
@clipo1979 
Check out #774 
This should remedy the issue you were experiencing.