ZoneMinder/zoneminder #726: Use ARM NEON instructions when built on ARM

connortechnology, 2015-02-22:
Just putting this here so that bounties can be posted to it.  

NEON is the ARM equivalent of SSE.  Thanks to @mastertheknife we have support for SSE, but as I work on getting ZM onto raspberry pi and other cameras which are normally ARM, it would be nice to have NEON support.  Also we will likely need every once of performance we can get for these architectures.

connortechnology, 2017-03-03:
@mastertheknife

knnniggett, 2017-04-28:
I have reassigned the $190 bounty source dollars I received from #432 to this issue.

@mastertheknife It's yours to claim once we merge #1823 
https://www.bountysource.com/issues/8808814-use-arm-neon-instructions-when-built-on-arm

knnniggett, 2017-05-18:
Implelmented in #1823

closed 2017-05-18 by knnniggett