ZoneMinder/zoneminder #727: Missing dependency for killall in rpm package

ewsizeit, 2015-02-23:
On centos 7 killall is not installed by default. It is contained in the package psmisc. ZoneMinder should depend on that to make sure killall is available.
/run/zoneminder/zm.pid fills up with error messages instead of only a pid if killall can not be found.

knnniggett, 2015-02-23:
Wow, nice catch.  Looks like all the redhat/fedora specfiles need that requirement added.
Just for the sake of knowing, what flavor of centos doesn't have that package by default?  Minimal install perhaps?  Something else?

knnniggett, 2015-02-23:
fixed in commit 111e75e8e91b5189b3f9b7a6580e8950f877472f

ewsizeit, 2015-02-23:
That was quick! Resolved within an hour.

We saw this on a centos 7 minimal install. It's on a VM server so no need for window system etc.

Minimal does have killall5, which is a System V version of killall. It doesn't have the same flags, so you really need psmisc.

I could imagine that on a full blown GUI machine it would be installed by some other package's dependency. Perhaps that's why no one has spotted it before.

closed 2015-02-23 by knnniggett