ZoneMinder/zoneminder #733: zmc can't set standard

skyglow, 2015-02-27:
Hi, it's been around 2 years since i'm using zoneminder, The issue i'm going to report might seem known but i've not found a real fix yet.

It happens with 2 cards i have out of 3.
The one that uses cx88xx as driver works fine.
stk1160 and tw68 [techwell 6816] (last one built driver by myself) do not.
All camera display correct video output with mplayer [ mplayer tv:// -tv driver=v4l2:input=0:width=640:height=480:device=/dev/video0 ]
After system reboot Conexant based one works fine zoneminder, the other 2 requires a strange practice to make them work.
For the techwell and Syntek one i need to continuosly restart zoneminder while having mplayer opened, till "I'm lucky" and they work perfectly until next reboot.
Cameras connected to the 3 cards are the same, video settings are done properly, device permission are set correct with apache in the video group.
While searching how to fix this issue i found someone saying that zoneminder is before accessing the 
card before setting it up.
The error continues any device format ,capture palette, colorspace, width and height i set.

Distribution: Ubuntu server 14.04 64bit lastest upgrades
Zoneminder: 1.28.1 from iconnor repository
Installation steps: the ones in zoneminder guild for my distribution

this is the weblog http://imgur.com/AU0huiE
card settings http://imgur.com/9j3UmAd
v2l-info http://pastebin.com/BEj9qNjK
zmu -q -d /dev/video0 -v       http://pastebin.com/H90ERGEr
mplayer output http://pastebin.com/ZPFgX6xT

Let me know if you need anyfurther information, in case needed i can also provide you an user access to the machine.

connortechnology, 2015-02-27:
Interesting. I don't see any reason off the top why it isn't working.

You will need to go into options -> logging and turn on debug logging at a level of 6 or so.  We will need the out put from that.

skyglow, 2015-02-27:
http://imgur.com/iAxrbIf,i58HLma#0
http://imgur.com/iAxrbIf,i58HLma#1

connortechnology, 2015-02-27:
please don't have anything else accessing it.  Those device or resource busy errors generally mean that someone else has it open.

skyglow, 2015-02-27:
nothing is accessing it.
i even checked with lsof nothing is accessing those devices

connortechnology, 2015-02-27:
Then I think you have two zmc processes for it, which should not happen. 

you have two pids for /dev/video0 9858 and 9841

skyglow, 2015-02-27:
That happens cause 9841 exits abnormally. So it later spawns 9858 that will exit again when trying to access the device

theccm, 2015-12-25:
I can confirm this bug with techwell 6804 Dvr card too since I moved to kernel > 4.x, the reason is before the tw68 module was not included in kernel source and since it has been rewritten almost entirely. 

So you will say this is a bug inside module code, but ffmpeg is working perfectly on command line using same libav version and same kernel version of course. So zoneminder must call ioctl making the device busy, before setting video format, ioctl VIDIOC_S_STD ?
I'm using 1.28.1 tagged version and ffmpeg 2.7.2 
I will try to update to master zoneminder branch and latest ffmpeg to test.

rob6502, 2016-01-19:
likewise, with Techwell devices that worked perfectly with the user-compiled Techwell driver on Mint 14, after upgrading to Mint 17.3 (with the new, built-in version of the Techwell driver), I got the same error as above in Zoneminder (in VLC, the same Techwell driver works OK) (kernel 3.18-3.19).

kingofkya, 2017-03-11:
Dusted off my old card.

03/11/17 00:30:00.232073 zmc_dvideo2[10835].DB1-zm_monitor.cpp/449 [Monitor Monitor-11 has no zones, adding one.]
03/11/17 00:30:00.232113 zmc_dvideo2[10835].DB1-zm_zone.cpp/51 [Initialised zone 0/All - 1 - 384x288 - Rgb:0000ff, CM:3, MnAT:15, MxAT:$
03/11/17 00:30:00.233254 zmc_dvideo2[10835].DB1-zm_monitor.cpp/459 [Monitor Monitor-11 has function 2]
03/11/17 00:30:00.233265 zmc_dvideo2[10835].DB1-zm_monitor.cpp/460 [Monitor Monitor-11 LBF = '%N - %d/%m/%y %H:%M:%S', LBX = 0, LBY = 0$
03/11/17 00:30:00.233270 zmc_dvideo2[10835].DB1-zm_monitor.cpp/461 [Monitor Monitor-11 IBC = 50, WUC = 25, pEC = 25, PEC = 25, EAF = 1,$
03/11/17 00:30:00.233878 zmc_dvideo2[10835].DB1-zm_zone.cpp/957 [Got 1 zones for monitor Monitor-11]
03/11/17 00:30:00.233906 zmc_dvideo2[10835].DB1-zm_zone.cpp/51 [Initialised zone 11/All - 1 - 384x288 - Rgb:0000ff, CM:3, MnAT:25, MxAT$
03/11/17 00:30:00.234967 zmc_dvideo2[10835].DB1-zm_monitor.cpp/2212 [Loaded monitor 11(Monitor-11), 1 zones]
03/11/17 00:30:00.234979 zmc_dvideo2[10835].INF-zmc.cpp/250 [Starting Capture version 1.30.0]
03/11/17 00:30:00.295512 zmc_dvideo2[10835].FAT-zm_local_camera.cpp/906 [Failed to set video standard 255: Device or resource busy]

Correct me if i am wrong. But is this error because zoneminder is missing the properly cased color pallets? I think the bt cards all list them in cap like RGB24 instead of rgb24.

ffmpeg -f v4l2 -list_formats all -i /dev/video0 
[video4linux2,v4l2 @ 0x87b360] Raw       :    rgb555le : 16-bit A/XRGB 1-5-5-5 :
[video4linux2,v4l2 @ 0x87b360] Raw       :    rgb555be : 16-bit A/XRGB 1-5-5-5 BE :
[video4linux2,v4l2 @ 0x87b360] Raw       :    rgb565le :     16-bit RGB 5-6-5 :
[video4linux2,v4l2 @ 0x87b360] Raw       :    rgb565be :  16-bit RGB 5-6-5 BE :
[video4linux2,v4l2 @ 0x87b360] Raw       :       bgr24 :     24-bit BGR 8-8-8 :
[video4linux2,v4l2 @ 0x87b360] Raw       :       rgb24 :     24-bit RGB 8-8-8 :
[video4linux2,v4l2 @ 0x87b360] Raw       :        bgr0 : 32-bit BGRA/X 8-8-8-8 :
[video4linux2,v4l2 @ 0x87b360] Raw       :        0rgb : 32-bit A/XRGB 8-8-8-8 :
[video4linux2,v4l2 @ 0x87b360] Raw       :     yuyv422 :           YUYV 4:2:2 :
[video4linux2,v4l2 @ 0x87b360] Raw       :     uyvy422 :           UYVY 4:2:2 :

connortechnology, 2017-03-11:
I don' think case is the problem, as they all become numbers in the code.  

I will refer you to this for ideas https://forums.zoneminder.com/viewtopic.php?t=23380
but even if we find a workaround, I do want to fixup our code.

You are going to have to turn on debugging and set the level pretty high.  Which palette are you setting it to?  PAL or NTSC?

kingofkya, 2017-03-12:
AH yeah i see what you mean the case is irelevent in code.

In case it helps 
debug log.
http://pastebin.com/8a2enLyv

smaller snip
``
03/11/17 17:22:30.257663 zmc_dvideo0[8242].DB9-zm_image.cpp/2391 [478 - 1: min_y: 0, max_y:479, min_x:339.00, 1/m:0.00]
03/11/17 17:22:30.257670 zmc_dvideo0[8242].DB9-zm_image.cpp/2441 [Deleting active_edge]
03/11/17 17:22:30.257675 zmc_dvideo0[8242].DB9-zm_image.cpp/2441 [Deleting active_edge]
03/11/17 17:22:30.258063 zmc_dvideo0[8242].DB1-zm_monitor.cpp/2212 [Loaded monitor 11(Monitor-11), 1 zones]
03/11/17 17:22:30.258073 zmc_dvideo0[8242].INF-zmc.cpp/250 [Starting Capture version 1.30.0]
03/11/17 17:22:30.290471 zmc_dvideo0[8242].DB3-zm_local_camera.cpp/685 [Opening video device /dev/video0]
03/11/17 17:22:30.290499 zmc_dvideo0[8242].DB2-zm_local_camera.cpp/691 [V4L2 support enabled, using V4L2 api]
03/11/17 17:22:30.290508 zmc_dvideo0[8242].DB3-zm_local_camera.cpp/696 [Checking video device capabilities]
03/11/17 17:22:30.290521 zmc_dvideo0[8242].DB3-zm_local_camera.cpp/706 [Setting up video format]
03/11/17 17:22:30.290531 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/714 [ v4l2_data.fmt.type = 00000001]
03/11/17 17:22:30.290539 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/715 [ v4l2_data.fmt.fmt.pix.width = 00000154]
03/11/17 17:22:30.290547 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/716 [ v4l2_data.fmt.fmt.pix.height = 000001e0]
03/11/17 17:22:30.290554 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/717 [ v4l2_data.fmt.fmt.pix.pixelformat = 33424752]
03/11/17 17:22:30.290561 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/718 [ v4l2_data.fmt.fmt.pix.field = 00000003]
03/11/17 17:22:30.290569 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/719 [ v4l2_data.fmt.fmt.pix.bytesperline = 000003fc]
03/11/17 17:22:30.290577 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/720 [ v4l2_data.fmt.fmt.pix.sizeimage = 00077880]
03/11/17 17:22:30.290585 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/721 [ v4l2_data.fmt.fmt.pix.colorspace = 00000001]
03/11/17 17:22:30.290593 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/722 [ v4l2_data.fmt.fmt.pix.priv = feedcafe]
03/11/17 17:22:30.290604 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/748 [ v4l2_data.fmt.type = 00000001]
03/11/17 17:22:30.290612 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/749 [ v4l2_data.fmt.fmt.pix.width = 00000154]
03/11/17 17:22:30.290620 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/750 [ v4l2_data.fmt.fmt.pix.height = 000001e0]
03/11/17 17:22:30.290627 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/751 [ v4l2_data.fmt.fmt.pix.pixelformat = 33424752]
03/11/17 17:22:30.290634 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/752 [ v4l2_data.fmt.fmt.pix.field = 00000003]
03/11/17 17:22:30.290641 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/753 [ v4l2_data.fmt.fmt.pix.bytesperline = 000003fc]
03/11/17 17:22:30.290649 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/754 [ v4l2_data.fmt.fmt.pix.sizeimage = 00077880]
03/11/17 17:22:30.290657 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/755 [ v4l2_data.fmt.fmt.pix.colorspace = 00000001]
03/11/17 17:22:30.290665 zmc_dvideo0[8242].DB4-zm_local_camera.cpp/756 [ v4l2_data.fmt.fmt.pix.priv = feedcafe]
03/11/17 17:22:30.290693 zmc_dvideo0[8242].DB3-zm_local_camera.cpp/798 [Setting up request buffers]
03/11/17 17:22:30.290703 zmc_dvideo0[8242].DB3-zm_local_camera.cpp/811 [Request buffers count is 8]
03/11/17 17:22:30.291169 zmc_dvideo0[8242].DB3-zm_local_camera.cpp/831 [Setting up data buffers: Channels 1 MultiBuffer 0 Buffers: 8]
03/11/17 17:22:30.291323 zmc_dvideo0[8242].DB3-zm_local_camera.cpp/881 [Configuring video source]
03/11/17 17:22:30.291344 zmc_dvideo0[8242].FAT-zm_local_camera.cpp/906 [Failed to set video standard 4096: Device or resource busy]

closed 2018-06-22 by stale[bot]