ZoneMinder/zoneminder #734: FI9821W v2 shows blue screen

duckmanz, 2015-02-28:
Hi people,
Configured my Foscam FI9821W v2 in zoneminder v1.28.1 as 
ffmpeg, 
source path: rtsp://[user]:[pw]@[IP]:88/videoMain. 
With 24 bit coulor and 1280x720 resolution. Remote method is RTP/RTSP/HTTP. 

The only thing I got is a blue screen with running time at the top. It's also showing fps!
With VLC stream works fine.

Does anyone else got this same model FI9821W working in Zoneminder? Tried a lot of different settings but nothing seems to work.

Second IPCam is working fine. (simple jpeg)

christophey2k, 2015-02-28:
hi, have the same problem with my foscam cam FI9821W V1 and FI8620, after zm > v1.26 and latest ffmpeg version, have blue screen or corrup live stream with ffmpeg RTP/unicast or RTP/RTSP 
libvlc have good live streming with foscam but since some month we can't use it du to a segment fault...

For me , only mjpeg cam working fine...

knnniggett, 2015-02-28:
Yes, I own a 9821W v2 and do my zoneminder development with it.  Some of the latest native RTSP improvements in 1.28.1 are based off testing with that specific camera model.

This camera has always worked with a source type of ffmpeg.  1.28.1 adds support using a source type of Remote, what we are calling native RTSP support.

Your description of how you populated the fields is not complete for a source type of Remote.  

Compare what you did to a screenshot of my working camera:
[image]

I was a bit surprised to see that the ffmpeg method was noticeably more cpu efficient than our native RTSP.  I would be interested in knowing if others experience this as well.  On my server, we are talking a difference of ~%16 cpu to ~%11 cpu.  Not much, but it will add up once in a typical system with multiple cameras.

duckmanz, 2015-02-28:
Thanks for the quick response guys!

knniggett your settings work brilliant!! Thanks!! The problem that it didn't work in my case, was the fact that I selected "ffmpeg" as source type, and for some reason the string "rtsp://[user]:[pw]@[IP]:88/videoMain"  doesn't work.

Again thanks for your help!

knnniggett, 2015-02-28:
Well, holdup.  Using a source type of ffmpeg works as well.  
That has always worked:

[image]

duckmanz, 2015-02-28:
I tried ffmpeg again; first deleted the cam, and configured it as ffmpeg and got a blue screen again. for "Remote Method" I select "RTP/RTSP/HTTP", if I select an other option there I don't get an image at all.

knnniggett, 2015-03-01:
RTP/Unicast is the correct Remote Method to use.
You have to look in your log files to see what is going on.
You should also be using a relatively new version of ffmpeg.

duckmanz, 2015-03-01:
If I use RTP/Unicast it doesn't even give me the blue screen, it just shows a broken link.
I think my version of ffmpeg is an old version:

ffmpeg version 0.8.16-4:0.8.16-0ubuntu0.12.04.1, Copyright (c) 2000-2014 the Libav developers

knnniggett, 2015-03-01:
Oh yeah, that's it.  Well, now we know.

christophey2k, 2015-03-08:
Hi, all 
with the first method of knnniggett for setup FI9821W V1 the cam work  but ...

if i use RTP/RTSP have good image but the system figed after few minutes or rapidly when moving camera, after bug zm is innusable for 1-3 minutes with no response and no image
after this period zm and live come back for few minute or less and rebug ...

2015-03-08 09:41:15.273974	web_php	25943	ERR	socket_sendto( /var/lib/zoneminder/sock/zms-614446s.sock ) failed: No such file or directory	includes/functions.php	2342
2015-03-08 09:41:05.266625	web_php	25855	ERR	socket_sendto( /var/lib/zoneminder/sock/zms-614446s.sock ) failed: No such file or directory	includes/functions.php	2342
2015-03-08 09:40:55.262059	web_php	25755	ERR	socket_sendto( /var/lib/zoneminder/sock/zms-614446s.sock ) failed: No such file or directory	includes/functions.php	2342
2015-03-08 09:40:45.254710	web_php	25757	ERR	socket_sendto( /var/lib/zoneminder/sock/zms-614446s.sock ) failed: No such file or directory	includes/functions.php	2342
2015-03-08 09:40:35.247786	web_php	25852	ERR	socket_sendto( /var/lib/zoneminder/sock/zms-614446s.sock ) failed: No such file or directory	includes/functions.php	2342
2015-03-08 09:40:16.742150	zmdc	25788	ERR	'zmc -m 1' exited abnormally, exit status 255	zmdc.pl	
2015-03-08 09:40:16.714878	zmc_m1	25921	ERR	Failed to pre-capture monitor 1 (0/1)	/usr/src/zoneminder/src/zmc.cpp	273
2015-03-08 09:40:16.176099	zmc_m1	25922	ERR	RTSP timed out	/usr/src/zoneminder/src/zm_rtsp.cpp	714
2015-03-08 09:38:24.246860	zmdc	25788	ERR	'zmc -m 1' exited abnormally, exit status 255	zmdc.pl	
2015-03-08 09:38:24.220823	zmc_m1	25912	ERR	Failed to pre-capture monitor 1 (0/1)	/usr/src/zoneminder/src/zmc.cpp	273
2015-03-08 09:38:23.682679	zmc_m1	25913	ERR	RTSP timed out

015-03-08 09:37:15.112307	web_php	25756	ERR	socket_sendto( /var/lib/zoneminder/sock/zms-270774s.sock ) failed: No such file or directory	includes/functions.php	2342
2015-03-08 09:36:52.328550	zmdc	25788	ERR	'zmc -m 1' exited abnormally, exit status 255	zmdc.pl	
2015-03-08 09:36:52.301408	zmc_m1	25892	ERR	Failed to pre-capture monitor 1 (0/1)	/usr/src/zoneminder/src/zmc.cpp	273
2015-03-08 09:36:51.762533	zmc_m1	25893	ERR	RTSP timed out	/usr/src/zoneminder/src/zm_rtsp.cpp	714
2015-03-08 09:35:28.387330	zmdc	25788	ERR	'zmc -m 1' exited abnormally, exit status 255	zmdc.pl	
2015-03-08 09:35:28.331428	zmc_m1	25875	ERR	Failed to pre-capture monitor 1 (0/1)	/usr/src/zoneminder/src/zmc.cpp	273
2015-03-08 09:35:27.791374	zmc_m1	25876	ERR	RTSP timed out

if i use the RTSP/Unicast the image is very buggy specially with many image activity (with static image that correct) but the system do not crash or figed with this RTSP/Unicast setup ...

I made this test with only one FOSCAM FI9821W V1 actived and enabled in zm

arjunroychowdhury, 2015-03-23:
knnniggett, I just noticed this - that the remote method sidesteps ffmpeg. I'm testing this now - I've set up one of my foscams (9831W) to use the remote method while the others are using ffmpeg. I'm trying to eliminate the occasional grey frame/smear.

Quick question: I looked at am_rtsp.cpp (traced from zm_monitor.cpp) and am I right in assuming that if its set to "remote" and RTSP protocol and RTP/RTSP method, you are forcing TCP (even though foscam also supports UDP RTSP)?

arjunroychowdhury, 2015-03-23:
I've been testing the remote option. It worked great for all of 10 minutes or so, after that, it just doesn't seem to work anymore.

My port # is 20001, but in the logs I see

2015-03-23 18:39:44.255471	zmc_m3	713	INF	Recieved new Content-Base in DESCRIBE reponse header. Updated device Url to: 'rtsp://192.168.1.125:65534/videoMain/'	zm_rtsp.cpp
2015-03-23 18:42:10.721264	zmc_m3	770	FAT	No RTSP sources	zm_remote_camera_rtsp.cpp

tstrike34, 2015-04-11:
Arjun,

I experience the same with DLINK cameras. I have 1.28.1 but I do not have the new Remote option.

Weird.

arjunroychowdhury, 2015-04-11:
Tstrike, life is full of surprises when it comes to ZoneMinder. I think how it behaves depends on what you ate the morning you installed it :-)) I re-installed zone minder from source 2 days ago via cmake -- and now there remote option is working perfectly! I have been running one of HD cameras on their remote mode and its been running extremely well - no smudding

tstrike34, 2015-04-11:
wow ok .... looks like I will need to compile it.... Grab it from github?  Thanks Arjun!

tstrike34, 2015-04-11:
I did the upgrade from the PPA on Ubuntu 14.04... Is there something wrong with the PPA? I would hate to have to compile it from source. 
[image]

tstrike34, 2015-04-11:
[image]
Yup got streaking.... you must have the latest source code drop

arjunroychowdhury, 2015-04-12:
Ah yes I did use source. I think iconnor does the ppas. I'm not sure his
ppas have all the latest and greatest.

tstrike34, 2015-04-12:
Ok drats.... *sigh* I guess tonight's project is installing from source and performing an upgrade...

I upgraded the very STALE VMware Zoneminder image....

arjunroychowdhury, 2015-04-12:
Wait hold on. The remote option works (or not ffmpeg) but you are getting
smudging? I dont think compiling from source will fix that.

tstrike34, 2015-04-12:
Not I dont even have the Remote Option....

The only remote option I have is on the first page of the Monitor set up under source type.

tstrike34, 2015-04-12:
Oh Wait I see what you did!!!   Remote on the first page ... on the 2nd tab I need to choose RTSP instead of HTTP... Duh!

arjunroychowdhury, 2015-04-12:
Right.

tstrike34, 2015-04-12:
[image]

I had to use UniCast...  the others did not work at all... RTP/RTSP drove my CPU NUTS...

tstrike34, 2015-04-12:
[image]
Here is what the log is saying

knnniggett, 2015-04-12:
Is your camera really set to 320 x 176 resolution to match what you've got in the monitor config???
Bump your colorspace to 32bit for a performance improvement.

If your monitor config is correct, those types of errors mean zoneminder is getting garbage from the camera.  It could be a libav/ffmpeg issue as that library does the interpreting for zoneminder. 

Items to look into:
- use a newer version of ffmpeg (switch to ffmpeg if you are using debain/ubuntu and have libav). This can be more complicated than it sounds because zoneminder detects the version of ffmpeg/libav at build time.  You could do what another user did... fire up a new vm with a distro like centos or fedora then install zonemidner from zmrepo.  zmrepo has ffmpeg 0.10.x
- network issues
- broken camera

arjunroychowdhury, 2015-04-12:
@knnniggett - he is using the remote method in source. In a forum post you mentioned this switches to your own RTSP player and does not use ffmpeg. Your post above seems to indicate that the native player also uses ffmpeg internally - can you clarify?

knnniggett, 2015-04-12:
The monitor config screenshot from the previous post shows a source tab that reflect a source type of "ffmpeg", not "Remote". The Remote source type does not have the "Source Path" field in the Source tab. Instead is has "Remote Protocol".

arjunroychowdhury, 2015-04-12:
I think you are looking at the old screenshot. He posted a new one 10 hours ago. That shows the correct screen with Remote protocol. https://github.com/ZoneMinder/ZoneMinder/issues/734#issuecomment-91976099

knnniggett, 2015-04-12:
Well, I guess I should finish my morning coffee before posting.

tstrike34, 2015-04-12:
Hey guys good morning... just woke up... Yes I am aware my PWD was there (its a fake one hehehe).
 
So I let ZM run overnight with RTP/Unicast... CPU seems to be ok, images appear to be ok, still got those errors indicated in the logfile... I forgot to tell you guys what cameras I am using (DOH)..

The cameras are DLINK DCS-2330L.

@knnniggett I am seriously thinking about switching to Centos (I really like Centos for more weighty applications like ZM and others)..  I am using the Ubuntu distro because of the ease of use and updating the VM appliance. I was seriously thinking about updating the applicance and resubmitting it to VMWare but didnt want the hassle of its upkeep.

tstrike34, 2015-04-12:
by the way Knnniggett, that was a GREAT forum post!

I wonder if ZM will ever get the audio recording portion working... do you guys use anything like ZM to record audio that corresponds with the video?

tstrike34, 2015-04-13:
Hey guys, I tried again with RTP/RTSP... it is hiccuping something ugly... it is if it doesnt want to make a proper connection or transmission over TCP/IP .... the smearing is gone on UDP.. but uggh. I switched back to RTSP/Unicast.

Is there something I can tweak in my network configs on the server itself? I already changed up the shmmax... I think I am going to up /run/shm to see if that changes things up.

knnniggett, 2015-06-17:
sorry for the late response on this one.  
@tstrike34 Your camera is not the same make/model as the OP.  If you are still having issues, please create a new issue and provide a Level 5 debug, showing the RTSP handshake process (e.g. DESCRIBE, SETUP, PLAY, etc.), and please post your monitor config settings as well for this camera.

closed 2015-06-17 by knnniggett

tstrike34, 2015-06-17:
OK will do                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 and Playbook on the Verizon Wireless 4G LTE network.                                                                                                                                                                                                                From: Andrew BauerSent: Wednesday, June 17, 2015 10:21 AMTo: ZoneMinder/ZoneMinderReply To: ZoneMinder/ZoneMinderCc: tstrike34Subject: Re: [ZoneMinder] FI9821W v2 shows blue screen (#734)sorry for the late response on this one.
@tstrike34 Your camera is not the same make/model as the OP.  If you are still having issues, please create a new issue and provide a Level 5 debug, showing the RTSP handshake process (e.g. DESCRIBE, SETUP, PLAY, etc.), and please post your monitor config settings as well for this camera.