ZoneMinder/zoneminder #736: Frame rate doesn't match set rate when in modect

river100, 2015-03-01:
If I use an IP cam streaming in H264 at 15 fps, any events using Modect have from 8 to maybe 9.5 fps with ZM vers 1.28.
The vers I have for reference is 1.25 which will record in Modect at the set frame rate of the cam.
I've tried this on ubuntu 14.04 LTS 32bit and 64bit. The ZM vers is 1.28 built using git and the instructions on the main page.  I've turned off all cams except one and still get the same response.

The hardware is server MOBO (Supermicro) 2Xs quad core ZEON L5420, 8GB ram SATA II HDs.

On the same hardware vers 1.25.0 I can get 13.5 to 15.0 in Modect. Using ffmpeg as source.
When using Mocord or Record modes the frame rate is 13.5 to 14.5, which is about what I got with 1.25.0
I'm going to set the source as Libvlc and check the frames.

river100, 2015-03-01:
Some more info on this bug. I took one cam and set the frame rate to 25 fps.  I get 22.14 fps in Mocord.
So I tried "Forced Web", with the cam at 25, I gor 8.0 fps.  No Motion to see what the 25 fps will do but looks like any "alarm" gives less than 10 no matter if set at higher rate.
Guess I need to check what happens if rate is set at less then 10 fps.

connortechnology, 2015-03-02:
I've been looking at the code that controls fps... and I think it is flawed, although I'm having trouble wrapping my head around it.

closed 2017-03-18 by SteveGilvarry