ZoneMinder/zoneminder #738: [Critical] 1.28.0 to 1.28.1 seems to have lost ability for RTSP Authentication

jamesarbrown, 2015-03-01:
Upgraded Ubuntu via PPA (1.28.1 Trusty amd64) and suddenly cameras are not working.

Switched on Debug and got
```
03/01/15 16:44:11.243044 zmc_m1[787].DB2-zm_rtsp.cpp/74 [Received RTSP response: RTSP/1.0 401 Unauthorized
Server: Rtsp Server/2.0
CSeq: 1
WWW-Authenticate: Basic realm="RtspSever2.0"

 (109 bytes)]
03/01/15 16:44:11.243165 zmc_m1[787].DB2-zm_rtsp.cpp/94 [Got 401 access denied response code, check WWW-Authenticate header and retry]
03/01/15 16:44:11.243209 zmc_m1[787].DB2-zm_rtsp_auth.cpp/218 [Found auth line at 3]
03/01/15 16:44:11.243222 zmc_m1[787].DB2-zm_rtsp_auth.cpp/223 [Analyze auth line WWW-Authenticate: Basic realm="RtspSever2.0"]
03/01/15 16:44:11.243243 zmc_m1[787].DB2-zm_rtsp_auth.cpp/67 [Set authMethod to Basic]`
```

So I uninstalled, downloaded
*https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder/+build/6474856/+files/zoneminder_1.28.0-trusty_amd64.deb*

Which worked out of the box

Cameras are set in this format
[image]

knnniggett, 2015-03-01:
RTSP authentication is working fine here.

That 401 response is coming from the camera itself.  To troubleshoot further, we need to see what zoneminder is sending to the camera. Turn your debug up to level 5 and then post the entire RTSP handshake showing all the SETUP, DESCRIBE, PLAY, etc steps.  Please show the whole log.  Don't cherry pick the errors out.

Knowing the make and model of your camera would help as well.

NOTE: You will get a performance improvement if you select 32bit colorspace, instead of 24.

jamesarbrown, 2015-03-01:
Please give me the specific log setup you need to see? Firing ZM up creates some 10 log files and soon starts spawning a more files.... and level 5 does not seem to spawn any extra RTSP info?

A full log file below for camera zmc_m1. (zmc_m2 is also doing same)

As soon as I installed 1.28.1 over 1.28.0 these cameras went red (with no config change). 

VLC connects. 1.28.0 connects. 1.28.1 does not connect.

The 401 response appears to be as the camera is wanting ZM to be authenticated before the RTSP Options query, which is is either not doing or not achieving.

This system as been working for months on 1.28.0

Camera API is here
https://www.dropbox.com/s/5pi7ybrb72h98lj/Nexus%20API%20information.pdf?dl=0

```
::::::::::::::
zm_debug.log.07285
::::::::::::::
03/01/15 17:54:01.285193 zmc_m1[7285].DB1-zm_logger.cpp/230 [LogOpts: level=DB5/
DB5, screen=OFF, database=INF, logfile=DB5->/var/log/zm/zm_debug.log.07285, sysl
og=ERR]
03/01/15 17:54:01.285245 zmc_m1[7285].DB1-zm_utils.cpp/262 [Detected a x86\x86-6
4 processor with SSSE3]
03/01/15 17:54:01.285659 zmc_m1[7285].DB1-zm_monitor.cpp/2486 [Got 1 monitors]
03/01/15 17:54:01.285671 zmc_m1[7285].DB1-zm_monitor.cpp/2521 [Got 1 for v4l_cap
tures_per_frame]
03/01/15 17:54:01.285688 zmc_m1[7285].DB2-zm_camera.cpp/39 [New camera id: 1 wid
th: 2048 height: 1536 colours: 3 subpixelorder: 6 capture: 1]
03/01/15 17:54:01.286860 zmc_m1[7285].DB1-zm_thread.cpp/241 [Creating thread]
03/01/15 17:54:01.286887 zmc_m1[7285].DB2-zm_rtsp.cpp/197 [RTSP Local SSRC is 2b
eae058]
03/01/15 17:54:01.286902 zmc_m1[7285].DB1-zm_thread.cpp/281 [Starting thread]
03/01/15 17:54:01.286947 zmc_m1[7287].DB2-zm_thread.cpp/253 [Invoking thread]
03/01/15 17:54:01.286989 zmc_m1[7285].DB1-zm_thread.cpp/302 [Started thread 7287
]
03/01/15 17:54:01.287010 zmc_m1[7285].DB2-zm_image.cpp/167 [Blend: Using SSE2 fa
st blend function]
03/01/15 17:54:01.287043 zmc_m1[7285].DB2-zm_image.cpp/203 [Delta: Using SSSE3 d
elta functions]
03/01/15 17:54:01.287049 zmc_m1[7285].DB2-zm_image.cpp/247 [Deinterlace: Using S
SSE3 delta functions]
03/01/15 17:54:01.287054 zmc_m1[7285].DB2-zm_image.cpp/260 [Image buffer copy: U
sing SSE2 aligned memcpy]
03/01/15 17:54:01.287058 zmc_m1[7285].DB3-zm_image.cpp/267 [Setting up static co
lour tables]
03/01/15 17:54:01.287093 zmc_m1[7285].DB1-zm_monitor.cpp/369 [monitor purpose=1]
03/01/15 17:54:01.287098 zmc_m1[7285].DB1-zm_monitor.cpp/377 [mem.size=660604960
]
03/01/15 17:54:01.287120 zmc_m1[7285].DB1-zm_monitor.cpp/522 [Unable to map file
 /dev/shm/zm.mmap.1 (660604960 bytes) to locked memory, trying unlocked]
03/01/15 17:54:01.287129 zmc_m1[7285].DB1-zm_monitor.cpp/524 [Mapped file /dev/s
hm/zm.mmap.1 (660604960 bytes) to locked memory, unlocked]
03/01/15 17:54:01.287485 zmc_m1[7287].DB2-zm_rtsp.cpp/48 [Sending RTSP message: 
OPTIONS rtsp://192.168.1.201:554/ RTSP/1.0
User-Agent: ZoneMinder/1.28.1
CSeq: 1

]
03/01/15 17:54:01.306961 zmc_m1[7287].DB2-zm_rtsp.cpp/74 [Received RTSP response
: RTSP/1.0 401 Unauthorized
Server: Rtsp Server/2.0
CSeq: 1
WWW-Authenticate: Basic realm="RtspSever2.0"

 (109 bytes)]
03/01/15 17:54:01.307032 zmc_m1[7287].DB2-zm_rtsp.cpp/94 [Got 401 access denied 
response code, check WWW-Authenticate header and retry]
03/01/15 17:54:01.307059 zmc_m1[7287].DB2-zm_rtsp_auth.cpp/218 [Found auth line 
at 3]
[... 56 lines cut ...]
03/01/15 17:54:01.648935 zmc_m1[7285].DB3-zm_zone.cpp/823 [Parsing polygon strin
g '0,0 2047,0 2047,1535 0,1535']
03/01/15 17:54:01.648941 zmc_m1[7285].DB3-zm_zone.cpp/861 [Got coordinate 0,0 fr
om polygon string]
03/01/15 17:54:01.648945 zmc_m1[7285].DB3-zm_zone.cpp/861 [Got coordinate 2047,0
 from polygon string]
03/01/15 17:54:01.648949 zmc_m1[7285].DB3-zm_zone.cpp/861 [Got coordinate 2047,1
535 from polygon string]
03/01/15 17:54:01.648953 zmc_m1[7285].DB3-zm_zone.cpp/861 [Got coordinate 0,1535
 from polygon string]
03/01/15 17:54:01.648959 zmc_m1[7285].DB3-zm_zone.cpp/881 [Successfully parsed p
olygon string]
03/01/15 17:54:01.648964 zmc_m1[7285].DB1-zm_zone.cpp/51 [Initialised zone 5/All
 - 1 - 2048x1536 - Rgb:0000ff, CM:2, MnAT:40, MxAT:0, MnAP:94371, MxAP:2359296, 
FB:3x3, MnFP:31457, MxFP:1887436, MnBS:157286, MxBS:0, MnB:1, MxB:0, OF: 5, AF: 
0]
03/01/15 17:54:01.705295 zmc_m1[7285].DB1-zm_monitor.cpp/2774 [Loaded monitor 1(
FrontCamera), 2 zones]
03/01/15 17:54:01.705312 zmc_m1[7285].INF-zmc.cpp/195 [Starting Capture]
03/01/15 17:54:01.755582 zmc_m1[7285].DB2-zm_remote_camera_rtsp.cpp/149 [Waiting
 for sources]
03/01/15 17:54:11.763278 zmc_m1[7285].FAT-zm_remote_camera_rtsp.cpp/155 [No RTSP
 sources]

```

knnniggett, 2015-03-01:
Referring to chapter 3.2Authentication, which authentication mode is the camera currently set to?
From the log, the camera is saying it is set to basic.  If so, set it to digest and the note the difference.

jamesarbrown, 2015-03-02:
Will look later this evening (GMT+0 here), there does not seem to be an option for Digest in the web interface. There is an option for "trusted IP" which I can try for my issue, but does not help yourselves identify what part of the changes are now causing this issue.

knnniggett, 2015-03-02:
Yes, exactly.  We didn't change anything with basic authentication, so I don't see where the problem is at the moment.  We recently added support for digest authentication so maybe looking at the logs when attempting to use digest will shed some light.

knnniggett, 2015-03-02:
Also, make sure nothing else is attempting to stream from the camera.  While most cameras these days can handle multiple streams, many cannot do so using the exact same url.

connortechnology, 2015-03-02:
We will need you to switch to our master dev branch, by using my ppa
ppa:iconnor/zoneminder-master

I have built fresh images with a possible fix.

jamesarbrown, 2015-03-02:
Yes that works for me. +1.

Is there anything anyone wants me to try before this is closed off? 

Probably not the place, but save me creating a bug in Launchpad.

1/
PPA (during yesterday) may need checking for what I thought was a "clean" install (without a /etc/zm directory), complains zm.conf does not exist and aborts. I put a /etc/zm/zm.conf from a backup and it worked.

2/ 
PPA  (Master today) had some quirks, but maybe as I was also trying with a Git build yesterday.

```
root@pvr:/etc/apt/sources.list.d# apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  libavcodec-extra-53 libavformat-extra-53 libavutil-extra-51 linux-generic
  linux-headers-generic linux-image-generic mytharchive mythgallery
  mythtv-common mythtv-frontend mythweather mythweb mythzoneminder
The following packages will be upgraded:
  zoneminder
1 upgraded, 0 newly installed, 0 to remove and 13 not upgraded.
Need to get 1,037 kB of archives.
After this operation, 430 kB disk space will be freed.
Do you want to continue? [Y/n] y
Get:1 http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu/ trusty/main zoneminder amd64 1.28.1+1-trusty-SNAPSHOT2015030201 [1,037 kB]
Fetched 1,037 kB in 0s (1,543 kB/s)   
(Reading database ... 382626 files and directories currently installed.)
Preparing to unpack .../zoneminder_1.28.1+1-trusty-SNAPSHOT2015030201_amd64.deb ...
Stopping ZoneMinder: success

Unpacking zoneminder (1.28.1+1-trusty-SNAPSHOT2015030201) over (1.28.0-trusty) ...
dpkg: warning: unable to delete old directory '/var/run/zm': Directory not empty
dpkg: warning: unable to delete old directory '/build/buildd': Directory not empty
dpkg: warning: unable to delete old directory '/build': Directory not empty
Processing triggers for ureadahead (0.100.0-16) ...
Processing triggers for man-db (2.6.7.1-1ubuntu1) ...
Setting up zoneminder (1.28.1+1-trusty-SNAPSHOT2015030201) ...
Installing new version of config file /etc/zm/zm.conf ...
Stopping ZoneMinder: Zoneminder already stopped

Initiating database upgrade to version 1.28.1 from version 1.28.0

Upgrading database to version 1.28.1
Loading config from DB
Saving config to DB
Upgrading DB to 1.28.0 from 1.28.0

Database successfully upgraded to version 1.28.0.
Upgrading DB to 1.28.1 from 1.28.0

Database successfully upgraded to version 1.28.1.
Upgrading DB to 1.28.99 from 1.28.0

Database successfully upgraded to version 1.28.99.

Database upgrade to version 1.28.1 successful.

ZoneMinder is stopped
invoke-rc.d: initscript zoneminder, action "status" failed.

Database already at version 1.28.1, update aborted.
```

PR #739 (2015-03-02): when auth is needed, try command again before dying.
I believe this fixes #738
closed 2015-03-24 by connortechnology