ZoneMinder/zoneminder #741: FI8620 ERR	Error while decoding frame

christophey2k, 2015-03-08:
Hi, all
with ZM 1.28.1 and foscam FI8620 ("only h264" camera at 25fps from foscam)
this cam work only in RTSP/unicast mode and now have ton of this error
config: rtsp://admin:password@192.168.253.97:554/11  

Error while decoding frame 4166	/usr/src/zoneminder/src/zm_remote_camera_rtsp.cpp	302
2015-03-08 09:03:03.274128	zmc_m3	23950	ERR	8: 00 00 01 06 e5 01 a2 80	/usr/src/zoneminder/src/zm_remote_camera_rtsp.cpp	303
2015-03-08 09:03:03.273310	zmc_m3	23950	ERR	Error while decoding frame 4163	/usr/src/zoneminder/src/zm_remote_camera_rtsp.cpp	302
2015-03-08 09:03:03.272817	zmc_m3	23950	ERR	8: 00 00 01 06 e5 01 9e 80	/usr/src/zoneminder/src/zm_remote_camera_rtsp.cpp	303
2015-03-08 09:03:03.271721	zmc_m3	23950	ERR	Error while decoding frame 4163	/usr/src/zoneminder/src/zm_remote_camera_rtsp.cpp	302
2015-03-08 09:03:03.172386	zmc_m3	23950	ERR	8: 00 00 01 06 e5 01 9c 80	/usr/src/zoneminder/src/zm_remote_camera_rtsp.cpp	303
2015-03-08 09:03:03.171996	zmc_m3	23950	ERR	Error while decoding frame 4161	/usr/src/zoneminder/src/zm_remote_camera_rtsp.cpp	302
2015-03-08 09:03:03.171551	zmc_m3	23950	ERR	8: 00 00 01 06 e5 01 9a 80	/usr/src/zoneminder/src/zm_remote_camera_rtsp.cpp	303
2015-03-08 09:03:03.170371	zmc_m3	23950	ERR	Error while decoding frame 4161	/usr/src/zoneminder/src/zm_remote_camera_rtsp.cpp	302
2015-03-08 09:03:02.968207	zmc_m3	23950	ERR	8: 00 00 01 06 e5 01 98 80	/usr/src/zoneminder/src/zm_remote_camera_rtsp.cpp	303
2015-03-08 09:03:02.967430	zmc_m3	23950	ERR	Error while decoding frame 4157	/usr/src/zoneminder/src/zm_remote_camera_rtsp.cpp	302
2015-03-08 09:03:02.817112	zmc_m3	23950	ERR	8: 00 00 01 06 e5 01 96 80	/usr/src/zoneminder/src/zm_remote_camera_rtsp.cpp	303

SteveGilvarry, 2015-06-17:
Duplicate #841 closing the older one.

closed 2015-06-17 by SteveGilvarry

atimonin, 2015-12-03:
I see similar errors, but only if I use substream (no errors in main stream)
And I see error in every 25-th frame (one per second). Maybe it'll help to investigate?
I may provide tcpdump or debug info if needed.

curlyboi, 2017-01-04:
This error is produced when using Remote source with RTSP and is produced everytime the camera sends a keyframe (also known as I-Frame). Solution is to use FFMpeg instead of Remote (but this has better performance on weak networks)