ZoneMinder/zoneminder #743: replace all .gifs with .pngs or .jpegs as appropriate.

connortechnology, 2015-03-08:
There is no reason to use gifs instead of pngs anymore.

kylejohnson, 2015-03-10:
Agreed, but this sounds like a really small win, and something that is a non-issue in the new UI.

knnniggett, 2015-08-17:
Here are the list of .gif's currently in the project:
```./web/api/lib/Cake/Console/Templates/skel/webroot/img/cake.power.gif
./web/api/lib/Cake/Test/test_app/Plugin/TestPlugin/webroot/img/cake.icon.gif
./web/api/lib/Cake/Test/test_app/View/Themed/TestTheme/webroot/img/cake.power.gif
./web/api/lib/Cake/Test/test_app/webroot/img/cake.power.gif
./web/api/lib/Cake/Test/test_app/webroot/theme/test_theme/img/cake.power.gif
./web/api/app/webroot/img/cake.power.gif
./web/skins/classic/graphics/arrow-s-d.gif
./web/skins/classic/graphics/center.gif
./web/skins/classic/graphics/seq-u.gif
./web/skins/classic/graphics/arrow-l.gif
./web/skins/classic/graphics/point-r.gif
./web/skins/classic/graphics/arrow-dr.gif
./web/skins/classic/graphics/point-o.gif
./web/skins/classic/graphics/seq-d.gif
./web/skins/classic/graphics/arrow-ul.gif
./web/skins/classic/graphics/point-g.gif
./web/skins/classic/graphics/arrow-r.gif
./web/skins/classic/graphics/arrow-l-d.gif
./web/skins/classic/graphics/arrow-dl.gif
./web/skins/classic/graphics/arrow-d.gif
./web/skins/classic/graphics/arrow-s-u.gif
./web/skins/classic/graphics/arrow-l-u.gif
./web/skins/classic/graphics/arrow-u.gif
./web/skins/classic/graphics/arrow-ur.gif
./web/graphics/transparent.gif
./web/graphics/spinner.gif```

These are all the little images used for web page rendering.

Doing a search for gif's in other projects installed on my machine, I noticed that the following projects have gif's in one or more of their folders: firefox, thunderbird, libreoffice, openssl, and pidgin.
The Gif format seems to still be used by modern projects.

We can't really make a claim for quality by converting them in this case, because we don't have the original bitmaps the .gif's may have originated from.

I personally don't see a need to convert these at all.  but, in any case, can we either convert them and/or close this issue out?

connortechnology, 2015-08-17:
For giggles, I did arrow-d.gif to arrow-d.png, and I replaced the white background with  transparency, which makes them a little more useful in other colour schemes.  I was able to remove 60 bytes.  

Is the effort worth it?  I don't know.  Probably a whole new set of images would be a better investment in time.

newburns, 2015-08-17:
WHHAAAATTTT?????
I think I can actually help (FINALLY).
How do I convert the images? Can they just be images that I create, which closely resemble the gif?
I will try to tackle this in Photoshop or Illustrator.... Is there any specific size requirement? I'll just

connortechnology, 2015-08-17:
I tried image magick.. but the png file was larger...

SO I used the gimp.  

Load image, convert to rgb mode, add transparency layer, do colour to alpha choosing white.  convert to indexed mode, save as png.

newburns, 2015-08-17:
Will do!
What code do I need to change so the UI will load the png vs. gif?
Just a find and replace on all the files for .gif --> .png?

connortechnology, 2015-08-17:
yup.

SteveGilvarry, 2017-03-18:
PR done, they are bigger files, didn't do transparent  as some had a fade in from white

closed 2017-03-18 by SteveGilvarry

connortechnology, 2017-03-18:
You can often improve the png compression.  I think the util is pngcrush... or something like that.

SteveGilvarry, 2017-03-18:
This was just gimp open export as.

connortechnology, 2017-03-18:
Yeah gimp will generate a non-optimal png.  Use pngcrush to shrink them. They will end up smaller than gifs.  I think what it basically does is used lower bit depth if possible.

SteveGilvarry, 2017-03-20:
pngcrush -brute -reduce arrow-d.png arrow-d-b.png got me 3 bytes, so must not be using right options. Give it a go when you get back.

connortechnology, 2017-03-20:
I played around a bit... switching to rgb and then back to indexed mode got me down to 216 bytes on arrow-d. The gif was 293.  

An even better approach is to merge them all into 1 image, and use css rules to snip the correct image out of the sheet.  Not sure that is worth it though.  I think there are arrows in the bootstrap supplied icons which we could use instead of these.

connortechnology, 2017-03-20:
removing creation time, etc takes several bytes out too.