ZoneMinder/zoneminder #745: zmc 1.28.1 crashes on gentoo box

qpaz, 2015-03-13:
hi folks,

first off all, you are really doing great stuff here, appreciate that. 
i just wanted to update from version 1.27.99.0 to 1.28.1, but it fails. i downgraded after that immediately back to 1.27.99, cause this is kind of prod system. i read about the new path (https://github.com/ZoneMinder/ZoneMinder/pull/524) and created the appropriate with perms. its on gentoo 3.10.25 with vlc 2.1.5.-r1 and libav-9.17. i also tried with fresh database, but doesnt work out. hints? if the addr2line command is needed, i will upgrade again, but maybe its my fault, just wanted to know if someone is aware of thins...

Mar 12 14:34:46 localhost zmc_m5[10905]: INF [Starting Capture]
Mar 12 14:34:46 localhost zmdc[8666]: ERR ['zmc -m 10' exited abnormally, exit status 11]
Mar 12 14:34:46 localhost zmc_m5[10905]: INF [Priming capture from rtsp://10.9.3.12:554/live2.sdp]
Mar 12 14:34:47 localhost zmc_m5[10905]: ERR [Got signal 11 (Segmentation fault), crashing]
Mar 12 14:34:47 localhost zmc_m5[10905]: ERR [Signal address is 0xffffffffffffffe8, from 0x477990]
Mar 12 14:34:47 localhost zmc_m5[10905]: ERR [Backtrace 0: /usr/local/bin/zmc(_Z14zm_die_handleriP9siginfo_tPv+0x6c) [0x487a7c]]
Mar 12 14:34:47 localhost zmc_m5[10905]: ERR [Backtrace 1: /lib64/libpthread.so.0(+0x11240) [0x7fc24d972240]]
Mar 12 14:34:47 localhost zmc_m5[10905]: ERR [Backtrace 2: /usr/local/bin/zmc(_ZN13Authenticator5resetEv+0x10) [0x477990]]
Mar 12 14:34:47 localhost zmc_m5[10905]: ERR [Backtrace 3: /usr/lib64/libliveMedia.so.4(_ZN13AuthenticatoraSERKS_+0x1e) [0x7fc1bb3974fe]]
Mar 12 14:34:47 localhost zmc_m5[10905]: ERR [Backtrace 4: /usr/lib64/libliveMedia.so.4(_ZN10RTSPClient18sendOptionsCommandEPFvPS_iPcEP13Authenticator+0x2f) [0x7fc1bb37af2f]]
Mar 12 14:34:47 localhost zmc_m5[10905]: ERR [Backtrace 5: /usr/lib64/vlc/plugins/access/liblive555_plugin.so(+0x657d) [0x7fc1bb5d357d]]
Mar 12 14:34:48 localhost zmc_m5[10905]: ERR [Backtrace 6: /usr/lib64/vlc/plugins/access/liblive555_plugin.so(+0x6afa) [0x7fc1bb5d3afa]]
Mar 12 14:34:48 localhost zmc_m5[10905]: ERR [Backtrace 7: /usr/lib64/libvlccore.so.7(+0x6c9c8) [0x7fc2464809c8]]
Mar 12 14:34:48 localhost zmc_m5[10905]: ERR [Backtrace 8: /usr/lib64/libvlccore.so.7(vlc_module_load+0x2c0) [0x7fc246480dc0]]
Mar 12 14:34:48 localhost zmc_m5[10905]: ERR [Backtrace 9: /usr/lib64/libvlccore.so.7(+0x36e38) [0x7fc24644ae38]]
Mar 12 14:34:48 localhost zmc_m5[10905]: ERR [Backtrace 10: /usr/lib64/libvlccore.so.7(+0x42093) [0x7fc246456093]]
Mar 12 14:34:48 localhost zmc_m5[10905]: ERR [Backtrace 11: /usr/lib64/libvlccore.so.7(+0x4303d) [0x7fc24645703d]]
Mar 12 14:34:48 localhost zmc_m5[10905]: ERR [Backtrace 12: /usr/lib64/libvlccore.so.7(+0x461c6) [0x7fc24645a1c6]]
Mar 12 14:34:48 localhost zmc_m5[10905]: ERR [Backtrace 13: /lib64/libpthread.so.0(+0x91da) [0x7fc24d96a1da]]
Mar 12 14:34:49 localhost zmc_m5[10905]: ERR [Backtrace 14: /lib64/libc.so.6(clone+0x6d) [0x7fc24a94c8cd]]
Mar 12 14:34:49 localhost zmc_m5[10905]: INF [Backtrace complete, please execute the following command for more information]
Mar 12 14:34:49 localhost zmc_m5[10905]: INF [addr2line -e /usr/local/bin/zmc 0x487a7c 0x7fc24d972240 0x477990 0x7fc1bb3974fe 0x7fc1bb37af2f 0x7fc1bb5d357d 0x7fc1bb5d3afa 0x7fc2464809c8 0x7fc246480dc0 0x7fc24644ae38 0x7fc246456093 0x7fc24645703d 0x7fc24645a1c6 0x7fc24d96a1da 0x7fc24a94c8cd]
Mar 12 14:34:49 localhost zmdc[8666]: ERR ['zmc -m 5' exited abnormally, exit status 11]

christophey2k, 2015-03-13:
hi for me same ptobleme recent zm don't wotk 
with libvlc.

i'm not sure if i upgraded vlc and zm at the same time or if this is a pure zm bug

the workaround is to use rtsp method or ffmpeg for connection

but live stream quality is poor

Envoyé depuis Yahoo Mail pour Android

knnniggett, 2015-03-13:
Looks like this is the same as #343.  
I've experienced this myself... one moment it works fine and the next it doesn't.

All we know is that the fault appears to be happening in the live555 library, not in directly zoneminder.  However, if someone could figure out exactly what conditions are causing this, it would be very helpful towards working around the problem.  Maybe it is caused when the version of vlc on the system is different than the version zoneminder was built against, maybe certain combinations of parameters (framerate, resolution, etc) are causing this, or maybe it is something else.  

For production systems, you should switch to your source type to ffmpeg or remote until someone figures this one out.

qpaz, 2015-03-16:
hi,

in my situation its not caused by the vlc version on the system, vlc 2.1.5.-r1 and libav-9.17 runs without any problems with zm 1.27.99 but not with zm 1.28.1 (after problems with 1.28 i build 1.27.99 fresh from cloned older github sources). so i think it has nothing to do with the vlc version build against. 

when i use ffmpeg as source type, then i got really weird pictures and the only thing which works really nice with vivotek ip8364c cams is with libvlc (in rtsp).

knnniggett, 2015-06-17:
The issue is in the live555 library. We currently do not know how to fix this, other than to use a different source type. Issue #343 is set up to track this.  Please post new information to that, rather than create a new issue.

closed 2015-06-17 by knnniggett