ZoneMinder/zoneminder #747: Integrate Make Movie Branch

newburns, 2015-03-16:
What would be required by the devs to allow the Make-Movie branch to be integrated?
Does it just have to play nice with the UI and utilize the default DB configs, or is there more required?
https://github.com/lbdc/zm-movie
This PHP script seems to be what Zoneminder needs for retrieving video, and I am willing to work on it's integration if I knew what it needed to have to be considered for implementing
@knnniggett @connortechnology

newburns, 2015-04-14:
Just so everyone knows, https://github.com/lbdc/zm-movie is not dependent on Mencoder anymore.
I think this would be a much better implementation of video retrieval/creation.
Am I chasing a dead issue? Barking up the wrong tree?

knnniggett, 2015-06-17:
That looks to be a good idea, but it is currently not something we are willing to directly integrate.  The reasoning is that it is just a single file and can be easily integrated manually ...but perhaps more importantly we are devoting our efforts towards zoneminder streaming direct to a video container file, which would make zm-movie an unnecessary feature.

closed 2015-06-17 by knnniggett