ZoneMinder/zoneminder #748: Event(s) not deleted from disk when deleted in Zoneminder

JoelLinn, 2015-03-18:
When I use a filter to delete multiple or simply delete a single event, the event is removed from the interface only. The event folder is symlinked to a directory on a btrfs mount.
Sadly I can't do further testing on the production machine, like changing filesystem etc...

Machine is Debian 7.7 running on XenServer, ZoneMinder 1.28.0

knnniggett, 2015-03-18:
If OPT_FAST_DELETE is enabled under options (the default) then this is by design.  You should read the help text concerning this option for a description.  The zmaudit daemon will clean up orphans the next time it runs.

Also, you should not symlink the event folder.  Instead, follow the instructions in the Wiki if this filesystem is on a local drive: http://www.zoneminder.com/wiki/index.php/Using_a_dedicated_Hard_Drive

Naturally, adjust the parameters to match your filesystem.

closed 2015-08-10 by knnniggett

doublehp, 2017-01-12:
I just ran /usr/bin/zmaudit.pl manualy.

According to ZM web interface, the oldest event for camera 8 is 2016-dec-14 th.

According to filesystem ...

546# ls /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8
16  17
547# ls /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8/16/12/
08  14  15  16  17  18  19  20  21  22  23  24  25  26  27  28  29  30  31
548# ls /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8/16/12/08/
18
549# ls /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8/16/12/08/18/
42
550# ls /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8/16/12/08/18/42/
55
552# ls /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/8/16/12/08/18/42/55/ | wc
    615     615   11070

So I have an event for dec 8th that is out if all ZM dbs. It may never be deleted by any later script.

This is an issue for me; I have writen a script that relies on the filesystem to determine the oldest even of a camera. This kind of error by ZM will produce false resuts for me.

Several cameras have similar residues.