ZoneMinder/zoneminder #752: Playback video problem in v1.26.5

demisx, 2015-03-24:
Hi guys. I am running v1.26.5 on Ubuntu 14 and recently, I am seeing issues playing back recorded videos. The video plays OK for a few seconds, and then it splits vertically, changes color to purple with `Time to next event` message plus a countdown timer shows up in the middle of it (see attached image). Do you have any idea what the issue may be? Everything else seems to work OK. Thank you.

[image]

connortechnology, 2015-03-25:
I thought 1.26.5 wouldn't work on ubuntu 14...

No idea what the issue would be.  You could turn on more debugging... go into options -> logging check relevant checkboxes.  turn level up to 6 or so.. 

most of the heavy lifting is done using ffmpeg libraries. 

My next suggestion would be upgrading to 1.28.1 or master dev branch.

demisx, 2015-03-25:
Thank you for the recommendations. I'll try to upgrade and report back. I've noticed that this happens only if I select "All Events" in the `Replay` dropdown. If I select "Gapless Events", then this doesn't happen.

P.S. FYI, I've been running it on Ubuntu 14 for  a few months and haven't noticed any obvious issues, except this one.

SteveGilvarry, 2015-06-17:
Let us know if you are seeing this post upgrade to 1.28.1 and we can re-open issue

closed 2015-06-17 by SteveGilvarry