ZoneMinder/zoneminder #753: Make Uninstall no longer removes Perl modules

knnniggett, 2015-03-25:
I just noticed the following when issuing a make uninstall from a from-source build:

```
Making uninstall in ZoneMinder
make[2]: Entering directory `/home/abauer/ZoneMinder/scripts/ZoneMinder'

Uninstall is unsafe and deprecated, the uninstallation was not performed.
We will show what would have been done.

unlink /usr/share/man/man3/ZoneMinder.3pm
unlink /usr/share/man/man3/ZoneMinder::Base.3pm
unlink /usr/share/man/man3/ZoneMinder::Config.3pm
unlink /usr/share/man/man3/ZoneMinder::ConfigAdmin.3pm
unlink /usr/share/man/man3/ZoneMinder::ConfigData.3pm
unlink /usr/share/man/man3/ZoneMinder::Control.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::3S.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::AxisV2.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::FI8608W_Y2k.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::FI8620_Y2k.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::FI8908W.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::FI9821W_Y2k.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::M8640.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::Ncs370.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::PanasonicIP.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::PelcoD.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::PelcoP.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::SkyIPCam7xx.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::TVIP862.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::Toshiba_IK_WB11A.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::Visca.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::Wanscam.3pm
unlink /usr/share/man/man3/ZoneMinder::Control::mjpgStreamer.3pm
unlink /usr/share/man/man3/ZoneMinder::Database.3pm
unlink /usr/share/man/man3/ZoneMinder::General.3pm
unlink /usr/share/man/man3/ZoneMinder::Logger.3pm
unlink /usr/share/man/man3/ZoneMinder::Memory.3pm
unlink /usr/share/man/man3/ZoneMinder::Trigger::Channel.3pm
unlink /usr/share/man/man3/ZoneMinder::Trigger::Channel::File.3pm
unlink /usr/share/man/man3/ZoneMinder::Trigger::Channel::Handle.3pm
unlink /usr/share/man/man3/ZoneMinder::Trigger::Channel::Inet.3pm
unlink /usr/share/man/man3/ZoneMinder::Trigger::Channel::Serial.3pm
unlink /usr/share/man/man3/ZoneMinder::Trigger::Channel::Spawning.3pm
unlink /usr/share/man/man3/ZoneMinder::Trigger::Channel::Unix.3pm
unlink /usr/share/man/man3/ZoneMinder::Trigger::Connection.3pm
unlink /usr/share/man/man3/ZoneMinder::Trigger::Connection::Example.3pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Base.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Config.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/ConfigAdmin.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/ConfigData.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/3S.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/AxisV2.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/FI8608W_Y2k.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/FI8620_Y2k.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/FI8908W.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/FI9821W_Y2k.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/LoftekSentinel.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/M8640.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/Ncs370.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/PanasonicIP.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/PelcoD.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/PelcoP.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/SkyIPCam7xx.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/TVIP862.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/Toshiba_IK_WB11A.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/Visca.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/Wanscam.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/mjpgStreamer.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Control/onvif.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Database.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/General.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Logger.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Memory.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Memory/Mapped.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Memory/Shared.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Trigger/Channel.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Trigger/Channel/File.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Trigger/Channel/Handle.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Trigger/Channel/Inet.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Trigger/Channel/Serial.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Trigger/Channel/Spawning.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Trigger/Channel/Unix.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Trigger/Connection.pm
unlink /usr/share/perl5/vendor_perl/ZoneMinder/Trigger/Connection/Example.pm
unlink /usr/lib64/perl5/vendor_perl/auto/ZoneMinder/.packlist

Uninstall is unsafe and deprecated, the uninstallation was not performed.
Please check the list above carefully, there may be errors.
Remove the appropriate files manually.
Sorry for the inconvenience.

make[2]: Leaving directory `/home/abauer/ZoneMinder/scripts/ZoneMinder'
```

This is going to cause problems if/when a zoneminder upgrade places the new Perl modules into a different location.

Still searching for a solution to this.

onlyjob, 2015-03-31:
Quick searching with [duckduckgo](https://duckduckgo.com) gave me the following links where I found information about `UNINST=1` trick:

  * http://perldoc.perl.org/ExtUtils/MakeMaker.html
  * https://github.com/Perl-Toolchain-Gang/ExtUtils-MakeMaker/blob/master/Changes
  * https://github.com/Perl-Toolchain-Gang/ExtUtils-MakeMaker/issues/129

If you're not afraid of consequences (why uninstall was disabled) you can try something like `make install UNINST=1`.

----

Why bother about `make uninstall` anyway? If I'm not mistaken it has been deprecated  for nearly 20 years. These days everything should be installed using package management software, especially if you're using binary GNU/Linux distro.

knnniggett, 2015-03-31:
I completely agree that packaging is the way to go.  Personally, I refuse to install anything from source on a production machine.  However, there are still a lot of users out there, Debian users for example, whose only option to get a modern version of zoneminder is to build from source.

onlyjob, 2015-03-31:
Debian users can build package from source and then install this package. It would be silly to install directly from source `make install` way even if it goes to `/usr/local`.
Besides Debian users can install Zoneminder package from official Debian repository:
 https://packages.qa.debian.org/z/zoneminder.html

SteveGilvarry, 2015-03-31:
LOL well they can now.

knnniggett, 2015-04-01:
@onlyjob 
I don't disagree with what you are saying, but we have a large user base who, unfortunately, have a history of building from source.  Some have been doing it for a long time.  I'd like to say we have most of the Ubuntu, CentOS, & Fedora users using the packages from our repos, but there are still some gaps.  

Pulling the rug out from them by stopping support is not the answer, however.  I would rather not  deal with the hords of angry nerd posts we would get if we just stopped supporting builds from source.  We aren't Microsoft! :-)

Instead, we need to ease them away from it by offering them a better solution.  now that your Debian packages are going, I think that will be a big step.

knnniggett, 2015-06-17:
I don't believe there is a direct solution for this.

Going to close this one out, making note of the following workarounds:
- Install from package, rather than from source, as the packaging scripts will properly remove the Perl modules on package removal.
- Alternatively, the user can add "UNINST=1" to ZM_PERL_MM_PARMS.  We can't turn this on by default, because it breaks packaging scripts.

closed 2015-06-17 by knnniggett