ZoneMinder/zoneminder #754: Max FPS and Alarm Max FPS disable when Source Type IP Camera

SteveGilvarry, 2015-03-26:
Mentioned on irc that we should fix this so logging the issue so we remember to get to it. We should be able to manage in the GUI to clear the values and hide the fields when FFMPEG, Remote, cURL and Libvlc Source Types are selected. I assume these sources are only used for IP Cameras?

The other option is to leave the fields as is but make sure the values have no impact, but that could still confuse users.

knnniggett, 2015-03-26:
Agreed.
What complicates this is that there is at least one case I can think of where we have a source type of "remote" and we need those fields to exists.
(typical older) IP cameras that create snapshots, rather than send an mjpeg or h264 stream will need those fields.

knnniggett, 2015-03-26:
I just thought about another potential case....
If an ffmpeg camera goes offline, the fps spikes which pegs the cpu.  A possible workaround is to place a value in the MAX Fps field higher than the normal frame rate.  Of course, this is just a workaround so one could argue that we should address the root cause instead.  

Perhaps we could make the fields "advanced", requiring a check box to be checked in order to make them re-appear.

newburns, 2015-03-26:
I was going to put in my 2 cents, but Andrew beat me to it.
I put in the Max FPS so that my CPU doesn't go to 100 load.

SteveGilvarry, 2015-03-26:
Yes I wanted to find the root cause, as I move my dev laptop between home and office and have different cameras so if I forget to disable and enable then things go nuts and I spend 20minutes waiting for responses to shut down. I am all for it reconnecting, but it needs a throttle on it. Do we have an issue for that also?
Back to the jpeg case I will have to see if I can find some examples of how people use them to see if we can find a combination that says enable Max FPS or disable.

SteveGilvarry, 2015-03-26:
Related https://github.com/ZoneMinder/ZoneMinder/issues/365

connortechnology, 2015-03-26:
I think maybe they should be relevant.  

For example ffmpeg... maybe we don't need to do motion detection on every frame.  Maybe every 5th.. even though you record all.  

I agree that they should only be available if they have an effect.

arjunroychowdhury, 2015-03-29:
Hello, I am curious why the CPU spikes if the camera is offline? I had kept my FPS in ZM blank, but given that it does spike, I've kept my ZM FPS MAX at a value higher than the in-camera FPS. But it wasn't evident to me why this is happening? I took at look at zm_ffmpeg_camera.cpp --> seems if it fails to open an ffmpeg stream, it waits for the configured ffmpeg timeout and launches another thread to try again - I don't see max-fps being used here. So I'm curious how this helps (I confirmed, it does help)

PR #1033 (2015-12-19): add warning and help text for maxfps fields
fixes #754 

I don't find this solution visually elegant, and I'm open to suggestions if anyone has other ideas.

Screenshots are incoming....

NOTE that I did experiment with triggering a popup when the user clicked into the maxfps field, but that got annoying real quick.  I settled for a note, in red, appearing to the right of the box.

Also, what you see in the screenshots below does not appear for existing monitors with a type of "local' or "file" since the issue does not affect those types.
closed 2015-12-19 by SteveGilvarry