ZoneMinder/zoneminder #755: bttv crashing zoneminder rpm

clipo1979, 2015-03-26:
Zoneminder crashes after several hours of running then reboots but after several days / weeks the system become very unstable and refuses to reboot.

I have installed both centos 6 and 7 rpms on fresh installs of both is with the same results.

No trial has been done on a from source build.

If you let me know what information is needed to debug I will post.

connortechnology, 2015-03-26:
zm can't ( shouldn't be able to) crash your box.  Sounds like a bug in the bttv driver.  We aren't responsible for that.

You can turn up your logging levels and send us the output.

connortechnology, 2015-03-27:
What card is it, and what bttv options are you using?

clipo1979, 2015-03-27:
Options use with bttv.conf are

options bttv card=77 tuner=4 radio=0 triton1=0 vsfx=0 autoload=0

I will increase debug and post results.

knnniggett, 2015-04-04:
I don't use local devices anymore so I won't be able to help much, but the last time I did, I used the following for a generic pico2000 card:
```
# bttv module setting for the pico2000 card
# simply copy this file to your /etc/modprobe.d folder 
options i2c-algo-bit bit_test=1
options bttv gbuffers=32 card=77 tuner=4 radio=0 coring=1 full_luma_range=1 chroma_agc=1 combfilter=1 autoload=0 triton1=0 vsfx=0 no_overlay=0
```

I no longer remember what the options mean.

knnniggett, 2015-06-17:
Closing due lack to data suggesting the problem lies within zoneminder.
You may reopen this issue if you acquire new information that suggests otherwise.

It may be worth posting your inquiry on the zoneminder forum.  I'm sure there are many others, that use the bttv driver, who can share their experiences.

closed 2015-06-17 by knnniggett