ZoneMinder/zoneminder #757: Probe of cURL monitor fails

onlyjob, 2015-03-30:
Monitor capture images successfully but probing fails as follows:
~~~~
Unable to probe local cameras, status is '255'
~~~~

SteveGilvarry, 2015-03-30:
I believe this is related to the fact that cURL and Libvlc cameras all added later and not setup in Probe. A few issues already discuss it, I think #674 is still open to address at some point. Even then I believe there is only a very small subset of cameras that work. Onvif discovery could be the solution, well for Onvif cameras anyway.

knnniggett, 2015-03-30:
The root cause of the error message is because your system cannot find the arp command.
This has been fixed in master with a series of commits: 
7ef503ff591a5fd15e23552dc83bc1cbcb51bcce
438456ce93dccb7f3035bc41136d5e7013784f40
13c6be548d7852a15eb2c0a1010a2dd6b9b688b8

The monitor probe depends on a small, static table of known mac address OUI's.  In practice it is not very helpful.  @SteveGilvarry mentioned the open issue to improve it.  Please place your comments there if you wish to discuss.

closed 2015-03-30 by knnniggett

onlyjob, 2015-03-30:
@knnniggett, I applied three commits you've mentioned to 1.28.1 and I'm still having this issue. Not sure why it is closed -- please re-open or shall we continue in #674 which seems to be about probing different type of cameras (non-cURL )?

knnniggett, 2015-03-30:
Rather than cherry pick the commits, can you build from master?  It is quite possible I missed a commit as there were different attempts to address this.  If you are still having an issue after you build from master, then you can re-open this issue and post your logs.  Those commits have added (hopefully better) feedback as to what is going on and we need to see that.

Also not that the Probe feature doesn't know anything about the Source Type you've selected from teh drop down, whether it be curl, remote, local, ffmpeg, etc.  It only knows about IP cameras, and only a very few of them as you can see here:
https://github.com/ZoneMinder/ZoneMinder/blob/master/web/skins/classic/views/monitorprobe.php#L280

Any comments about how to make this better should go in #674