ZoneMinder/zoneminder #758: zmc: Invalid JPEG file structure: two SOI markers

onlyjob, 2015-03-30:
I'm pulling JPEG frames from web camera but Zoneminder is unable to process them: `zmc` complains to log:

~~~~
Invalid JPEG file structure: two SOI markers
~~~~

from `zoneminder-1.28.1/src/zm_jpeg.cpp` line _40_.

knnniggett, 2015-03-30:
These are coming from the jpeg library, not zoneminder.  I'm not sure there is anything we can do about this.

onlyjob, 2015-03-30:
I'm not sure either but it seems that some sort of workaround is necessary as Zoneminder seems to be the only application having trouble with such files. Strangely enough browsers and viewers have no problems displaying such JPEG...

knnniggett, 2015-03-30:
True, but the other applications you mention probably are not trying to break the stream up into individual jpeg files and save them to disk.... they are just showing the stream, which is why they don't fail. 

If you Google "Two SOI Markers", you can see zoneminder is not the only application to experience this issue.  I don't know of any way to easily work around this, but maybe someone else does.  

We have a development branch going to modernize our recording method so we record to a container file rather than jpeg.  Hopefully that should extend to local devices and not just ip cameras.

knnniggett, 2015-07-07:
Was reading about something else when I came across the following post:
https://community.ubnt.com/t5/UniFi-Video/FW-1-2-and-ZoneMinder/td-p/538086

connortechnology, 2015-07-07:
Yeah I've seen that mentioned before too.  I think the header parsing code has issues.  I hate even more that we have two versions of it.

I have some cameras that sometimes give the two SOI markers message.  I am slowly auditing all this code.  Would love help.

connortechnology, 2015-07-07:
maybe we should junk our parser code and use something like:
https://github.com/joyent/http-parser

knnniggett, 2015-07-30:
@onlyjob have you had a chance to followup with the latest suggestion?

I had no idea until just now that this same suggestion is also in the wiki:
http://www.zoneminder.com/wiki/index.php/FAQ#What_causes_.22Invalid_JPEG_file_structure:_two_SOI_markers.22_from_zmc_.281.24.x.29

onlyjob, 2015-07-31:
In my case _simple_ or _regex_ methods make no difference -- with either method I'm getting `Invalid JPEG file structure: two SOI markers...

hpsilva, 2015-08-08:
regex does not solve either in my case.

What i find interesting is that i am using a pair of cameras from the same model, and one works perfectly whilst the model with more recent firmware fails. This makes me believe this might be an issue related with the encoding algorithm from the camera, which in the later model is not compatible with the lib used by zoneminder or linux

So to solve the issue, i am considering downgrading the camera firmware.

Any other suggestion?

onlyjob, 2018-06-16:
No activity doesn't mean that the problem is gone.

closed 2018-08-22 by stale[bot]

Simpler1, 2023-05-08:
I know this is an old issue, but I thought this might help someone (or even future me).

I was getting this error when I had 
General -> Source Type: Remote

Once I changed this to
General -> Source Type: Ffmpeg

The errors went away.