ZoneMinder/zoneminder #760: source code mess

onlyjob, 2015-03-30:
### Some Perl modules contain POD stubs:

~~~~
Below is stub documentation for your module. You'd better edit it!
~~~~

and some modules are undocumented.

### Bad code style

All over the sources there are unusually long lines.
For instance `CMakeLists.txt` files are hard to read and edit because of long lines.
Wrapping is easy enough and it makes code a lot more readable.

Same issues exist in CPP code, etc.

### `cppcheck` errors

~~~~
[zm_image.cpp:644]: (error) Resource leak: outfile
[zm_logger.cpp:540]: (error) Dangerous usage of 'classString' (strncpy doesn't always null-terminate it).
[zm_user.cpp:80]: (error) Mismatching allocation and deallocation: User::monitor_ids
[zm_utils.cpp:81]: (error) Using reference 'format' as parameter for va_start() results in undefined behaviour.
[zm_zone.cpp:908]: (error) Mismatching allocation and deallocation: str_ptr
[zm_zone.cpp:924]: (error) Mismatching allocation and deallocation: str_ptr
~~~~

Perhaps it would be a good idea to routinely use `cppcheck` especially during pre-release clean-up.

SteveGilvarry, 2015-03-30:
FYI I was looking at how to wrap lines in cmakelists.txt, found the continuation character, but also found this.
"CMake versions prior to 3.0 do not support continuation with \. They report errors in quoted arguments containing lines ending in an odd number of \ characters."
Project doesn't have any official coding standards so might be a first step to adopt and document some guidelines.

onlyjob, 2015-03-30:
Well at least we could add end-of-line characters to separate arguments inside round brackets... It would be a big readability improvement even if we won't break any quoted parameters (just start a new argument on new line). Perhaps it is worth checking with CMake v2 just to make sure but I think I've seen such syntax in other projects...

onlyjob, 2015-04-05:
`perlcritic /usr/bin/zm*.pl` found multiple issues in `.pl` scripts:

~~~~
usr/bin/zmaudit.pl: Subroutine prototypes used at line 85, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/bin/zmaudit.pl: Subroutine prototypes used at line 86, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/bin/zmaudit.pl: Subroutine prototypes used at line 87, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/bin/zmaudit.pl: Glob written as <...> at line 150, column 27.  See page 167 of PBP.  (Severity: 5)
/usr/bin/zmaudit.pl: Glob written as <...> at line 158, column 35.  See page 167 of PBP.  (Severity: 5)
/usr/bin/zmaudit.pl: Glob written as <...> at line 255, column 24.  See page 167 of PBP.  (Severity: 5)
/usr/bin/zmaudit.pl: Subroutine prototypes used at line 448, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/bin/zmaudit.pl: Subroutine prototypes used at line 461, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/bin/zmaudit.pl: Subroutine prototypes used at line 501, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/bin/zmdc.pl: Bareword file handle opened at line 236, column 10.  See pages 202,204 of PBP.  (Severity: 5)
/usr/bin/zmdc.pl: Two-argument "open" used at line 236, column 10.  See page 207 of PBP.  (Severity: 5)
/usr/bin/zmupdate.pl: Stricture disabled at line 665, column 13.  See page 429 of PBP.  (Severity: 5)
~~~~

There are even more issues in modules `perlcritic /usr/share/perl5/ZoneMinder/*.pm`:

~~~~
/usr/share/perl5/ZoneMinder/Config.pm: Bareword file handle opened at line 75, column 2.  See pages 202,204 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/Config.pm: Two-argument "open" used at line 75, column 2.  See page 207 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/Control.pm: Subroutine prototypes used at line 78, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/Control.pm: Subroutine prototypes used at line 148, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/Database.pm: Subroutine prototypes used at line 73, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/Database.pm: Subroutine prototypes used at line 97, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/Database.pm: Subroutine prototypes used at line 113, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/Database.pm: Subroutine prototypes used at line 155, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/Database.pm: Subroutine prototypes used at line 171, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 79, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 93, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 165, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 199, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 216, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Bareword file handle opened at line 253, column 9.  See pages 202,204 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Two-argument "open" used at line 253, column 9.  See page 207 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Bareword file handle opened at line 263, column 9.  See pages 202,204 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Two-argument "open" used at line 263, column 9.  See page 207 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 275, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 294, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 306, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 320, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 450, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 458, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 491, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 544, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 588, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 599, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 601, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Subroutine prototypes used at line 652, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/General.pm: Expression form of "eval" at line 712, column 18.  See page 161 of PBP.  (Severity: 5)
[... cut ...]
/usr/share/perl5/ZoneMinder/Memory.pm: Subroutine prototypes used at line 643, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/Memory.pm: Subroutine prototypes used at line 662, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/Memory.pm: Subroutine prototypes used at line 677, column 1.  See page 194 of PBP.  (Severity: 5)
/usr/share/perl5/ZoneMinder/Memory.pm: Subroutine prototypes used at line 692, column 1.  See page 194 of PBP.  (Severity: 5)
~~~~

SteveGilvarry, 2015-04-15:
Any issues with making some PSR-2 php coding style changes? 
Finding this pretty useful.  https://scrutinizer-ci.com/g/SteveGilvarry/ZoneMinder/issues/master
We could setup on the ZoneMinder/ZoneMinder, instead of just on my own. We also might want to split this thread off into Perl, C and PHP, adopt a coding standard for each and also see what options we have to automate checking of PR's meeting the standard.

connortechnology, 2015-04-15:
I think it's a good idea.

commit 2549865c (2015-05-19): Merge pull request #812 from SteveGilvarry/Cmakelists_CleanUp

Fixes #760 in part Clean up CMakeLists.txt
closed 2015-05-19 by knnniggett