ZoneMinder/zoneminder #762: libAV deprecations

onlyjob, 2015-03-30:
Compilation on Debian Jessie revealed multiple deprecations:

~~~~
src/zm_ffmpeg.cpp:79:18: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)
src/zm_ffmpeg.cpp:79:38: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)
src/zm_ffmpeg.cpp:85:19: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)
src/zm_ffmpeg.cpp:85:39: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)

src/zm_local_camera.cpp:614:16: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)
src/zm_local_camera.cpp:614:36: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)
src/zm_local_camera.cpp:844:24: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)
src/zm_local_camera.cpp:844:44: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)

src/zm_mpeg.cpp:326:14: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)
src/zm_mpeg.cpp:326:35: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)
src/zm_mpeg.cpp:347:19: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)
src/zm_mpeg.cpp:347:40: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)

src/zm_remote_camera_rtsp.cpp:194:17: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)
src/zm_remote_camera_rtsp.cpp:194:37: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)
src/zm_remote_camera_rtsp.cpp:197:14: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)
src/zm_remote_camera_rtsp.cpp:197:34: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:3195)

src/zm_sdp.cpp:407:45: warning: ‘AVCodecContext::codec_name’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:1065)
src/zm_sdp.cpp:407:45: warning: ‘AVCodecContext::codec_name’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:1065)
src/zm_sdp.cpp:407:112: warning: ‘AVCodecContext::codec_name’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:1065)
src/zm_sdp.cpp:407:112: warning: ‘AVCodecContext::codec_name’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:1065)
src/zm_sdp.cpp:423:45: warning: ‘AVCodecContext::codec_name’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:1065)
src/zm_sdp.cpp:423:45: warning: ‘AVCodecContext::codec_name’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:1065)
src/zm_sdp.cpp:423:113: warning: ‘AVCodecContext::codec_name’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:1065)
src/zm_sdp.cpp:423:113: warning: ‘AVCodecContext::codec_name’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:1065)
src/zm_sdp.cpp:431:30: warning: ‘AVCodecContext::codec_name’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:1065)
src/zm_sdp.cpp:431:30: warning: ‘AVCodecContext::codec_name’ is deprecated (declared at /usr/include/libavcodec/avcodec.h:1065)

SteveGilvarry, 2015-03-30:
Already logged in #580

knnniggett, 2015-03-30:
Staying on top of the libav/ffmpeg mess seems to be a full time job in and of itself.
I agree we need to address it, but for simplicity lets keep the conversation in #580 as @SteveGilvarry mentioned.

closed 2015-03-30 by knnniggett