ZoneMinder/zoneminder #764: Remote Zoneminder installations

newburns, 2015-03-30:
How hard would it be to implement a remote Zoneminder install?
Basically:
multiple processing servers - locally or intranet wide
Single frontend server for montage or server group views
All commands, monitors, and SQL changes can be done from the single front-end view. All of the processing happens on independent servers.

connortechnology, 2015-03-30:
Some of the work has already been done by me.  Lots more work to do to make it work better.

newburns, 2015-06-04:
So how does it work @connortechnology ?
I have a site with over 40 monitors, and Zoneminder is not handling them well. Load is ridiculously high.
We have been through enormous amounts of RAM, CPU cores, and Disk I/O (which is above 600 Mb/s).
I want to get things started for Remote installs

connortechnology, 2015-06-04:
It works by assigning a recording server to each monitor.  So you have 1 central db (preferably on it's own machine).  You need a shared disk space SAN or something else.  I'm working on a storage area patch so that each server can record to a local disk and just share that disk via nfs to the other servers.

newburns, 2015-06-04:
I have zoneminder running in a Proxmox Virtualized Cluster. It has several SAN storage areas.
Ideally, I assume you mean barebone installs of Zoneminder, but I see no reason my scenario would not work.
>It works by assigning a recording server to each monitor
So I would need essentially 40+ CentOS installs of Zoneminder for each monitor?
Is there a central interface to view all monitors? How does montage work?

Do you have a branch with the Remote Setup working?

connortechnology, 2015-06-04:
The iconnor dev branch on github has my stuff in it.  However it also has a ton of other stuff like the h264 video branch merged in and may or may not be totally stable.  
I am in the process of merging the cluster stuff into master.  It is going to take a while though.

There is a relatively stable ubuntu ppa of my stuff at ppa:~iconnor/zoneminder-cluster

connortechnology, 2015-06-04:
No, you just as many servers as you need.  You can put 10 (or more, or less) monitors on each server.
The web interface can be run on any of the servers, or all of them, or a dedicated machine.  Montage works the same as always.

knnniggett, 2015-06-17:
Closing as this thread does not describe a bug or a feature request.  Please take discussions like this to the ZoneMinder user forum.

closed 2015-06-17 by knnniggett