ZoneMinder/zoneminder #766: Issue when viewing events when users are restricted to limited cameras

goosfraba01, 2015-03-30:
We ran into this problem for our users using IE when they are restricted to only a few cameras.  Users with access to all cameras don't have the issue.

Limited access get this
[image]

Full access get this
[image]

In my digging I think the issue is with the following codes in skins/classic/views/event.php call to dbEscape().

```php
if ( $user['MonitorIds'] )
    $midSql = " and MonitorId in (".join( ",", preg_split( '/["\'\s]*,["\'\s]*/', dbEscape($user['MonitorIds']) ) ).")";
else
    $midSql = '';
```
dbEscape adds quotes around the string so what should be ("31,21,29") in the SQL query becomes ("'31,32,32'").  So it wants to match the full string instead of each number individually.  

I added the following in our environment to fix the issue, but am not sure if that is the right way to fix the issue.

```php
if ( $user['MonitorIds'] )
//    $midSql = " and MonitorId in (".join( ",", preg_split( '/["\'\s]*,["\'\s]*/', dbEscape($user['MonitorIds']) ) ).")";
    $midSql = " and MonitorId in (".trim(join( ",", preg_split( '/["\'\s]*,["\'\s]*/', dbEscape($user['MonitorIds']) ) ), "'" ).")";
else
    $midSql = '';
```
This is all documented in this thread on the forum (http://www.zoneminder.com/forums/viewtopic.php?f=32&t=22993)

Thanks in advance,

Kenny

connortechnology, 2015-03-30:
This has already been fixed.

closed 2015-03-30 by connortechnology

goosfraba01, 2015-03-30:
Sorry I missed that.  Thanks again,

Kenny